introlab/rtabmap #563: Camera Pose Extraction

Zumbalamambo, 2020-06-11:
I have created a db file using Google Tango-RTABMap application. How do I extract the camera pose corresponding to each and every image from the tango db?

matlabbe, 2020-06-11:
You can open the database with `rtabmap-databaseViewer` (or in RTAB-Map->File->Export poses...) and do File->Export poses... you have also the options to extract all corresponding RGB and depth frames.

Zumbalamambo, 2020-06-11:
thank you!

1. May I know what do these values represent?

"0.188878 -0.700506 0.688198 0.000599 -0.981995 -0.132266 0.134879 -0.018083 -0.003458 -0.701283 -0.712875 0.011538"

I'm more eager to find the translation params x,y,z and  quaternions w,rx,ry,rz params

2. How can I map the camera pose to the image that corresponds to the pose since the filename of the image is not mentioned in the poses.txt? 

3. While exporting the pose, there were two options.. 1. Camera 2. Robot 3. Scan.. what do they represent?

Zumbalamambo, 2020-06-12:
@matlabbe I have exported the poses.txt file and the rgb frames. there are totally 146 images while there are 155 poses. How can I sort this out?

matlabbe, 2020-06-23:
The options "1. Camera 2. Robot 3. Scan.." are used to export the poses of the corresponding frame. For example, we may want the poses in camera frame instead of robot frame if we are exporting the poses to a photogrammetry software.

Did you export with databaseViewer or RTAB-Map? can you share the database?

closed 2020-09-01 by Zumbalamambo

shrinivas96, 2023-05-21:
Hello @matlabbe, thanks for your insights. I have some questions, which I am not sure if this is the right place to ask. If not, please let me know and I will change it accordingly.

 I also want to extract the poses after recording a session of the robot driving around in an environment (in my case, turtlebot3 house world). I recorded for nearly 190 seconds.

1. When I do the "RTAB-Map->File->Export poses...->Raw format (*.txt)" I get a file containing about 119 lines, each containing 12 values. What do these numbers represent? How can I make use of them? (While exporting poses it opens a dialog box asking "Which poses?" with the available options "Map's graph (see Graph view)" or "Odometry". Both text files have 12 values in each line, but the total number of lines differ.)

2. Exporting in RGBD-SLAM format gives a file with values that are explained on top of the file. Namely, the timestamp, 3D pose, and orientation in quaternions. Plotting the `(x, y)` from this file results in a nearly same trajectory as the robot actually took. Screenshot: 
[image]

How do I get more poses from the entire session? That is, I seem to have only 119 different poses for the entire session, and I would like to be able to query the pose and the scans at any given time.

3. You mentioned that while exporting poses there is an option to extract images. I assume the functionality has changed since you wrote that and now "Extract images" is its own thing in the File menu. Nonetheless, clicking on that option and selecting a RGBD format gives 3 folders of 119 images and yaml files. Is it again possible to get more images? And instead of images is it possible to get laser scans?

I assume the `.db` file is naturally not saving everything or else the file size would be too huge, so maybe it is not possible from the `.db` file. Alternatively, could you point me to a resource which I could read to understand where I can get these while the `rtabmap_slam` node is operational?

matlabbe, 2023-05-29:
The raw format (12 values) is the 3x4 transformation matrix, line by line.
```
Matrix:
R00 R01 R02 Tx 
R10 R11 R12 Ty 
R20 R21 R22 Tz

or in 12 values:
R00 R01 R02 Tx R10 R11 R12 Ty R20 R21 R22 Tz
```

The RGBD-SLAM format + Index is maybe the most useful format if you don't care working with quaternions.

You could get more poses with `Rtabmap/CreateIntermediateNodes=true` (actually all odometry poses would be saved). If you need also data of those poses, set `Mem/IntermediateNodeDataKept=true`.

For exporting the laser scans individually, you may use File->Export 3D Map menu option, uncheck "From RGB-D images..." and "Assemble clouds/meshes...". You may set "Set th enumber of k nearest neighbors..." to 0 and "Voxel size" to 0 to export original scans. The "Output frame" can be used to export the scans in map frame (already transformed based on pose in the map), in base frame (relative to base link of the robot) or in scan frame (relative to lidar sensor itself).

cheers,
Mathieu