introlab/rtabmap #564: Alicevision

tkircher, 2020-06-15:
In this commit, the variable me was renamed to mesh (with the original mesh being renamed to geoMesh), and PointsVisibilities was removed from the Texturing class (and moved to Mesh). It also appears that some of the datatypes have changed, making an update nontrivial.

https://github.com/alicevision/AliceVision/commit/62bf6d4e0fe9c48e18a9e78bf87afde55bfa1590#diff-6e6d218f593d49b0e3ec3188b6c545f8

So essentially, the latest version of AliceVision supported here for now is 2.1.0.

matlabbe, 2020-06-22:
It seems that I use v2.2.0 here: https://github.com/introlab/rtabmap/blob/c5ec4f337bb077a550f16d00a699b99f5900bb6d/docker/bionic/Dockerfile#L87
but with a [patch](https://gist.githubusercontent.com/matlabbe/469bba5e7733ad6f2e3d7857b84f1f9e/raw/edaa88ed38344219af1cc919a5597f5a74445336/alice_vision_eigen.patch). I'll try with latest AliceVision version.

tkircher, 2020-06-22:
You're right, the v2.2.0 release doesn't have this issue, it's just the current branch.

```
/home/tkircher/clean_build/rtabmap/corelib/src/util3d_surface.cpp:2248:12: error: ‘struct aliceVision::mesh::Texturing’ has no member named ‘me’; did you mean ‘mesh’?
  texturing.me = new mesh::Mesh();
            ^~
            mesh
/home/tkircher/clean_build/rtabmap/corelib/src/util3d_surface.cpp:2249:12: error: ‘struct aliceVision::mesh::Texturing’ has no member named ‘me’; did you mean ‘mesh’?
  texturing.me->pts = new StaticVector<Point3d>(cloud2.size());
            ^~
            mesh
/home/tkircher/clean_build/rtabmap/corelib/src/util3d_surface.cpp:2250:12: error: ‘struct aliceVision::mesh::Texturing’ has no member named ‘pointsVisibilitie’; did you mean ‘remapVisibilities’?
  texturing.pointsVisibilities = new mesh::PointsVisibility();
            ^~~~~~~~~~~~~~~~~~
            remapVisibilities

tkircher, 2020-11-20:
Incidentally, this is still an outstanding issue. Alicevision has had significant changes since 2.2.0. If nothing else, a stub checking for the version and throwing an error if it's too new.

matlabbe, 2020-11-20:
Hi @tkircher,

I tried v2.3.0. First, we need to apply this [patch](https://gist.github.com/matlabbe/9c5be02afa1b5e65583facb69a41e44c) to avoid segmentation fault when using g2o or GTSAM (built with system's eigen version):
```bash
cd AliceVision
    git checkout v2.3.0
    wget https://gist.githubusercontent.com/matlabbe/9c5be02afa1b5e65583facb69a41e44c/raw/3f56541e3acd1873acef2f152e3d9dd0792bd257/alice_vision_eigen_v2_3_0.patch
    git apply alice_vision_eigen_v2_3_0.patch
```

While the final OBJ with AliceVision can be exported, the code seg fault just after when freeing aliceVision::sfmData object. Here is the output:
```gdb
(gdb) bt
#0  0x00007fffda6479fd in __GI___libc_free (mem=0x41) at malloc.c:3103
#1  0x00007fffcb7da89f in std::_Rb_tree<unsigned int, std::pair<unsigned int const, std::shared_ptr<aliceVision::sfmData::View> >, std::_Select1st<std::pair<unsigned int const, std::shared_ptr<aliceVision::sfmData::View> > >, std::less<unsigned int>, Eigen::aligned_allocator<std::pair<unsigned int const, std::shared_ptr<aliceVision::sfmData::View> > > >::_M_erase(std::_Rb_tree_node<std::pair<unsigned int const, std::shared_ptr<aliceVision::sfmData::View> > >*) () at /usr/local/lib/libaliceVision_sfmData.so.2
#2  0x00007fffcb7da875 in std::_Rb_tree<unsigned int, std::pair<unsigned int const, std::shared_ptr<aliceVision::sfmData::View> >, std::_Select1st<std::pair<unsigned int const, std::shared_ptr<aliceVision::sfmData::View> > >, std::less<unsigned int>, Eigen::aligned_allocator<std::pair<unsigned int const, std::shared_ptr<aliceVision::sfmData::View> > > >::_M_erase(std::_Rb_tree_node<std::pair<unsigned int const, std::shared_ptr<aliceVision::sfmData::View> > >*) () at /usr/local/lib/libaliceVision_sfmData.so.2
#3  0x00007fffcb7da875 in std::_Rb_tree<unsigned int, std::pair<unsigned int const, std::shared_ptr<aliceVision::sfmData::View> >, std::_Select1st<std::pair<unsigned int const, std::shared_ptr<aliceVision::sfmData::View> > >, std::less<unsigned int>, Eigen::aligned_allocator<std::pair<unsigned int const, std::shared_ptr<aliceVision::sfmData::View> > > >::_M_erase(std::_Rb_tree_node<std::pair<unsigned int const, std::shared_ptr<aliceVision::sfmData::View> > >*) () at /usr/local/lib/libaliceVision_sfmData.so.2
#4  0x00007fffcb7da875 in std::_Rb_tree<unsigned int, std::pair<unsigned int const, std::shared_ptr<aliceVision::sfmData::View> >, std::_Select1st<std::pair<unsigned int const, std::shared_ptr<aliceVision::sfmData::View> > >, std::less<unsigned int>, Eigen::aligned_allocator<std::pair<unsigned int const, std::shared_ptr<aliceVision::sfmData::View> > > >::_M_erase(std::_Rb_tree_node<std::pair<unsigned int const, std::shared_ptr<aliceVision::sfmData::View> > >*) () at /usr/local/lib/libaliceVision_sfmData.so.2
#5  0x00007fffcb7da875 in std::_Rb_tree<unsigned int, std::pair<unsigned int const, std::shared_ptr<aliceVision::sfmData::View> >, std::_Select1st<std::pair<unsigned int const, std::shared_ptr<aliceVision::sfmData::View> > >, std::less<unsigned int>, Eigen::aligned_allocator<std::pair<unsigned int const, std::shared_ptr<aliceVision::sfmData::View> > > >::_M_erase(std::_Rb_tree_node<std::pair<unsigned int const, std::shared_ptr<aliceVision::sfmData::View> > >*) () at /usr/local/lib/libaliceVision_sfmData.so.2
#6  0x00007fffcb7da875 in std::_Rb_tree<unsigned int, std::pair<unsigned int const, std::shared_ptr<aliceVision::sfmData::View> >, std::_Select1st<std::pair<unsigned int const, std::shared_ptr<aliceVision::sfmData::View> > >, std::less<unsigned int>, Eigen::aligned_allocator<std::pair<unsigned int const, std::shared_ptr<aliceVision::sfmData::View> > > >::_M_erase(std::_Rb_tree_node<std::pair<unsigned int const, std::shared_ptr<aliceVision::sfmData::View> > >*) () at /usr/local/lib/libaliceVision_sfmData.so.2
#7  0x00007fffcb7da875 in std::_Rb_tree<unsigned int, std::pair<unsigned int const, std::shared_ptr<aliceVision::sfmData::View> >, std::_Select1st<std::pair<unsigned int const, std::shared_ptr<aliceVision::sfmData::View> > >, std::less<unsigned int>, Eigen::aligned_allocator<std::pair<unsigned int const, std::shared_ptr<aliceVision::sfmData::View> > > >::_M_erase(std::_Rb_tree_node<std::pair<unsigned int const, std::shared_ptr<aliceVision::sfmData::View> > >*) () at /usr/local/lib/libaliceVision_sfmData.so.2
#8  0x00007fffcb7da875 in std::_Rb_tree<unsigned int, std::pair<unsigned int const, std::shared_ptr<aliceVision::sfmData::View> >, std::_Select1st<std::pair<unsigned int const, std::shared_ptr<aliceVision::sfmData::View> > >, std::less<unsigned int>, Eigen::aligned_allocator<std::pair<unsigned int const, std::shared_ptr<aliceVision::sfmData::View> > > >::_M_erase(std::_Rb_tree_node<std::pair<unsigned int const, std::shared_ptr<aliceVision::sfmData::View> > >*) () at /usr/local/lib/libaliceVision_sfmData.so.2
#9  0x00007fffcb7da875 in std::_Rb_tree<unsigned int, std::pair<unsigned int const, std::shared_ptr<aliceVision::sfmData::View> >, std::_Select1st<std::pair<unsigned int const, std::shared_ptr<aliceVision::sfmData::View> > >, std::less<unsigned int>, Eigen::aligned_allocator<std::pair<unsigned int const, std::shared_ptr<aliceVision::sfmData::View> > > >::_M_erase(std::_Rb_tree_node<std::pair<unsigned int const, std::shared_ptr<aliceVision::sfmData::View> > >*) () at /usr/local/lib/libaliceVision_sfmData.so.2
[... cut ...]  > > > > > const&, std::pair<float, float> const&, bool) () at /home/mathieu/workspace/rtabmap_cv_latest/bin/librtabmap_core.so.0.20
#16 0x000055555556cd47 in main ()
```
It seems again related to Eigen memory allocation. In the patch, I disable [this](https://github.com/alicevision/AliceVision/blob/63fe7f085d6169e4848a9b8ec6aa3c12301ed338/src/aliceVision/numeric/numeric.hpp#L11-L23), not sure if it is related. It is not explained why they don't support default Eigen aligment. 

Regards,
Mathieu

tkircher, 2020-11-21:
Thanks for looking into this again! I'll be talking to the alicevision devs about this.