introlab/rtabmap #565: 2D costmap issue

chiaueng, 2020-06-15:
Dear Sir,
I am currently using the ZED camera in order to generate a 2D costmap in RTABmap. I put the database of my testing environment in https://drive.google.com/file/d/195tDODDRV_LQ62dbloImD8zjfqr9BBxi/view?usp=sharing. 
In this database, there is a ceiling in id 44 to 50. I try to do some experiments to avoid the ceiling seen as an obstacle. For example, setting the restrictions for the obstacle height from 0.8m to1.2m:
[image] 
However, I still get poor result as below (
I draw the red frames around the places that should not be regarded as an obstacle)
[image]

I looked into the database, and it seems there is a big bias for z value (Ranged from -6~3 meters). With this condition, it becomes meaningless to set the obstacle height. May I have your kindly suggestion for how to handle this condition ?

i ask this question because I would like to ignore the obstacles which is too high. For example, the ceiling should not be an obstacle for the navigation.  I also hope the viewers can understand the 
traversable path from the 2D grip map.
Thank you in advance.

Sincerely,
Chiaueng

matlabbe, 2020-06-30:
If you know that the floor should be always the same height, you could enable `Reg/Force3DoF` to avoid drift along z axis. If you are using ZED odometry, not sure if they have the option to make it 2D.

For the ceiling, it is because the parameter `Grid/MaxObstacleHeight` is done against the base frame of the robot. In your case, there is no TF between the base of he robot (ground) and the camera. The referential is the camera frame, so the height is relative to that frame. In database viewer, we can see that the occupancy grid is at the height of the camera:
[image]
generating black pixels on this frame. But if I set `Grid/MaxObstacleHeight` to `0.5`, the ceiling is ignored:
[image]

If you want to similate the hight of the camera on a robot, you could add a [static_transform_publisher](http://wiki.ros.org/tf#static_transform_publisher) between `base_link` and `camera_link` frames with z=1.3 meters, and set `frame_id` of rtabmap to `base_link` instead of `camera_link`.

chiaueng, 2020-07-07:
Thank you for the suggestion. I tried to set Grid/MaxObstacleHeight = 0.5, and the ceiling is disappeared as you said.
However, I saw other problems from the database, may I have your comments for how to improve it ?
1. The Z value drift a lot while I go back by the same route, which results in two layers of grounds as below. 
[image]
PS: In the above database, I have already enabled Reg/Force3DoF, but it still doesn't help.
[image]
2. The occupancy grid in my database is almost on the sky. But the occupancy grid you show me on above image is not so high as my result. Is there a way to tune it to similar as your result ?
[image]

matlabbe, 2020-07-11:
The whole map is tilted. It looks like the camera started tilted. For the z drift, it is a common problem. However, if you close the loop, it would correct that drift. You didn't say which odometry approach you used, some are more robust to drift than others. Is the zed camera the original ZED without IMU?

`Reg/Force3DoF=true` will affect only graph optimization if you are not using rgbd_odometry.