introlab/rtabmap #566: parameter conflicts/ during reprocess

oxidane-lin, 2020-06-17:
Hi, matlabbe:

I am exploring through your excellent work, but i met this issue: when I use rtabmap-reprocess to reprocess my database with some parameters, the changes I made won't work. To be clear: when I run the following command:

`rtabmap-reprocess --Kp/DetectorStrategy 2 --Vis/FeatureType 2 from.db to.db`

In my from.db, the Kp/DetectorStrategy=6, Vis/FeatureType=6; the generated database to.db is still the same size with from.db. Checking in the databaseViewer, in the info column, the parameters are changed to 2(=ORB); but the features on the image are still GFTT/BRIEF, not ORB. My current solution is to only set Kp/DetectorStrategy=2, at the same time set Vis/FeatureType to another number, and it seems to work, I can see the images with ORB features.

Here is my question:
1. Are the parameters with prefix Kp works the same with a specific parameter, such as Kp/MaxDepth and Vis/MaxDepth? I've digged a little into the code, they seem to work on different priorities.
2. Am I using the rtabmap-reprocess the right way?

my platform: Ubuntu 16.04, rtabmap 0.20 from recent build.

matlabbe, 2020-07-06:
Set `Mem/UseOdomFeatures` to false to make sure features are reextracted with the detector you are using:
```bash
rtabmap-reprocess --Mem/UseOdomFeatures false --Kp/DetectorStrategy 2 --Vis/FeatureType 2 from.db to.db
```
For convenience, I also updated reprocess tool in the commit above to set it to false automatically when we change the detector.

oxidane-lin, 2020-07-13:
Thank you! This worked for me. Will close this issue.

closed 2020-07-13 by oxidane-lin