introlab/rtabmap #568: FindGeogram.cmake should be included in cmake_modules

tkircher, 2020-06-21:
For certain build configurations, cmake will fail if this is not included.

matlabbe, 2020-06-22:
[geogram is a dependency of AliceVision](https://github.com/introlab/rtabmap/blob/c5ec4f337bb077a550f16d00a699b99f5900bb6d/docker/bionic/Dockerfile#L76-L84), normally we should not have to explicitly search for it if we already search for AliceVision. What is the cmake error?

crossref with https://github.com/introlab/rtabmap/issues/564 if it also a version problem

tkircher, 2020-06-22:
```
-- AliceVision Find_Package
-- ----
-- Found AliceVision version: 2.1.0
-- Installed in: /opt
-- Used AliceVision libraries: 
-- ----
CMake Error at CMakeLists.txt:517 (find_package):
  By not providing "FindGeogram.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Geogram", but
  CMake did not find one.
```

I built and installed the official version of geogram externally rather than the internal AliceVision repository version.

matlabbe, 2020-06-22:
Just saw the Geogram dependency is explicitly searched here:
https://github.com/introlab/rtabmap/blob/ff5695878cca85794b960dce6d1c763ecfba83ef/CMakeLists.txt#L516-L517
I think it works in the docker as `FindGeogram.cmake` would be installed in `/usr/local/lib/cmake/modules`. You may check where `FindGeogram.cmake` is installed on your computer when you `make install` Geogram.

tkircher, 2020-06-22:
Yep, my mistake. I must have forgotten to set the CMAKE_MODULE_PATH.

closed 2020-06-22 by tkircher

matlabbe, 2020-06-23:
For info, the line in Geogram where it installs FindGeogram.cmake:
https://github.com/alicevision/geogram/blob/98def2ce2393c04c9a6dbf034483deefd690da7b/CMakeLists.txt#L149

tkircher, 2020-06-23:
It's in the same place in the official release as well.