introlab/rtabmap #569: INSTALL_FAILED_OLDER_SDK

borongyuan, 2020-06-23:
It seems the latest RTABMap-debug.apk in docker image is not compatible with tango tablet, which is running android 4.4.2

matlabbe, 2020-06-23:
For Tango phones, use tango docker image ([instructions](https://github.com/introlab/rtabmap/wiki/Installation#rtab-map-tango-apk)):
```
$ docker pull introlab3it/rtabmap:tango
```

borongyuan, 2020-06-28:
Thanks for your help. I have tried `introlab3it/rtabmap:tango-api19 `, but encountered the same problem. The Yellowstone tablet was released before Phab2Pro. Maybe they are outdated :( I could use an apk file from an early build of `introlab3it/rtabmap:tango` released this year. The compatibility issue seems to be cause by commit 012439aa0b3fdb3067023b33fa30ff29ed8895fe.
I also notice that you have supported Huawei P30 Pro. I happen to have this device too. The apk file from `introlab3it/rtabmap:android24` works fine on this device :)

matlabbe, 2020-07-10:
I've found the problem, the arm64-v8a package cannot include both 32 and 64 bits binaries for android API19. The resulting package is automaticcally API21. For the Project Tango tablet, use the package in armeabi-v7a instead. Based on the [docker instructions](https://github.com/introlab/rtabmap/wiki/Installation#rtab-map-tango-apk):

```bash
$ docker run --name=rtabmap-tango introlab3it/rtabmap:tango-api19
$ docker cp rtabmap-tango:/root/rtabmap-tango/build/armeabi-v7a/app/android/bin/RTABMap-debug.apk .
$ docker rm rtabmap-tango # cleanup container
$ adb install -r RTABMap-debug.apk
```
Just tested on a tango tablet and it works.

closed 2020-07-10 by matlabbe