introlab/rtabmap #572: RTAB-Map for Android not compatible anymore

InfSA, 2020-07-19:
Hi when I try to install on my Sony XPERIA 1 I receive the following:
error: "your device is't compatible with this version"
Phone has TOF, Lidar etc.
Would love to use the app as is shown in the video "RTAB-Map Tango 0.11.14 on Phab2Pro"

matlabbe, 2020-07-30:
As ARCore's depth API doesn't work exactly the same on all devices, it is difficult to support all of them without having to tweak for each device... buying all the phones to support them is not a solution for this free project... it is why I don't officially support android app. Currently only the Huawei P30Pro should be working out of the box with RTAB-Map android-26+ app. I still support Tango phones like Zenfone and Phab2Pro because I own them.

InfSA, 2020-07-31:
Thank you, very much.
Which one of the Zenfone or Phab2Pro would you suggest?
Any specific model?

matlabbe, 2020-07-31:
[ZenFone AR (ZS571KL)](https://www.asus.com/Phone/ZenFone-AR-ZS571KL/)

closed 2020-07-31 by InfSA

Vicssss, 2023-12-02:
Hello , just for other people , it works on Samsung S20+ 5G with TOF camera , first time i open it crash when trying to start mapping but after some day i try and it works perfectly ! now going to try to add ROS

parkerlreed, 2023-12-04:
> Hello , just for other people , it works on Samsung S20+ 5G with TOF camera , first time i open it crash when trying to start mapping but after some day i try and it works perfectly ! now going to try to add ROS

Really??? Where are you getting the APK? Play Store wont let you install it.

parkerlreed, 2023-12-04:
I tried this release and it just instantly crashes running on the S20 Plus.

https://github.com/introlab/rtabmap/releases/download/0.20.2/RTABMap-0.20.2-tango.apk

parkerlreed, 2023-12-04:
No need to experiment. There's "11*" devices with ARCore and TOF sensor. Should work well for any of these.

For non TOF devices, you can enable ARCore NDK driver and turn on Motion based depth under Mapping but this produces wildly garbled results.

* Google duplicated the S10 and counting the major model revisions there's only 6

Source: https://developers.google.com/ar/devices

* LG	V60 ThinQ	Supports multiple GPU texture resolutions - 1080p, 720p, 480p
Supports Depth API
Supports time-of-flight (ToF) hardware depth sensor

* LG	V60 ThinQ 5G	Supports multiple GPU texture resolutions - 1080p, 720p, 480p
Supports Depth API
Supports time-of-flight (ToF) hardware depth sensor

* Samsung	Galaxy A80	Supports Depth API
Supports time-of-flight (ToF) hardware depth sensor

* Samsung	Galaxy Note10+	Supports multiple GPU texture resolutions - 1080p, 720p, 480p
Supports Depth API
Supports time-of-flight (ToF) hardware depth sensor

* Samsung	Galaxy Note10+ 5G	Supports multiple GPU texture resolutions - 1080p, 720p, 480p
Supports Depth API
Supports time-of-flight (ToF) hardware depth sensor

* Samsung	Galaxy S10 5G	Supports multiple GPU texture resolutions - 1080p, 720p, 480p
Supports Depth API
Supports time-of-flight (ToF) hardware depth sensor

* Samsung	Galaxy S20+	Supports Depth API
Supports time-of-flight (ToF) hardware depth sensor

* Samsung	Galaxy S20+ 5G	Supports Depth API
Supports time-of-flight (ToF) hardware depth sensor

* Samsung	Galaxy S20 Ultra 5G	Supports Depth API
Supports time-of-flight (ToF) hardware depth sensor

* Sharp	AQUOS R5G	Supports Depth API
Supports time-of-flight (ToF) hardware depth sensor

Vicssss, 2023-12-05:
Ok , thanks , so last things to do its to add ROS outputs to this app hahaha , did you have any idea ?

parkerlreed, 2023-12-05:
No idea there, sorry. My experience with RTAB-Map has always been for standalone scanning. 

I have made my setup slightly fancier though.

I'm using KDE Connect on the phone to share the databases to my computer and auto opening in the desktop version of RTAB-Map

```
#!/bin/sh
export LD_LIBRARY_PATH=/home/deck/.local/lib
inotifywait -m -q -r -e moved_to --format '%f' "$HOME/Downloads/KDEConnect" |
        while IFS='' read -r filename; do
                echo "$filename"
                case "$filename" in
                    *.db)
                        mv "$HOME"/Downloads/KDEConnect/"$filename" "$HOME"/Documents/RTAB-Map/
                        LD_LIBRARY_PATH=" " rtabmap "$HOME"/Documents/RTAB-Map/"$filename" &
                        ;;
                    *)
                        echo "Not a known file"
                esac
        done
```

giuseppepaglia, 2024-03-11:
Hi @matlabbe,
There is any android device compatible with rtabmap that can be used with L515 realsense?

parkerlreed, 2024-03-11:
> Hi @matlabbe, There is any android device compatible with rtabmap that can be used with L515 realsense?

I do not believe the Android version supports external sensors. It has drivers for sensors found built-into devices but nothing externally.

cerrajeriasabaris, 2024-06-23:
Hello... does anyone know of this compressed app to use a realsense with android as a 3D scanner?
something like dot3d or similar

parkerlreed, 2024-06-24:
The Android application as far as I'm aware does not support any external scanners. 

It either uses the depth hardware if available or the regular camera through ARCore estimation.

cerrajeriasabaris, 2024-06-25:
thank you parkerlreed 
 How could I get your apk to test?
I don't know how to program and I can't get an apk for a samsung s20

Vicssss, 2024-06-25:
> thank you parkerlreed 
>  How could I get your apk to test?
> I don't know how to program and I can't get an apk for a samsung s20

Hello , i don't remember where exactly but you can find the APK in a docker container on this repo (rtabmap) , then you can use ADB command to install it to your phone , i use it on S20+,  got very good result !

cerrajeriasabaris, 2024-06-25:
Yes, I saw your YouTube video, but I have no idea what the apk is or how to install it with adb, I have an s22 ultra but I am willing to buy an s20 ultra to have a scanner... as soon as I find an apk

Vicssss, 2024-06-25:
I didn't have any video on youtube 😅 but here is a link to the apk i have :
 https://drive.google.com/file/d/1Y2RGWFzaqTP6FrJ16HEfXJFW_vHRKPaw/view?usp=drivesdk 

download and try , it dosent work the first time but for unknow reason it work some day after !

parkerlreed, 2024-06-25:
`docker save introlab3it/rtabmap:android30`

Should give you a tarball you can extract to get the file. It's annoying there's no easy way to grab a single file from an online Docker image.

parkerlreed, 2024-06-25:
Ok so that didn't work. Ended up having to use skopeo

https://github.com/containers/skopeo

```
[parker@rogally bin]$ ./skopeo copy docker://introlab3it/rtabmap:android30 docker-archive:$PWD/android.tar
Getting image source signatures
Copying blob 284055322776 done   | 
Copying blob da5eb365e843 done   | 
Copying blob 730bc09a9a49 done   | 
Copying blob 29d1155c5e78 done   | 
Copying blob a61a6998e17f done   | 
Copying blob 624651a90a0e done   | 
Copying blob 26397dcec9b9 done   | 
Copying blob 26397dcec9b9 done   | 
Copying blob 52bc5a5a3363 done   | 
Copying blob b8ebe7e5f90d done   | 
Copying blob 4f4fb700ef54 done   | 
Copying blob 7ff7f9c5a53e done   | 
Copying config 5590f52bc2 done   | 
Writing manifest to image destination
[parker@rogally bin]$ tar -xOf android.tar 912b68b0424832ab726e88823816b8dd8e8b0f6afbb89decbeddbc5fb2b7f2d0.tar | tar --strip-components 7 -x root/rtabmap-tango/build/arm64-v8a/app/android/bin/RTABMap-debug.apk[parker@rogally bin]$ ls -lah RTABMap-debug.apk 
-rw-r--r-- 1 parker parker 18M Jun 16 14:27 RTABMap-debug.apk
```

matlabbe, 2024-06-26:
I added 0.21.4 apk in the release directly for convenience: https://github.com/introlab/rtabmap/releases/tag/0.21.4

cerrajeriasabaris, 2024-06-26:
ehhh... android30 apk works perfectly on s22 ultra... of course it doesn't have tof... connect the d435i and of course it doesn't work... but I think you have done a great job (from my ignorance) it is one of the best applications what I have tried.

cerrajeriasabaris, 2024-06-26:
https://github.com/state-of-the-art/Handy3DScanner    This application works perfectly with d400 and d435i but is not capable of doing what rtab-map does

 on  samsung s22ultra