introlab/rtabmap #574: L515 support

VisibleMeasurement, 2020-07-20:
Tried to use with my new L515 from Intel.  It does not show up in the supported list.  Wanted to know if there are plans for inclusion in a later release.  Thanks!

TheBricktop, 2020-07-29:
Lidar module would be awesome to use with rtanmap for scanning, its precision is unmatched with any other cheap modules.

blakely3dnm, 2020-07-29:
For the time being you can use it with rtabmap-ros via realsense-ros.

VisibleMeasurement, 2020-07-30:
Is there any chance you could point me to a guide on how to do that?

blakely3dnm, 2020-07-31:
A good start would be the RGBD tutorial for rtabmap-ros. 
http://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping

Since the L515 is new there are no tutorials for that camera specifically, but you can adapt documentation you find for other realsense cameras with minimal changes. Some understanding of ros topics and how to remap them in launch files would be very helpful. There is documentation for this on the ROS website. 

run the realsense2_camera node
run "rostopic list" and verify that the l515 is publishing a depth topic. something like "/camera/depth/image_rect_raw". 

You can then edit the examples for realsense cameras such as the D435i to use the depth topics from the L515 instead. just change the relevant lines in the ros launch file.

VisibleMeasurement, 2020-07-31:
Thanks!  I will take a look shortly.

matlabbe, 2020-08-02:
If the L515 uses a similar interface than D400 series from realsense2 library, you could try selecting realsense2 driver and see if it works. The only thing blocking me right now to integrate officially the L515 is that I didn't get my hands on one yet. We are opened for pull request if someone has one and want to look in [CameraRealSense2](https://github.com/introlab/rtabmap/blob/master/corelib/src/camera/CameraRealSense2.cpp) class to make it compatible with its L515 (I guess it would be pretty similar than with D435i if the lidar data is on a depth stream). 

Using the ROS interface is a workaround if you are using already ROS.

blakely3dnm, 2020-08-02:
I am not a developer but I tried to update CameraRealSense2.cpp to support the L515. 

Adding the following at line 574 seems to allow for initialization of the camera
`                 else if ("L500 Depth Sensor" == module_name)
                  {
                          sensors[0] = elem;
                          sensors[0].set_option(rs2_option::RS2_OPTION_LASER_POWER,100);
                  }

`
However, a problem arises with the RS2 wrapper in the Rtabmap GUI. L5 sensor 0 (depth) and sensor 1 (RGB) do not share common stream resolution options, so a given resolution will either fail for the depth or the RGB.

> [ERROR] (2020-08-02 13:10:21.671) CameraRealSense2.cpp:758::init() Given stream configuration is not supported by the device! Stream Index: 0, Width: 640, Height: 480, FPS: 30
> 
> [ERROR] (2020-08-02 13:10:53.760) CameraRealSense2.cpp:758::init() Given stream configuration is not supported by the device! Stream Index: 1, Width: 1920, Height: 1080, FPS: 30

This was causing issues for me when I tried RGBD odometry with the L515 in rtabmap-ros, and in order to fix it I had to use the align_depth launch parameter for the realsense2 camera node. That then publishes RGB and depth topics with matching dimensions. Doing this in c++ is a bit beyond my comprehension however. 

For reference, here is the relevant output from rs-enumerate-devices:

> 
> Stream Profiles supported by L500 Depth Sensor
>  Supported modes:
>     stream       resolution      fps       format   
>     Confidence	  1024x768	@ 30Hz	   RAW8
>     Confidence	  640x480	@ 30Hz	   RAW8
>     Confidence	  384x1024	@ 30Hz	   RAW8
>     Confidence	  240x640	@ 30Hz	   RAW8
>     Infrared	  1024x768	@ 30Hz	   Y8
>     Infrared	  768x1024	@ 30Hz	   Y8
>     Infrared	  640x480	@ 30Hz	   Y8
>     Infrared	  480x640	@ 30Hz	   Y8
>     Depth	  1024x768	@ 30Hz	   Z16
>     Depth	  768x1024	@ 30Hz	   Z16
>     Depth	  640x480	@ 30Hz	   Z16
>     Depth	  480x640	@ 30Hz	   Z16
> 
> Stream Profiles supported by RGB Camera
>  Supported modes:
>     stream       resolution      fps       format   
>     Color	  1920x1080	@ 30Hz	   RGB8
>     Color	  1920x1080	@ 30Hz	   Y16
>     Color	  1920x1080	@ 30Hz	   BGRA8
>     Color	  1920x1080	@ 30Hz	   RGBA8
>     Color	  1920x1080	@ 30Hz	   BGR8
>     Color	  1920x1080	@ 30Hz	   YUYV
>     Color	  1920x1080	@ 15Hz	   RGB8
>     Color	  1920x1080	@ 15Hz	   Y16
>     Color	  1920x1080	@ 15Hz	   BGRA8
>     Color	  1920x1080	@ 15Hz	   RGBA8
>     Color	  1920x1080	@ 15Hz	   BGR8
>     Color	  1920x1080	@ 15Hz	   YUYV
>     Color	  1920x1080	@ 6Hz	   RGB8
>     Color	  1920x1080	@ 6Hz	   Y16
>     Color	  1920x1080	@ 6Hz	   BGRA8
>     Color	  1920x1080	@ 6Hz	   RGBA8
>     Color	  1920x1080	@ 6Hz	   BGR8
>     Color	  1920x1080	@ 6Hz	   YUYV
>     Color	  1280x720	@ 60Hz	   RGB8
>     Color	  1280x720	@ 60Hz	   Y16
>     Color	  1280x720	@ 60Hz	   BGRA8
>     Color	  1280x720	@ 60Hz	   RGBA8
>     Color	  1280x720	@ 60Hz	   BGR8
>     Color	  1280x720	@ 60Hz	   YUYV
>     Color	  1280x720	@ 30Hz	   RGB8
>     Color	  1280x720	@ 30Hz	   Y16
>     Color	  1280x720	@ 30Hz	   BGRA8
>     Color	  1280x720	@ 30Hz	   RGBA8
>     Color	  1280x720	@ 30Hz	   BGR8
>     Color	  1280x720	@ 30Hz	   YUYV
>     Color	  1280x720	@ 15Hz	   RGB8
>     Color	  1280x720	@ 15Hz	   Y16
>     Color	  1280x720	@ 15Hz	   BGRA8
>     Color	  1280x720	@ 15Hz	   RGBA8
>     Color	  1280x720	@ 15Hz	   BGR8
>     Color	  1280x720	@ 15Hz	   YUYV
>     Color	  1280x720	@ 6Hz	   RGB8
>     Color	  1280x720	@ 6Hz	   Y16
>     Color	  1280x720	@ 6Hz	   BGRA8
>     Color	  1280x720	@ 6Hz	   RGBA8
>     Color	  1280x720	@ 6Hz	   BGR8
>     Color	  1280x720	@ 6Hz	   YUYV
>

blakely3dnm, 2020-08-02:
In case it was an easy fix I gave it a try. 
The L515 produces a D4xx format depth stream so in theory it should be easy.  I tried editing the CameraRealSense2 class by simply adding to line 571
```
		else if ("L500 Depth Sensor" == module_name)
		{
		}
```
but the depth and RGB streams for the L515 do not share any common resolution settings and therefore any resolution entered in the GUI will fail to initialize the camera. The CameraRealSense2 class assumes they are the same. I can't see an easy way to modify that behavior, and even if there were, I believe rtabmap would not make registered depth images if depth and RGB are different sizes. I ran into this problem with rtabmap-ros but was able to work around the issue by using the "align_depth" launch parameter in the Realsense2_Camera node, which then publishes RGB and depth topics with the same dimensions. I am not sure if there is a way to expose those data formats in a non-ROS implementation.

matlabbe, 2020-08-03:
The align depth function is called in CameraRealSense2 here:
https://github.com/introlab/rtabmap/blob/0a9d237ac2968463d36c4c9b4436871a6c3ea0ca/corelib/src/camera/CameraRealSense2.cpp#L1194-L1197

The depth stream from D400 is from **Stereo Module**, I think if the depth is coming from **L500 Depth Sensor**, we would have to set:
```cpp
if ("L500 Depth Sensor" == module_name)
{
	sensors[1] = elem;
}
```

Then for this part:
https://github.com/introlab/rtabmap/blob/0a9d237ac2968463d36c4c9b4436871a6c3ea0ca/corelib/src/camera/CameraRealSense2.cpp#L683-L696

If the depth sensor gives `RS2_FORMAT_Z16`, then there would be no change to do. You can put the logger in Preferences->Logging in **Info** level to see this log: 
https://github.com/introlab/rtabmap/blob/0a9d237ac2968463d36c4c9b4436871a6c3ea0ca/corelib/src/camera/CameraRealSense2.cpp#L635-L643
It will show all supported profiles from the depth sensor. If you can copy paste here that section of the log it would be useful.

blakely3dnm, 2020-08-03:
I still have the problem with this section

line 652
```
				//D400 series:
				if (video_profile.width()  == cameraWidth_ &&
					video_profile.height() == cameraHeight_ &&
					video_profile.fps()    == cameraFps_)
```
that condition can only be true for sensors[0] or sensors[1] - so only for either RGB or depth, but not both. 
Setting the resolution to be compatible with the depth stream we see:

>[ERROR] (2020-08-02 22:02:09.143) CameraRealSense2.cpp:779::init() Given stream configuration is not supported by the device! Stream Index: 0, Width: 640, Height: 480, FPS: 30

Or if we set it to a compatible resolution for RGB we get:

> [ERROR] (2020-08-02 22:03:39.263) CameraRealSense2.cpp:779::init() Given stream configuration is not supported by the device! Stream Index: 1, Width: 1280, Height: 720, FPS: 30

 

> Stream Profiles supported by L500 Depth Sensor
>  Supported modes:
>     stream       resolution      fps       format
>     Confidence	  1024x768	@ 30Hz	   RAW8
>     Confidence	  640x480	@ 30Hz [...]

matlabbe, 2020-08-03:
There is no compatible resolution between both indeed.  Instead of:
https://github.com/introlab/rtabmap/blob/0a9d237ac2968463d36c4c9b4436871a6c3ea0ca/corelib/src/camera/CameraRealSense2.cpp#L653-L655

For a fast test, can you try:
```cpp
if ((!ir_ && !irDepth_ && video_profile.stream_type() == RS2_STREAM_DEPTH) // L515 fix, use first available depth stream
    ||
    (video_profile.width()  == cameraWidth_ &&
    video_profile.height() == cameraHeight_ &&
    video_profile.fps()    == cameraFps_))
```

blakely3dnm, 2020-08-03:
With that edit, I am now unchecking both "IR emitter enabled" and "use depth image in IR mode". Entering 640 x 480 in the GUI results in the same error. However, setting 1920 x 1080 gives a new error:

```
terminate called after throwing an instance of 'rs2::not_implemented_error'
  what():  Requested extrinsics are not available!
Aborted (core dumped)
```
I have tried erasing the default extrinsics from the GUI in case it was trying to use that but the same error remains. I don't think it should look for extrinsic calibration if we are using the L515 in single mode. I will see if I can find a place where it goes down the "dual" path in the code...

blakely3dnm, 2020-08-05:
I have a feeling it is actually related to getting the extrinsic between depth and IMU or RGB and IMU (around line 854). I'm not quite sure how to see the index number of the sensors on the L5 but perhaps the index needs to be shifted in this case due to not being a stereo module.  I can sort of read C++ but I would need to study a lot to modify the code here.

peta-peta, 2020-08-23:
Any progress on the L515 integration? I have one and am using Ubuntu 18.04 LTS. I am happy to test any new developments on this topic, since I am very keen on some real time Lidar-odometry.

peta-peta, 2020-09-08:
> A good start would be the RGBD tutorial for rtabmap-ros.
> http://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping
> 
> Since the L515 is new there are no tutorials for that camera specifically, but you can adapt documentation you find for other realsense cameras with minimal changes. Some understanding of ros topics and how to remap them in launch files would be very helpful. There is documentation for this on the ROS website.
> 
> run the realsense2_camera node
> run "rostopic list" and verify that the l515 is publishing a depth topic. something like "/camera/depth/image_rect_raw".
> 
> You can then edit the examples for realsense cameras such as the D435i to use the depth topics from the L515 instead. just change the relevant lines in the ros launch file.
> 
> If you get both realsense-ros and rtabmap-ros installed, I can send you a launch file that works for me. Currently I have tried ICP odometry only.
> 
> P.S. I am not affiliated with the RTABMap project, but I wouldn't be surprised if there will be native support for the L515 soon.

Hi blakely3dnm, 
I am wondering if you could share your launch script. I am having problems to get `roslaunch rtabmap_ros rtabmap.launch ` running. The Realsense-ros node seems to run correctly, since I can view all topics, but I am not able to see any results in rtabmap. I start the realsense-ros node with `roslaunch realsense2_camera rs_camera.launch `Your help is appreciated.

pauldeee, 2020-09-08:
@peta-peta This should be a good place to start from... I've been using L515 with T265 on a 3D printed mount. This should launch L515 and map via ICP if you have everything installed properly. Change the SERIAL_NUMBER to match your serial number(s). Also, keep in mind the ICP will fail easily during daylight or if you're out of L515's usable range.

```<launch>
    <!-- use_sim_time?! T/F  -->
    <arg name="sim_time" value="false"/>
    <arg name="use_t265" value="false"/>
    <!-- start l515 -->
    <param unless="$(arg sim_time)" name="/l515/motion_module/global_time_enabled" value="true" />
    <param unless="$(arg sim_time)" name="/l515/rgb_camera/global_time_enabled" value="true" />
    <param unless="$(arg sim_time)" name="/l515/rgb_camera/enable_auto_exposure" value="false"/>
    <param unless="$(arg sim_time)" name="/l515/l500_depth_sensor/sensor_mode" value ="0"/>
    <!-- <param unless="$(arg sim_time)" name="/l515/rgb_camera/gain" value ="4096"/> -->
    <!-- HD MODE? true/false  --> 
    <arg unless="$(arg sim_time)" name="HD" value="false"/>
    <!-- Launch l515  -->  
    <include unless="$(arg sim_time)" file="$(find realsense2_camera)/launch/rs_camera.launch">
        <arg name="serial_no" value="L515_SERIAL_NUMBER"/>
        <arg name="unite_imu_method" value="linear_interpolation"/>
        <arg name="camera" value="l515"/>
        <arg name="enable_gyro" value="true"/>
        <arg name="enable_accel" value="true"/>
        <arg name="accel_fps" value="400"/>
        <arg name="gyro_fps" value="400"/>
        <arg name="filters" value="pointcloud"/>
        <arg name="color_fps" value="30"/>
        <arg name="pointcloud_texture_stream" value="RS2_STREAM_ANY"/>
        <arg name="align_depth" value="false"/>
        <!-- <arg name="initial_reset" value="true"/> -->
        <!-- High DEF CAM -->
        <arg if="$(arg HD)" name="depth_width" value="1024"/>
        <arg if="$(arg HD)" name="depth_height" value="768"/>
        <arg if="$(arg HD)" name="color_width" value="1920"/>
        <arg if="$(arg HD)" name="color_height" value="1080"/>        
        <!-- STANDARD DEF CAM -->
        <arg unless="$(arg HD)" name="depth_width" value="640"/>
        <arg unless="$(arg HD)" name="depth_height" value="480"/>
        <arg unless="$(arg HD)" name="color_width" value="1280"/>
        <arg unless="$(arg HD)" name="color_height" value="720"/>
    </include>

    <node if="$(arg use_t265)" pkg="tf" type="static_transform_publisher" name="static_tf" args="0 0 0.05 0 0 0 t265_link l515_link 10" />
    <node if="$(arg sim_time)" pkg="tf" type="static_transform_publisher" name="static_tf_1" args="0 0 0 0 0 0 t265_pose_frame t265_link 10" />

    <param if="$(arg use_t265)" name="/t265/tracking_module/enable_mapping" value="false"/>
    <include if="$(arg use_t265)" file="$(find realsense2_camera)/launch/rs_t265.launch">
        <arg name="serial_no" value="T265_SERIAL_NUMBER"/>
        <arg name="camera" value="t265"/>
        <arg name="unite_imu_method" value="linear_interpolation"/>
        <arg name="enable_fisheye1" value="true"/>
        <arg name="enable_fisheye2" value="true"/>
        <arg name="enable_sync" value="true"/>
        <arg name="publish_odom_tf" value="true"/>          
    </include>

    <arg name="rtabmapviz"    default="true"/>
    <arg name="scan_rate"    default="30"/>
    <arg name="rate"            value="2"/>
    <arg if="$(arg sim_time)" name="use_sim_time" value="true"/>
    <arg name="scan_cloud_assembling" value="false"/>
    <arg name="use_imu" value="true"/>
    <arg name="imu_topic" value="/imu/data"/>
    <arg name="frame_id" default="l515_link"/>
    <arg name="voxel_size" value="0.1"/>
    <!-- individual scan max size -->
    <arg name="scan_cloud_max_points" value="0"/>

    <arg name="wait_imu_to_init" value="true"/>
    <node pkg="imu_complementary_filter" type="complementary_filter_node" name="complementary_filter_node">
        <remap from="imu/data_raw" to="/l515/imu"/>
      </node>
[... 87 lines cut ...]
        <param name="Icp/PointToPlane"               type="string" value="true"/>
        <param name="Icp/Iterations"                 type="string" value="10"/>
        <param name="Icp/Epsilon"                    type="string" value="0.0001"/>
        <param name="Icp/MaxTranslation"             type="string" value="1"/>
        <param name="Icp/MaxCorrespondenceDistance"  type="string" value="$(arg voxel_size)"/>
        <param name="Icp/PM"                         type="string" value="true"/> 
        <param name="Icp/PMOutlierRatio"             type="string" value="0.9"/>
        <param name="Icp/CorrespondenceRatio"        type="string" value="0.01"/>
        <param name="Icp/PointToPlaneMinComplexity" value="0.0001"/> 

      </node>

    <!-- RTABMAPVIZ NODE  -->
      <node if="$(arg rtabmapviz)" name="rtabmapviz" pkg="rtabmap_ros" type="rtabmapviz" output="screen">
        <param name="frame_id" type="string" value="$(arg frame_id)"/>
        <param name="odom_frame_id" type="string" value="odom"/>
        <param name="subscribe_odom_info" type="bool" value="true"/>
        <param name="subscribe_scan_cloud" type="bool" value="true"/>
        <param name="approx_sync" type="bool" value="false"/>
        <remap if="$(arg scan_cloud_assembling)" from="scan_cloud" to="assembled_cloud"/>
        <remap unless="$(arg scan_cloud_assembling)" from="scan_cloud" to="/l515/depth/color/points"/>

      </node>
  </group>
</launch>```

peta-peta, 2020-09-09:
@pauldeee thank you for your quick answer. I created a launch with the content from your message above. I then executed it 
 with `roslaunch rtabmap_ros rtabmap_pauldeee.launch`. 

Following two modification were necessary:

1. I replaced `L515_SERIAL_NUMBER` with my serial number that I got with `rs-enumerate-devices` and than substituted it here:  <arg name="serial_no" value="L515_SERIAL_NUMBER"/> 
2. I had to disable the IMU here:   <arg name="use_imu" value="false"/>, because it was not detected. I don't know if it has something to do that I am not using a T265 in addition. Can it also run solely with the L515?

At least I am one step further. I am able to see a point cloud  in the viewer :) and it is also mapping, when I move the L515. However, there is no camera stream visible. Any Idea?

Next steps:
- How can I export the mapped pointcloud?

pauldeee, 2020-09-10:
@peta-peta If you want you use RGB you need to have the rtabmap node subscribe to it. Something like the following inside of the rtabmap_ros node. (which I slightly modified from the above launch)

```
<param name="subscribe_rgb" value="true"/>
<remap from="rgb/image"   to="/l515/color/image_raw" />
<remap from="rgb/camera_info" to="/l515/color/camera_info"/>
```

You may have to play with the [realsense2_camera launch parameters](https://github.com/IntelRealSense/realsense-ros#launch-parameters) to obtain what you would like. I disable color pointclouds to keep fast 30hz scan rate.

In order for the IMU to be used in this launch file it is dependent on the [IMU_complementary_filter](http://wiki.ros.org/imu_complementary_filter), make sure that is installed.

For exporting a pointcloud, simply pause RTAB-Map with the button near top left of GUI window. Then under edit -> view point clouds. If you like it then, file -> export 3D...

peta-peta, 2020-09-10:
@pauldeee I confirm that your hints are working for a) the rgb image and b) for exporting the pointcloud. I had switch off rgb-d field - untick first checkbox in the export function. Apparently the depth stream is missing?!

I ll try your gibt for the IMU as a next step. What is the purpose of the IMU in your case?

blakely3dnm, 2020-09-11:
Yes there are no depth images using that launch file, instead everything is done with laser scans. So odometry and mapping are done by geometry only. I have a similar launch file for RGBD odometry. It works but it is far from optimized. 

[..will edit to include launch..]

mdasari823, 2020-09-25:
Hi @blakely3dnm @pauldeee @peta-peta do you have a stable odometry with L515? I tried the launch files from both pauldee and blakely, but it seems they are giving really unstable odometry. 

Btw I am getting warnings like below when I run pauldee's launch file. Do you also face these warnings?

`[ WARN] [1600996614.940854309]: IcpOdometry: Transferring value 0.3 of "Icp/VoxelSize" to ros parameter "scan_voxel_size" for convenience. "Icp/VoxelSize" is set to 0.
[ WARN] [1600996614.941276012]: IcpOdometry: Transferring value 10 of "Icp/PointToPlaneK" to ros parameter "scan_normal_k" for convenience.
[ WARN] (2020-09-24 21:16:54.941) RegistrationIcp.cpp:428::parseParameters() Parameter Icp/PM is set to true but RTAB-Map has not been built with libpointmatcher support. Setting to false.
[ WARN] (2020-09-24 21:16:54.941) OdometryF2M.cpp:143::OdometryF2M() OdomF2M/BundleAdjustment=1 cannot be used with registration not done only with images (Reg/Strategy=1), disabling bundle adjustment.`

`[ WARN] [1600996615.300430879]: Setting "Grid/FromDepth" parameter to false (default true) as "subscribe_scan" or "subscribe_scan_cloud" is true. The occupancy grid map will be constructed from laser scans. To get occupancy grid map from cloud projection, set "Grid/FromDepth" to true. To suppress this warning, add <param name="Grid/FromDepth" type="string" value="false"/>
[ INFO] [1600996615.301135433]: RTAB-Map detection rate = 2.000000 Hz
[ INFO] [1600996615.301351539]: rtabmap: Deleted database "/home/wings/.ros/rtabmap.db" (--delete_db_on_start or -d are set).
[ INFO] [1600996615.301381363]: rtabmap: Using database from "/home/wings/.ros/rtabmap.db" (0 MB).
[ WARN] (2020-09-24 21:16:55.302) RegistrationIcp.cpp:428::parseParameters() Parameter Icp/PM is set to true but RTAB-Map has not been built with libpointmatcher support. Setting to false.
[ WARN] (2020-09-24 21:16:55.302) RegistrationIcp.cpp:428::parseParameters() Parameter Icp/PM is set to true but RTAB-Map has not been built with libpointmatcher support. Setting to false.`

pauldeee, 2020-09-25:
@mdasari823 For RTAB-Map's best performance with LIDAR you should have built [RTAB-Map from source](https://github.com/introlab/rtabmap_ros#build-from-source) and built with [libpointmatcher support](https://github.com/ethz-asl/libpointmatcher#quick-start).

According to the error you posted above:
```
Parameter Icp/PM is set to true but RTAB-Map has not been built with libpointmatcher support. Setting to false.
```

mdasari823, 2020-09-29:
Thanks @pauldeee. I rebuilt the RTAB-Map with libpointmatcher support. You are right it seems better now, with slow motion of course. But, I am curious why it fails so badly when there is heavy motion or jerks. For example, the Intel real sense T265 stereo camera does not fail even under such heavy motion. I would expect the d435i or L515 with depth info, it should be performing better than T265 right? I am not sure if this is a limitation of libpointmatcher or these depth/Lidar sensors are not as good as the stero cameras for Odometry. Any idea?

blakely3dnm, 2020-09-29:
@mdasari823 what update rate are you getting for your odometry? ICP is more expensive than VIO, so its possible that your update frequency is too low to accommodate fast motions. the T265 does use depth information from the stereo cameras and registers the visual features to the stereo depth. In the case of the T265 there are likely a few hundred to a few thousand points used for correspondences while ICP from the L515 uses hundreds of thousands of points even in VGA mode. Depending on the environment you could throw out a random fraction of the points in the lidar scans through a libpointmatcher option. With the limited testing I've done, I get good results with ~50% thrown out but doing much more than that results in lower accuracy.

mdasari823, 2020-09-29:
Thanks @blakely3dnm for the suggestions. I am not sure about the update rate, how do you get this? When I do the '`rostopic echo /rtabmap/odom/pose/pose`', I see really faster updates. Is that what you are referring to? 

Also, Isn't it that the computation should not be an issue because ICP is running on the host, i.e., my Laptop, not on the sensor itself?  So my question is more of-- if there is enough computation capacity, which one of them is better for tracking-- T265 stereo camera or d435 RGB-D or L515 Lidar?

blakely3dnm, 2020-09-29:
You can see the update rate in the terminal 
`[ INFO] [1600664613.655947447]: Odom: ratio=0.046694, std dev=0.015367m|0.015367rad, update time=0.033356s `
That update time I am getting is good enough to do smooth handheld motions and mapping at walking speed. 
The answer to which camera is better depends on your application, but I am guessing it will be the T265. If you are just trying to do tracking, then the T265 will do great. An important caveat being that the T265 requires light, visual features (i.e. not endless white walls), and ideally not repetitive visual features such as brick walls. 
Mostly what I try to do with RTAB-Map is to make high resolution 3D scans of places by hand held methods, with local error ideally within what is listed for the sensor. For that I would prefer the L515 (or even better if I had money, I would get an Ouster lidar), since while the T265 does well with odometry, it does not create a very accurate representation of surfaces. @pauldeee showed me a nice hybrid example using the T265 for odometry and taking the L515 scans for recording a detailed geometric representation, and I think for my purposes that is the way to go.

mdasari823, 2020-09-29:
Got it. Btw, thanks a bunch. I am getting the somewhat decent pose after all the help I got from you and @pauldeee. I am getting the update time of 10ms for a static scenario, i.e. when I am not moving. And when I start moving, it is varying all the way from 10ms to 70ms. To your point, how do you throw out points? can we change this number of throw out points dynamically? as in while we are moving?  

About the T265 vs. L515, yeah I understand. The T625 was specifically developed for tracking, whereas the other depth sensors for more for the 3D understanding of the environment. So, looks like they are better together, and we cannot compare them individually it seems.

blakely3dnm, 2020-09-29:
You can try setting the Icp/DownsamplingStep to a higher number in the RTABMap launch file ( for some reason, in my hands this has adverse effects on odometry refresh rate), or you can use a libpointmatcher .yaml configuration file. The schema ethz-asl provides on their Github is a bit out dated. After some trial and error this is what I came up with. 
```
readingDataPointsFilters:
  - RandomSamplingDataPointsFilter:
      prob: 0.5
  - SurfaceNormalDataPointsFilter:
      knn: 10
      keepDensities: 1
  - SimpleSensorNoiseDataPointsFilter:
      sensorType: 3
  - ObservationDirectionDataPointsFilter
  - OrientNormalsDataPointsFilter

referenceDataPointsFilters:
  - SamplingSurfaceNormalDataPointsFilter:
      knn: 10
      keepDensities: 1
matcher:
  KDTreeMatcher:
    maxDist: 0.3
    epsilon: 3.16 

outlierFilters:
  - TrimmedDistOutlierFilter:
      ratio: 0.65

errorMinimizer:
  PointToPlaneErrorMinimizer

transformationCheckers:
  - CounterTransformationChecker:
      maxIterationCount: 40
  - DifferentialTransformationChecker:
      minDiffRotErr: 0.001
      minDiffTransErr: 0.01
      smoothLength: 4   
  - BoundTransformationChecker:
      maxRotationNorm: 0.8
      maxTranslationNorm: 0.4
#inspector:
#  NullInspector

inspector:
 VTKFileInspector:
     baseFileName: pointmatcher-run1
     dumpPerfOnExit: 0
     dumpStats: 0
     dumpIterationInfo: 1 
     dumpDataLinks: 0 
     dumpReading: 0 
     dumpReference: 0 

logger:
  NullLogger
#  FileLogger

```

The relevant setting is "prob" under RandomSamplingDataPointsFilter

mdasari823, 2020-09-29:
Wow, thanks a lot. One last question, do you know what's the purpose of IMU in @pualdeee's launch file? Is it fusing IMU+ICP Odometry?

blakely3dnm, 2020-09-30:
Yes the IMU can constrain the movement of the pose. If you turn on the option to see gravity vectors in rtabmap viz you can verify that the IMU data is being used.

matlabbe, 2020-10-18:
For rtabmap standalone, a new commit has been added for L515: https://github.com/introlab/rtabmap/commit/3047b7da6b734628bdfe67eef1550d61a928a0d7

Anderson191919, 2020-11-22:
Hello guys, 
I have an issue with the L515 on windows with rtabmap (v 0.20.6 from https://ci.appveyor.com/project/matlabbe/rtabmap/branch/master/artifacts). unfortunately I get following error(s) when starting on rtabmap

Anderson191919, 2020-11-22:
[image]

Anderson191919, 2020-11-22:
I have used these settings (https://github.com/introlab/rtabmap/issues/614#issuecomment-721266476). 

Thank you very much in advance. 
best regards

Anderson191919, 2020-11-22:
[image]

VisibleMeasurement, 2020-12-31:
L515 support is now included in the 20.7 release, closing issue.

closed 2020-12-31 by VisibleMeasurement

MoonUniverse, 2021-01-05:
hello,everyone!
        I am using version 0.20.7,but i still get the following error!
```
[ERROR] (2021-01-05 10:31:09) CameraRealSense2.cpp:855::rtabmap::CameraRealSense2::init() Given stream configuration is not supported by the device! Stream Index: 0, Width: 640, Height: 480, FPS: 30
[ERROR] (2021-01-05 10:31:09) CameraRealSense2.cpp:856::rtabmap::CameraRealSense2::init() Available configurations:
[ERROR] (2021-01-05 10:31:09) CameraRealSense2.cpp:867::rtabmap::CameraRealSense2::init() RGB8 1920 1080 30 0 Color type=2
[ERROR] (2021-01-05 10:31:09) CameraRealSense2.cpp:867::rtabmap::CameraRealSense2::init() Y16 1920 1080 30 0 Color type=2
[ERROR] (2021-01-05 10:31:09) CameraRealSense2.cpp:867::rtabmap::CameraRealSense2::init() BGRA8 1920 1080 30 0 Color type=2
[ERROR] (2021-01-05 10:31:09) CameraRealSense2.cpp:867::rtabmap::CameraRealSense2::init() RGBA8 1920 1080 30 0 Color type=2
[ERROR] (2021-01-05 10:31:09) CameraRealSense2.cpp:867::rtabmap::CameraRealSense2::init() BGR8 1920 1080 30 0 Color type=2
[ERROR] (2021-01-05 10:31:09) CameraRealSense2.cpp:867::rtabmap::CameraRealSense2::init() YUYV 1920 1080 30 0 Color type=2
[ERROR] (2021-01-05 10:31:09) CameraRealSense2.cpp:867::rtabmap::CameraRealSense2::init() RGB8 1920 1080 15 0 Color type=2
[ERROR] (2021-01-05 10:31:09) CameraRealSense2.cpp:867::rtabmap::CameraRealSense2::init() Y16 1920 1080 15 0 Color type=2
[ERROR] (2021-01-05 10:31:09) CameraRealSense2.cpp:867::rtabmap::CameraRealSense2::init() BGRA8 1920 1080 15 0 Color type=2
[ERROR] (2021-01-05 10:31:09) CameraRealSense2.cpp:867::rtabmap::CameraRealSense2::init() RGBA8 1920 1080 15 0 Color type=2
[ERROR] (2021-01-05 10:31:09) CameraRealSense2.cpp:867::rtabmap::CameraRealSense2::init() BGR8 1920 1080 15 0 Color type=2
[ERROR] (2021-01-05 10:31:09) CameraRealSense2.cpp:867::rtabmap::CameraRealSense2::init() YUYV 1920 1080 15 0 Color type=2
[ERROR] (2021-01-05 10:31:09) CameraRealSense2.cpp:867::rtabmap::CameraRealSense2::init() RGB8 1920 1080 6 0 Color type=2
[ERROR] (2021-01-05 10:31:09) CameraRealSense2.cpp:867::rtabmap::CameraRealSense2::init() Y16 1920 1080 6 0 Color type=2
[ERROR] (2021-01-05 10:31:09) CameraRealSense2.cpp:867::rtabmap::CameraRealSense2::init() BGRA8 1920 1080 6 0 Color type=2
[ERROR] (2021-01-05 10:31:09) CameraRealSense2.cpp:867::rtabmap::CameraRealSense2::init() RGBA8 1920 1080 6 0 Color type=2
[ERROR] (2021-01-05 10:31:09) CameraRealSense2.cpp:867::rtabmap::CameraRealSense2::init() BGR8 1920 1080 6 0 Color type=2
[ERROR] (2021-01-05 10:31:09) CameraRealSense2.cpp:867::rtabmap::CameraRealSense2::init() YUYV 1920 1080 6 0 Color type=2
[ERROR] (2021-01-05 10:31:09) CameraRealSense2.cpp:867::rtabmap::CameraRealSense2::init() RGB8 1280 720 60 0 Color type=2
[ERROR] (2021-01-05 10:31:09) CameraRealSense2.cpp:867::rtabmap::CameraRealSense2::init() Y16 1280 720 60 0 Color type=2
[ERROR] (2021-01-05 10:31:09) CameraRealSense2.cpp:867::rtabmap::CameraRealSense2::init() BGRA8 1280 720 60 0 Color type=2
[ERROR] (2021-01-05 10:31:09) CameraRealSense2.cpp:867::rtabmap::CameraRealSense2::init() RGBA8 1280 720 60 0 Color type=2
[ERROR] (2021-01-05 10:31:09) CameraRealSense2.cpp:867::rtabmap::CameraRealSense2::init() BGR8 1280 720 60 0 Color type=2
[ERROR] (2021-01-05 10:31:09) CameraRealSense2.cpp:867::rtabmap::CameraRealSense2::init() YUYV 1280 720 60 0 Color type=2
[ERROR] (2021-01-05 10:31:09) CameraRealSense2.cpp:867::rtabmap::CameraRealSense2::init() RGB8 1280 720 30 0 Color type=2
[ERROR] (2021-01-05 10:31:09) CameraRealSense2.cpp:867::rtabmap::CameraRealSense2::init() Y16 1280 720 30 0 Color type=2
[ERROR] (2021-01-05 10:31:09) CameraRealSense2.cpp:867::rtabmap::CameraRealSense2::init() BGRA8 1280 720 30 0 Color type=2
[ERROR] (2021-01-05 10:31:09) CameraRealSense2.cpp:867::rtabmap::CameraRealSense2::init() RGBA8 1280 720 30 0 Color type=2
[ERROR] (2021-01-05 10:31:09) CameraRealSense2.cpp:867::rtabmap::CameraRealSense2::init() BGR8 1280 720 30 0 Color type=2
[ERROR] (2021-01-05 10:31:09) CameraRealSense2.cpp:867::rtabmap::CameraRealSense2::init() YUYV 1280 720 30 0 Color type=2
[ERROR] (2021-01-05 10:31:09) CameraRealSense2.cpp:867::rtabmap::CameraRealSense2::init() RGB8 1280 720 15 0 Color type=2
[ERROR] (2021-01-05 10:31:09) CameraRealSense2.cpp:867::rtabmap::CameraRealSense2::init() Y16 1280 720 15 0 Color type=2
[ERROR] (2021-01-05 10:31:09) CameraRealSense2.cpp:867::rtabmap::CameraRealSense2::init() BGRA8 1280 720 15 0 Color type=2
[ERROR] (2021-01-05 10:31:09) CameraRealSense2.cpp:867::rtabmap::CameraRealSense2::init() RGBA8 1280 720 15 0 Color type=2
[ERROR] (2021-01-05 10:31:09) CameraRealSense2.cpp:867::rtabmap::CameraRealSense2::init() BGR8 1280 720 15 0 Color type=2
[ERROR] (2021-01-05 10:31:09) CameraRealSense2.cpp:867::rtabmap::CameraRealSense2::init() YUYV 1280 720 15 0 Color type=2
[ERROR] (2021-01-05 10:31:09) CameraRealSense2.cpp:867::rtabmap::CameraRealSense2::init() RGB8 1280 720 6 0 Color type=2
[ERROR] (2021-01-05 10:31:09) CameraRealSense2.cpp:867::rtabmap::CameraRealSense2::init() Y16 1280 720 6 0 Color type=2
[ERROR] (2021-01-05 10:31:09) CameraRealSense2.cpp:867::rtabmap::CameraRealSense2::init() BGRA8 1280 720 6 0 Color type=2
[ERROR] (2021-01-05 10:31:09) CameraRealSense2.cpp:867::rtabmap::CameraRealSense2::init() RGBA8 1280 720 6 0 Color type=2
[ERROR] (2021-01-05 10:31:09) CameraRealSense2.cpp:867::rtabmap::CameraRealSense2::init() BGR8 1280 720 6 0 Color type=2
[... cut ...]
[ WARN] (2021-01-05 10:31:17) CameraRealSense2.cpp:111::rtabmap::CameraRealSense2::~CameraRealSense2() stop_streaming() failed. UVC device is not streaming!                      
[ WARN] (2021-01-05 10:31:17) CameraRealSense2.cpp:111::rtabmap::CameraRealSense2::~CameraRealSense2() tried to stop sensor without starting it                                   
[ WARN] (2021-01-05 10:31:17) CameraRealSense2.cpp:111::rtabmap::CameraRealSense2::~CameraRealSense2() stop_streaming() failed. Hid device is not streaming!
```

matlabbe, 2021-01-08:
The latest L515 firmware (1.5.3) doesn't support RGB 640x480 anymore. The following commit fixes the problem by supporting 1280x720x30 RGB + 640x480x30 Depth: https://github.com/introlab/rtabmap/commit/b70d13cb9e1ef0aeed7ad0140056ad662214e4b2

eltonhcchan, 2021-01-16:
@MoonUniverse hi, I've encountered the same problem as you posted. Did you fix it? Just adjust to 640x480 by downgrading the firmware?

matlabbe, 2021-01-18:
The commit above fixed it for latest realsense firmware. I just released [0.20.8](https://github.com/introlab/rtabmap/releases/tag/0.20.8) with those changes.