introlab/rtabmap #575: Realsense 2 on Android Device

TheBricktop, 2020-07-21:
This was talked about before but since the librealsense2 works out of the box on android devices without root, could it be possible to build a version of rtabmap mobile with realsense2 libraries build in?
Its just to evaluate the possibility to use rtab map as a backend for mobile XR devices.

matlabbe, 2020-07-31:
Definitely possible, just tried it:
[image]

I didn't take a look deeply in the code, but at least doing the whole pipeline with RTAB-Map is possible. Using ARCore for pose estimation could be nice along the realsense images.

TheBricktop, 2020-08-01:
Oh, this is looking good, i've just forked rtabmap to see if i could build the apk with rtab map enabled or maybe even do an aar lib to use rtab map slam capabilities with realsense on android (arcore with multiroom relocation and map extraction would be marvellous as well)