introlab/rtabmap #578: Drift on no movement when odometry lost

shirbarzel, 2020-07-29:
Hi,
I'm using rtabmap and robot_localization with my D435i for VIO, I'm getting a drift as seen on the gif when the odometry get lost without any movement of the camera. Any idea why?

[image]

blakely3dnm, 2020-07-29:
I'm not an expert by any means but I have played with rtabmap off and on over the years. 

With VIO odometry will generally be lost if you don't have visual features that can be registered to valid depth measurements. It looks fine until you put your hand in front of the sensor and begin trying to solve pose with junk data. I'm guessing that much acceleration is within the noise range if your IMU and it is trying to fit to bad visual features on your hand.

matlabbe, 2020-08-25:
Is the accelerometer from the IMU used to estimate the position? If he filter doesn't receive anymore sensor update, it may use its internal representation to update the pose based on time and probably a constant velocity model. It will indeed take the latest velocity received and estimate its position while the sensor is blocked. Unless you have a super accurate IMU (like xsens) for pose tracking, it will be very difficult to know how much you moved when looking to a white wall. You have two choices, either the filter stop updating when it receives null odometry, or estimate as constant velocity (maybe use accelerometer to adjust a little) based on last one received. 

It is why in [kinect example](https://github.com/introlab/rtabmap/wiki/Kinect-mapping#lostodometry), when we cannot estimate odometry anymore, we just stop, forcing the user to come back where it worked and try to scan different the textureless area (like scanning from farther, or scanning to have always some features to track).