introlab/rtabmap #579: make fail

EasternH, 2020-07-30:
When I using make in ./build folder, I meet the error.

`[ 80%] Linking CXX executable ../../../bin/rtabmap
../../../bin/librtabmap_core.so.0.20.2: undefined reference to `g2o::csparse_extension::cs_chol_workspace(cs_di_sparse const*, cs_di_symbolic const*, int*, double*)'
../../../bin/librtabmap_core.so.0.20.2: undefined reference to `g2o::csparse_extension::writeCs2Octave(char const*, cs_di_sparse const*, bool)'
../../../bin/librtabmap_core.so.0.20.2: undefined reference to `g2o::csparse_extension::cs_cholsolsymb(cs_di_sparse const*, double*, cs_di_symbolic const*, double*, int*)'
collect2: error: ld returned 1 exit status
app/src/CMakeFiles/rtabmap.dir/build.make:464: recipe for target '../bin/rtabmap' failed
make[2]: *** [../bin/rtabmap] Error 1
CMakeFiles/Makefile2:416: recipe for target 'app/src/CMakeFiles/rtabmap.dir/all' failed
make[1]: *** [app/src/CMakeFiles/rtabmap.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
`

How can I solve it

EasternH, 2020-07-30:
I have install the g2o, but it's strange that it cannot use it

matlabbe, 2020-08-03:
Is this the binary g2o version (ros-melodic-libg2o) or the one from source?

matlabbe, 2020-08-03:
Duplicate of https://github.com/introlab/rtabmap/issues/56 ?

eliabntt, 2020-08-04:
Try delete the build folder inside rtabmap. I had problems too today while upgrading, apparently by g2o, after deleting the build folder everything resolved.