introlab/rtabmap #580: [Feature] Camera pose from link and displacement filter

eliabntt, 2020-08-03:
Hi, 
in my system I have a camera that is free to move w.r.t. the robot base. I've done various tests and it seems that loop closures are being detected correctly taking into account camera pose between links. I've tried also to create a portion of the map, pause rtabmap, rotate the camera, and started rotating the robot and still the loop closure doesn't ruin my odometry (as expected).

Still I'd want to ask you if there is a way to get camera pose from link information. I see [here](https://github.com/introlab/rtabmap/blob/master/corelib/src/Rtabmap.cpp#L1233) that you filter out poses based robot movement. Since my camera is moving, I'd like to add a couple of lines to consider also minimum angular displacement of the camera pose variation.(my camera is publishing a standalone "odometry" topic) I could also add a custom subscriber and elaborate the information but maybe you have some smarter/direct way.

matlabbe, 2020-08-03:
The transformation between base frame and the camera frame is included in the CameraModel's `localTransform` set in the node:
```cpp
// Get base to camera of the current node
signature->sensorData().cameraModels()[0].localTransform()
// Get base to camera of the previous node
s->sensorData().cameraModels()[0].localTransform()
```

eliabntt, 2020-08-04:
fine it seems working :) thanks!

closed 2020-08-04 by eliabntt