introlab/rtabmap #582: give a label name to a map sessions or when a new map_trigger function is called

NicoNorena, 2020-08-08:
@matlabbe question: 
is there support to give a name contained within the database of the mapping session? or whenever a new map_trigger function is called ?

NicoNorena, 2020-08-25:
okay thanks, so then how is the map frame called for the second session? by the fault the first map session is called "map", but  the preceding frame name of the map sessions is supposed to have a unit name.

when the function "int Rtabmap::triggerNewMap()" https://github.com/introlab/rtabmap/blob/775c9318d333aa09a00ee1ae06c6664eb1b83265/corelib/src/Rtabmap.cpp#L745
is called, does it start a new session with the next possible enumeration? 

so the "bool Rtabmap::labelLocation(int id, const std::string & label)"
https://github.com/introlab/rtabmap/blob/775c9318d333aa09a00ee1ae06c6664eb1b83265/corelib/src/Rtabmap.cpp#L792
is labeling the pose in one node in the graph (given id); this is not labeling the map session

matlabbe, 2020-08-25:
yes, `rtabmap::labelLocation()` is labelling a node in the graph, not a session.

`Rtabmap::triggerNewMap()` increment the map id. So if we have already 1 session (mapId=0), the next mapId will be 1.

The transform /map -> /odom is the correction of the current odometry pose to transform it in map frame. The map frame is related to the first session (when `RGBD/OptimizeFromGraphEnd=false`, default) in the merged sessions up to last node added to graph. For example, if you have disjoint session 0 and session 1 (no loop closures between them), the map to odom correction will be done against reference frame of the session 1. If there is a loop closure between session 0 and session 1, the map to odom correction will be done against reference frame of the session 0.

NicoNorena, 2020-08-25:
okay, I get it now, thanks.
I am going to close the ticket

closed 2020-08-25 by NicoNorena