introlab/rtabmap #584: NVM Format

Zumbalamambo, 2020-08-14:
How do I export pointcloud in NVM format?

Zumbalamambo, 2020-08-19:
and how do I open bundler.out with python?

matlabbe, 2020-08-24:
I've never used the NVM format: http://ccwu.me/vsfm/doc.html#nvm
I am not sure what they mean by having "multiple 3D models" in the same file.

For the bundler output, see [Bundler](http://www.cs.cornell.edu/~snavely/bundler/) page and this [example](https://github.com/introlab/rtabmap/wiki/Export-Raster-Layers-to-MeshLab) of export with rtabmap into MeshLab (also supporting bundler format). Can't you open it in a simple text app? It seems not a binary format, so you may be able to parse it in python. The bundler format is described [here](http://www.cs.cornell.edu/~snavely/bundler/bundler-v0.4-manual.html#S6).

Zumbalamambo, 2020-08-31:
@matlabbe thank you!... I have been trying to understand the bundler format but it's so hard since there are no meta tags that describe how it is structured so I can parse it. Is there any python example code that I can go through?

Also will it give me the points3d of the rgb images similar to that of the aachen dataset?

Zumbalamambo, 2020-08-31:
As an example, Lets assume the following,

```
# Bundle file v0.3
728 152190                                   -->num_cameras(corresponding to the image name), num_points
1735.36 0 0                                  --> <camera1>
0.027868 -0.999189 -0.0290513                --> [the focal length, followed by two radial distortion coeffs]
0.469858 -0.0125585 0.882653                 -->[a 3x3 matrix representing the camera rotation]
-0.882302 -0.0382478 0.469127
0.0747407 -0.0339194 -0.0254617
1735.36 0 0                                  --> </camera1>

```

But the documentation says     `<t>  [a 3-vector describing the camera translation]` is also present between <camera></camera> but I dont see it in the bundler.out file :(

closed 2020-09-01 by Zumbalamambo