introlab/rtabmap #586: RTABMAP + GTSAM results in segmentation fault.

thohemp, 2020-08-18:
Using rtabmap 0.20.3 and gtsam 4.0.0 ends in segmentation fault, when starting:

```
Thread 1 "rtabmap" received signal SIGSEGV, Segmentation fault.
__GI___libc_free (mem=0x21) at malloc.c:3103
3103	malloc.c: No such file or directory.
(gdb) b
Breakpoint 1 at 0x7ffff4dae9fd: file malloc.c, line 3103.
(gdb) bt
#0  0x00007ffff4dae9fd in __GI___libc_free (mem=0x21) at malloc.c:3103
#1  0x00007fffe4592ce4 in gtsam::noiseModel::Diagonal::Sigmas(Eigen::Matrix<double, -1, 1, 0, -1, 1> const&, bool) () at /usr/local/lib/libgtsam.so.4
#2  0x00007fffe441b364 in _GLOBAL__sub_I_lago.cpp ()
    at /usr/local/lib/libgtsam.so.4
#3  0x00007ffff7de5783 in call_init (env=0x7fffffffdbe8, argv=0x7fffffffdbd8, argc=1, l=<optimized out>) at dl-init.c:72
#4  0x00007ffff7de5783 in _dl_init (main_map=0x7ffff7ffe170, argc=1, argv=0x7fffffffdbd8, env=0x7fffffffdbe8) at dl-init.c:119
#5  0x00007ffff7dd60ca in _dl_start_user () at /lib64/ld-linux-x86-64.so.2
#6  0x0000000000000001 in  ()
#7  0x00007fffffffdf96 in  ()
#8  0x0000000000000000 in  ()
```

tkircher, 2020-08-20:
Probably an Eigen issue. Try configuring GTSAM with -DGTSAM_USE_SYSTEM_EIGEN=ON

thohemp, 2020-08-20:
Worked, thanks.

closed 2020-08-20 by thohemp