introlab/rtabmap #587: Issues with make

marcosab1991, 2020-08-19:
I'm trying to build rtabmap and already installed all the dependencies but make doesnt run:
[  4%] Built target rtabmap_utilite
[  5%] Built target res_tool
[  5%] Linking CXX shared library ../../../bin/librtabmap_core.so
/usr/bin/ld: não foi possível localizar -lBoost::serialization
/usr/bin/ld: não foi possível localizar -lBoost::thread
/usr/bin/ld: não foi possível localizar -lBoost::timer
/usr/bin/ld: não foi possível localizar -lBoost::chrono
collect2: error: ld returned 1 exit status
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/build.make:1851: ../bin/librtabmap_core.so.0.20.3] Error 1
make[1]: *** [CMakeFiles/Makefile2:894: corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2

thohemp, 2020-08-19:
Is boost installed?

marcosab1991, 2020-08-19:
> Is boost installed?

Yes

thohemp, 2020-08-20:
But it says it cannot find the boost. Can you find the missing libs in /usr/bin?
Otherwise try to reinstall boost, either from source or with
`sudo apt-get install libboost-all-dev`

matlabbe, 2020-08-24:
This error has been fixed in https://github.com/introlab/rtabmap/commit/8d63e9eae1c5c1b5511c29e7e9a029a3e17261cc

matlabbe, 2020-08-24:
Were you compiling against libpointmatcher too?