introlab/rtabmap #590: Information matrix/goodness of a node

eliabntt, 2020-08-24:
Hi, from what I understand information matrices that you save/publish are the covariance inverse.

Are those inf matrix updated when a loop closure happens or are them fixed?

Is there any other measure that can tell how likely it is to have a loop closure in any given pose?
Or, alternatively, a way to assess if a node is a "good" node or a "bad" node?
With "good" node I mean a node that has a lot of information and a high probability of triggering a loop closure.

matlabbe, 2020-08-25:
The information matrices in the links of the graph are fixed. 

RTAB-Map uses vision as loop closure detection approach, so if there are more visual words in an image, it means it would be more likely to trigger a loop closure. It is not always the case though, in highly visually repetitive environment. The loop closure detector tends to put more weight on discriminative features than common visual features. In the main window or `rtabmapviz`, you can open the View->Statistics panel, then under Loop subpanel, you can find many information about the current loop closure detection state (in particular the loop hypothesis value or the likelihood value). Those information are also available through `/rtabmap/info` topic on ROS.

eliabntt, 2020-09-09:
So when you do the k-th loop closure the final uncertainty is NOT based on the optimization of the covariance obtained by the (k-1)-th loop closure's optimization. Right? So if node 1 because of (k-1)-th loop closure should have a lower uncertainty because of g2o/gtsam, you don't have that reflected in the future but when the k-th loop closure happens the node 1 will contribute as if the previous closures didn't happened?

matlabbe, 2020-09-10:
After an optimization, the next optimization will use the previous optimized poses as guess, using the original (fixed) information matrices. It will converge faster and will be consistent with the previous optimization. If we ignore the past optimizations and take always the raw odometry poses for new optimization, the graph may not converge always to same solution (on a different local minimum). By using the last optimization as guess, the current optimization will stay near the last local minimum. For the uncertainties, gtsam/g2o will recompute the pose uncertainties based on the current loop closures.

eliabntt, 2020-09-21:
Maybe I wasn't clear enough. I'm not saying you are not using the raw odometry poses but maybe that you could update and use the updated information matrix. So at every loop closure you won't only update the poses but also the covariances/information matrixes

matlabbe, 2020-09-27:
Is g2o ot gtsam giving back the updated information matrices for each pose? Does it improve significantly the convergence time?

eliabntt, 2020-09-28:
You can get this info from both as we've discussed [here](https://github.com/introlab/rtabmap/issues/526#issuecomment-638035062). GTSAM gives the result out of the box, instead for g2o you need to do the computation for each link. I didn't test the convergence time since to do that I'd need to edit the covariances within the rtabmap db.

matlabbe, 2020-09-28:
This could be tested by exporting the graph in `*.g2o` format and use g2o tools directly (or rewrite a simple tool to load the g2o graph and do the optimization).

closed 2021-03-24 by eliabntt