introlab/rtabmap #592: make error

aaravrav142, 2020-08-31:


matlabbe, 2020-08-31:
cmake is building against opencv binaries from kinetic:
```
Found OpenCV: /opt/ros/kinetic/include/opencv-3.3.1-dev;/opt/ros/kinetic/include/opencv-3.3.1-dev/opencv
```
You are also building with VINS-Fusion:
```
--   With VINS-Fusion          = YES (License: GPLv3)
```
If VINS-Fusion has been built with your other OpenCV version with CUDA, then there is a problem of mismatch opencv version between rtabmap and its dependencies. If you do `cmake -DWITH_VINS=OFF`, is it able to build correctly with opencv binaries?

Note that even if you force rtabmap to build against your opencv version, make sure to rebuild also your cv_bridge package because you will have the same problems with rtabmap and rtabmap_ros not linking on same opencv version. 

When you have a custom opencv setup like this, it is better to just uninstall all your opencv system binaries, and rebuild all packages depending on opencv so that everyone are linking on same opencv version.

aaravrav142, 2020-09-01:
Ok following your advice, I rebuilt all libraries again to properly link with the rtabmap and rtabmap_ros. 
It built successfully this time. 

Thank you for the help.

closed 2020-09-01 by aaravrav142