introlab/rtabmap #593: Opening the depth image

Zumbalamambo, 2020-09-01:
What format the depth image is at?

I tried to parse the depth image to access the Z values with respect to the XY position using the following code,

```
import cv2
depth_img = cv2.imread("depth/2.png",cv2.IMREAD_ANYDEPTH)

for i in depth_img[0]:
    for j in depth_img[1]:
        print(depth_img[i][j])
```

Unfortunately, all the values are zero :(

matlabbe, 2020-09-01:
They are unsiged 16bits value in mm.

Zumbalamambo, 2020-09-01:
@matlabbe thank you! :)

The value always ranges between 1-4 in every images. Am I parsing it properly?

import numpy as np
from PIL import Image

im = Image.fromarray(np.array(Image.open("depth/1.png")).astype("uint16"))
img_numpy = np.array(im).astype(np.uint16)
h, w = img_numpy.shape

for i in range(h):
    for j in range(w):
        if(img_numpy[i][j] > 0):
            print("XY: ", (i,j)," Z: ", img_numpy[i][j])

Zumbalamambo, 2020-09-01:
@matlabbe thank you! :)

The value always ranges between 1-4 in every images. Am I parsing it properly? Is it possible to get the absolute XY position?

```
import numpy as np
from PIL import Image

im = np.array(Image.open("depth/1.png")).astype("uint16")
h, w = im.shape

for i in range(h):
    for j in range(w):
        if(im[i][j] > 0):
            print("XY: ", (i,j)," Z: ", im[i][j])
```

matlabbe, 2020-09-01:
I tried your code in your first post and it shows values over 1000 (over 1 meter) in my case. Can you share the database containing depth/2.png ?

Zumbalamambo, 2020-09-02:
@matlabbe I have fixed it by pulling the latest docker image. It seems to export with an older version of RTABMAP has created this issue..

May I also ask how I can get the absolute position of each of these `z values`? I'm using the first code

matlabbe, 2020-09-02:
With Google Tango phones, we accept lower resolution depth image corresponding to higher resolution rgb images. The depth image is assumed to be registered to rgb image, it means that it has exactly the same calibration of the rgb camera minor the scale of fx,fy,cx and cy.

To get the 3D position of each pixel in camera frame, you need the intrinsic camera matrix K of the camera. When doing he export, you may see a yaml file having the calibration. The formula is here:
https://github.com/introlab/rtabmap/blob/23d835354087919b6b0558d523cc0ba81b18dd2a/corelib/src/util3d.cpp#L232-L235

Zumbalamambo, 2020-09-02:
@matlabbe thank you! I have implemented the same in python as follows,

```
import cv2

# camera matrix
mtx = [8.6767822265625000e+02, 0., 4.8268579101562500e+02, 0.,
       8.6760119628906250e+02, 2.7117459106445312e+02, 0., 0., 1.]

# parse calib matrix
fx = mtx[0]
cx = mtx[2]

fy = mtx[4]
cy = mtx[5]

depth_img = cv2.imread("RTAB-Map/bundler/depth/1326.png", cv2.IMREAD_ANYDEPTH)
depth_row, depth_column = depth_img.shape

# to store 3d points
pts3d = []

for i in range(depth_row):
    for j in range(depth_column):

        # retrieve depth value
        depth = depth_img[i][j]
        x, y = i, j

        if depth > 0.0:
            x3D = (x - cx) * depth / fx
            y3D = (y - cy) * depth / fy
            z3D = depth

            # populate 3d points array
            pts3d.append([x3D, y3D, z3D])
        else:
            pts3d.append([-1, -1, -1])
```

Could you please confirm if I'm doing it right? I'm bit concerned about the shape of the `pts3d`. I will be using `pts3d` to perform 2d-3D matching. The shape of RGB image is `960x540` hence I'm afraid if it will give incorrect shape error :(

matlabbe, 2020-09-10:
Well, not sure what approach you will be using for 2D-3D matching, but the 3D projection seems okay.

closed 2020-10-23 by Zumbalamambo