introlab/rtabmap #594: Export Sequence and Odometry

ffrivera0, 2020-09-02:
Hi,

I was wondering if it is possible to export the video and its corresponding odometry when using realsense D435+T265 dual-mode, rather than the register frames and poses

matlabbe, 2020-09-03:
A mjpeg video could be created from the directory containing the RGB images. For the poses, what kind of "odometry" format do you want. RTAB-Map can export poses in multiple formats (like *.g2o), see File->Export poses. `rtabmap-report --poses` can also be used to export in [rgbd-dataset format](https://vision.in.tum.de/data/datasets/rgbd-dataset/file_formats).

ffrivera0, 2020-09-07:
Hi,
I mean a video record all the frames from the start point (eg. 30Hz or 60 Hz) and its corresponding odometry (also 30Hz or 60Hz). In rtabmap, I only find the output images and poses registered at a low frequency (2-3 Hz).

By the way, May I know exactly the frequency of new IDs (images) recorded?

matlabbe, 2020-09-10:
To record all odometry data in database, you should set `Rtabmap/CreateIntermediateNodes=true` and `Mem/IntermediateNodeDataKept=true`.