introlab/rtabmap #595: [Feature] Require nearby nodes list

MotorTrick, 2020-09-08:
Hi, I was looking into your code to see if there is already a service to require nearby nodes given a position on the map. As far as I understand it's what you are doing [here](https://github.com/introlab/rtabmap/blob/23d835354087919b6b0558d523cc0ba81b18dd2a/corelib/src/Rtabmap.cpp#L2248). Is there any service that does the same? Otherwise it should be fairly easy to get that right? Just by reproducing the code in a new service. I'm also interested in the camera orientation on those nearby poses..

matlabbe, 2020-09-28:
Hi,
I added a service in the commit above to do so:
```bash
# get nodes around latest node:
rosservice call /rtabmap/get_nodes_in_radius "{node_id: 0, x: 0.0, y: 0.0, z: 0.0, radius: 2.0}"
# get nodes around node 16:
rosservice call /rtabmap/get_nodes_in_radius "{node_id: 16, x: 0.0, y: 0.0, z: 0.0, radius: 2.0}"
# get nodes around landmark 2:
rosservice call /rtabmap/get_nodes_in_radius "{node_id: -2, x: 0.0, y: 0.0, z: 0.0, radius: 2.0}"
# get nodes around position x=5:
rosservice call /rtabmap/get_nodes_in_radius "{node_id: 0, x: 5.0, y: 0.0, z: 0.0, radius: 2.0}"
```

This returns the pose of base frame. If base frame is different than camera frame, you will have to multiply the local transform in the nodes. To get features and local transforms for a list of nodes (e.g., for nodes 1,2 and 3):
```bash
rosservice call /rtabmap/get_node_data "ids: [1,2,3]"
```

cheers,
Mathieu

eliabntt, 2020-09-29:
Nice, I've done something different though
In this way I've all the node data+features directly. It can be better adapted to accept flags for images/sensor data clearly but it's quite fast [especially by using the memory] and gives me all the info I need ;)
Feel free to use this
```
 bool CoreWrapper::getNearbyNodeDataCallback(rtabmap_ros::GetNearbyNodeData::Request &req,
                                                rtabmap_ros::GetNearbyNodeData::Response &res) {
        Transform pose = rtabmap_ros::transformFromPoseMsg(req.pose.pose, true);

        std::map<int, Transform> poses;
        std::multimap<int, rtabmap::Link> constraints;
        typedef std::multimap<int, rtabmap::Link>::iterator iter;

        rtabmap_.getGraph(poses,constraints, false, false, 0, !true, !true, !true, !true);

        std::map<int, Transform> nearestPoses;
        const rtabmap::Memory *mem = rtabmap_.getMemory();

        std::vector<int> nodes = graph::findNearestNodes(poses, pose, req.k > 0 ? req.k : 100);

        for (auto id : nodes) {
            int node_value = -1;

            Signature s = *mem->getSignature(id);

            if (s.id() > 0) {
                if (std::sqrt(std::pow(s.getPose().x() - pose.x(), 2) + std::pow(s.getPose().y() - pose.y(), 2)) <= req.rad) {

                    for (auto l:mem->getLinks(s.id())) {
                        rtabmap::Link::Type type = l.second.type();
                        if (type == rtabmap::Link::kGlobalClosure) {
                            node_value = (int) type;
                            break;
                        }
                        if (type == rtabmap::Link::kLocalSpaceClosure || type == rtabmap::Link::kNeighborMerged) {
                            node_value = type;
                        }
                    }

                    NodeData msg;
                    // data clearing - stuff I don't need - less bandwith
                    s.sensorData().clearCompressedData();
                    s.sensorData().clearRawData();
                    s.sensorData().clearOccupancyGridRaw();
                    s.sensorData().setOccupancyGrid(cv::Mat(),cv::Mat(),cv::Mat(), s.sensorData().gridCellSize(), s.sensorData().gridViewPoint());
                    std::multimap<int, cv::Mat> tmp;
                    s.setWordsDescriptors(tmp);
                    rtabmap_ros::nodeDataToROS(s, msg);
                    res.data.push_back(msg);
                    res.value.push_back(node_value);
                }
            }
        }
        return !res.data.empty();
    }
```

MotorTrick, 2020-09-30:
Thank you both!

closed 2020-09-30 by MotorTrick