introlab/rtabmap #597: Orientations of three axes when using realsense cameras

popDuorz, 2020-09-09:
Hi,

May I know the world coordinate of the exported pose (.g2o format, robot), eg what is the x axis, y axis, and z axis point to. For example in computer vision system, the X axis points to the right, the Y axis to the bottom, and the Z axis to the front as seen from the image. How about in rtabmap. Thnaks

matlabbe, 2020-09-09:
If you want to export the pose in camera frame (x->right, y->down, z->forward), choose camera frame in the dropdown box when exporting. The "robot" frame means the base frame of the robot, and it follows ROS REP105: https://www.ros.org/reps/rep-0105.html (x->forward, y->left, z->up).

popDuorz, 2020-09-10:
Thanks. Very helpful information :)

closed 2020-09-10 by popDuorz