introlab/rtabmap #598: SIGILL caused by rtabmap-export

moshe-rabaev, 2020-09-15:
When using the rtabmap:bionic image with multiband texturing.

This error happens prior to exporting a multiband obj file.

```Thread 1 "rtabmap-export" received signal SIGILL, Illegal instruction.
0x00007fb744b01579 in aliceVision::mvsUtils::MultiViewParams::loadMatricesFromSfM(int) () from /usr/local/lib/libaliceVision_mvsUtils.so.2
```

I am running it with docker, my host machine has the following specs:

- OS: ubuntu 20.04.1 LTS 
- Graphics card: GeForce GT 630/PCIe/SSE2
- Processor: Intel® Xeon(R) CPU E5-2620 0 @ 2.00GHz × 24

matlabbe, 2020-09-15:
On which database are you doing this? Are you using this command:
```bash
# Export texture mesh (bionic image to have AliceVision support, not needed if --multiband is not used)
$ docker run -it --rm -v ~/Downloads/MultiSessionTango:/MultiSessionTango \
   --workdir /MultiSessionTango --user $(id -u):$(id -g) \
   introlab3it/rtabmap:bionic \
   rtabmap-export \
      --texture \
      --texture_size 4096 \
      --texture_range 3 \
      --poisson_depth 9 \
      --max_polygons 500000 \
      --multiband \
      --color_radius 0 \
      --no_clean \
      merged.db
```
from this [tutorial](https://github.com/introlab/rtabmap/wiki/Multi-Session-Mapping-with-RTAB-Map-Tango)?

moshe-rabaev, 2020-09-15:
Hi,

I haven't tried the database from the tutorial.

I'm using the exact command, except `--user` and `--max_polygons 500000`

I will try it with the database in the link you sent.