introlab/rtabmap #601: Reg/Strategy

MotorTrick, 2020-09-18:
Hi, may I ask what's the difference between the three Reg/Strategy? Is ICP-Reg only lidar or is always based on visual features?

I'm currently using proximity detection + non robust loop closures and I use those to refine my odometry in an EKF (I use the loop closure as a "GPS" signal). I think that this cause the system to reject many valid loop closures because the covariance goes too low. In fact, by removing proximity detection it seems to get better (in the sense that it detects/accept more loop closures) but I'd like to have both.

Moreover, I've seen that you suggest to have a "static" covariance. Why is that? Shouldn't the covariance be an always-growing thing? (not the twist covariance but just the pose covariance)

matlabbe, 2020-09-27:
Hi,

- Reg/Strategy=0: all transforms are computed with only visual features
- Reg/Strategy=1: loop closure is computed with visual->icp (visual gives the rough guess to ICP), but proximity detection and neighbor refining are ICP only based on guess from odometry.
- Reg/Strategy=2: loop closure, proximity detection and neighbot refining are computed with visual->icp (visual gives the rough guess to ICP)

You can increase RGBD/OptimizeMaxError (default 3) to accept more loop closures if the covariance is small. The your odometry covariance is too small, rtabmap may not able to close loops as it will required to deform the graph more than the covariance received. You can also set RGBD/OptimizeMaxError to 0 to accept all loop closures. 

The covariance used in links of rtabmap is the twist covariance, not the pose covariance. The pose covariance would increase out of bound, but not the twist.

MotorTrick, 2020-09-30:
What do you suggest between those if one has camera+2d lrf? Strat1 or 2?

About covariance that is exactly the error I'm getting. I've already optimizeMaxError at 3. How much do you suggest to increase that? Is this better than using robust optimization?
I'm using a standard ekf to fuse imu+odometry+2nd IMU+icp odometry+as I was saying loop closures (local and global) as "GPS" even if this makes the robot "jump" a bit sometimes

matlabbe, 2020-10-08:
If the covariance is poorly chosen, robust optimization won't be better. For `Optimizer/Robust` or `RGBD/OptimizeMaxError`, if the covariance is not good, it would be preferred to disable them. Also for graph optimization with GTSAM or g2o, covariance is important. For example, if the covriance is unknown, it is possible to set Identity and use `Optimizer/Strategy`=0 (TORO), which is less sensible to covariance.

It is possible to override the covariance to a fixed one with those parameters for `rtabmap` node:
```
odom_tf_linear_variance:=0.01
odom_tf_angular_variance:=0.01
```

For localization, there will be always jumps on loop closures, because we are relocating at another place. See also this [post](http://official-rtab-map-forum.67519.x6.nabble.com/Filtering-rtabmap-localization-jumps-with-robot-localization-in-2D-td5931.html).

MotorTrick, 2020-10-12:
Actually Robust seems to work better, and also giving the odom as a guess for the TFs.
As of now I've around 0.003 for both linear and angular but I think I can change it by tuning the EKF.  What would be in your opinion a good value? 

What do you suggest between those if one has camera+2d lrf? Strat1 or 2?

HendrikClijsters123, 2020-11-13:
Hey @MotorTrick  

I am using a setup comparable to yours. Did you find out if one of the strategies is more beneficial than the others?

Cheers
Hendrik

MotorTrick, 2020-11-18:
Sincerely, no. But by being the visual features so noisy in a real world scenario I'm testing differences between 1 and 2.