introlab/rtabmap #605: how to export .db using rtabmap-databaseviewer, and then open with rtabmap

HuangTY96, 2020-09-28:
I use rtabmap-databaseviewer to open a 1.db file, and after I change some parameter, I export this 1.db file to another 2.db file. but when I open this 2.db using rtabmap, there is all blank, seems like there is no cloud. and could you please tell me how to solve it.

my system is ubuntu18.04+rtabmap 0.20.3+pcl 1.11.0

matlabbe, 2020-09-28:
The Export database option in DatabaseViewer exports only selected data to a new database, the graph is then lost. To convert a database with other parameters, use `rtabmap-reprocess` instead with your modified parameters (e.g., using `Vis/MinInliers` set to 15 and `Grid/CellSize` to 10cm):
```
$ rtabmap-reprocess --Vis/MinInliers 15 --Grid/CellSize 0.1 input.db output.db
```

HuangTY96, 2020-10-07:
> The Export database option in DatabaseViewer exports only selected data to a new database, the graph is then lost. To convert a database with other parameters, use `rtabmap-reprocess` instead with [...]

thanks a lot, but which parameter can be used to change the depth of point cloud?

matlabbe, 2020-10-07:
You can use `rtabmap --params | grep Grid/` to show Grid parameters. To one you want is `Grid/RangeMax`.