introlab/rtabmap #606: Building Rtabmap plugin for Android?

kaatrasa, 2020-09-28:
Hey,

I created Unity plugin for Rtabmap that I have been using with Windows.  Now I would like to use it with Android, but I am not sure how to build it with android version of Rtabmap. I am using the android-deps (https://github.com/introlab/rtabmap/tree/master/docker/bionic/android) docker image for the cross-compiled dependencies.

What I am currently doing:
1) Start the container & run deps.bash
2) Then I ran rtabmap.bash with arguments "/opt/android 24"(https://github.com/introlab/rtabmap/blob/master/docker/bionic/android/rtabmap_api26/rtabmap.bash), (I set DBUILD_SHARED_LIBS=ON and removed "make clean" from end (so that I can copy the librtabmap_core.so and librtabmap_utilite.so files to my Unity project).
3) Then for simple example I tried to build the loop-closure example as .so file (minor modifications to cmakelists.txt):

```
cmake_minimum_required(VERSION 2.8)
PROJECT( MyProject )

SET(CMAKE_MODULE_PATH "${PROJECT_SOURCE_DIR}")
FIND_PACKAGE(RTABMap COMPONENTS Core Utilite REQUIRED) 
FIND_PACKAGE(OpenCV REQUIRED)

SET(INCLUDE_DIRS
   ${RTABMap_INCLUDE_DIRS}
   ${OpenCV_INCLUDE_DIRS}
)

SET(LIBRARIES
   ${RTABMap_LIBRARIES}
   ${OpenCV_LIBRARIES} 
)

INCLUDE_DIRECTORIES(${INCLUDE_DIRS})

include (GenerateExportHeader)
add_library(RtabmapPlugin SHARED main.cpp)
TARGET_LINK_LIBRARIES(RtabmapPlugin ${LIBRARIES})
```
I run this script to try to build the project with arguments "/opt/android 24"
```
#!/bin/bash

if [ $# -ne 2 ]; then
    echo "rtabmap.bash android_install_prefix api_level (23 for tango, 24 for arengine)   # Example: build.bash /opt/android 24"
    exit 1
fi

prefix=$1
api=$2
pwd=$(pwd)

$pwd/cmake-3.17.0-Linux-x86_64/bin/cmake -DCMAKE_TOOLCHAIN_FILE=$ANDROID_NDK/build/cmake/android.toolchain.cmake -DANDROID_ABI=arm64-v8a -DANDROID_NDK=$ANDROID_NDK -DANDROID_NATIVE_API_LEVEL=$api -DBUILD_SHARED_LIBS=OFF -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=$prefix/arm64-v8a -DCMAKE_FIND_ROOT_PATH="$prefix/arm64-v8a/bin;$prefix/arm64-v8a;$prefix/arm64-v8a/share" -DBUILD_EXAMPLES=OFF -DBUILD_TOOLS=OFF -DRTABMap_DIR=$pwd/rtabmap-tango/build/arm64-v8a ./dockerbuilds
make
```
However, when I run this I get:
```
root@0e079fdb75fe:~# ./test.bash /opt/android 24
-- Configuring done
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
RTABMap_CORE_RELEASE
    linked by target "RtabmapPlugin" in directory /root/dockerbuilds
RTABMap_UTILITE_RELEASE
    linked by target "RtabmapPlugin" in directory /root/dockerbuilds

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
[ 50%] Building CXX object CMakeFiles/RtabmapPlugin.dir/main.cpp.o
In file included from /root/dockerbuilds/main.cpp:1:
In file included from /root/rtabmap-tango/corelib/include/rtabmap/core/Rtabmap.h:34:
In file included from /root/rtabmap-tango/corelib/include/rtabmap/core/SensorData.h:32:
/root/rtabmap-tango/corelib/include/rtabmap/core/Transform.h:35:10: fatal error: 'Eigen/Core' file not found
#include <Eigen/Core>
         ^~~~~~~~~~~~
1 error generated.
CMakeFiles/RtabmapPlugin.dir/build.make:79: recipe for target 'CMakeFiles/RtabmapPlugin.dir/main.cpp.o' failed
make[2]: *** [CMakeFiles/RtabmapPlugin.dir/main.cpp.o] Error 1
CMakeFiles/Makefile2:92: recipe for target 'CMakeFiles/RtabmapPlugin.dir/all' failed
make[1]: *** [CMakeFiles/RtabmapPlugin.dir/all] Error 2
Makefile:100: recipe for target 'all' failed
make: *** [all] Error 2
```
I am  new to CMake so I am not sure what is causing the error, why are rtabmap_core and utilite NOTFOUND and how is eigen not found eventough I was able to build your rtabmap application just fine.

Thank you in advance!

matlabbe, 2020-09-28:
Hi,

Based on the external build c++ example, I think that only by adding the PCL dependency it should fix your problem:
https://github.com/introlab/rtabmap/blob/bedc771fa4acde22c1497811a88de50c42a248e8/examples/RGBDMapping/CMakeLists.txt#L25
then add:
https://github.com/introlab/rtabmap/blob/bedc771fa4acde22c1497811a88de50c42a248e8/examples/RGBDMapping/CMakeLists.txt#L47
and:
https://github.com/introlab/rtabmap/blob/bedc771fa4acde22c1497811a88de50c42a248e8/examples/RGBDMapping/CMakeLists.txt#L58
This will add Eigen and other required dependencies at the same time.

cheers,
Mathieu

kaatrasa, 2020-09-28:
Hey,

thanks for the quick reply! Looks like adding PCL fixed all the missing dependencies. However I still get the Rtabmap NOTFOUND error:

```
root@0e079fdb75fe:~# ./test.bash /opt/android 24
-- Configuring done
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
RTABMap_CORE_RELEASE
    linked by target "RtabmapPlugin" in directory /root/dockerbuilds
RTABMap_UTILITE_RELEASE
    linked by target "RtabmapPlugin" in directory /root/dockerbuilds

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
[ 50%] Building CXX object CMakeFiles/RtabmapPlugin.dir/main.cpp.o
In file included from /root/dockerbuilds/main.cpp:1:
In file included from /root/rtabmap-tango/corelib/include/rtabmap/core/Rtabmap.h:34:
In file included from /root/rtabmap-tango/corelib/include/rtabmap/core/SensorData.h:32:
/root/rtabmap-tango/corelib/include/rtabmap/core/Transform.h:35:10: fatal error: 'Eigen/Core' file not found
#include <Eigen/Core>
         ^~~~~~~~~~~~
1 error generated.
CMakeFiles/RtabmapPlugin.dir/build.make:79: recipe for target 'CMakeFiles/RtabmapPlugin.dir/main.cpp.o' failed
make[2]: *** [CMakeFiles/RtabmapPlugin.dir/main.cpp.o] Error 1
CMakeFiles/Makefile2:92: recipe for target 'CMakeFiles/RtabmapPlugin.dir/all' failed
make[1]: *** [CMakeFiles/RtabmapPlugin.dir/all] Error 2
Makefile:100: recipe for target 'all' failed
make: *** [all] Error 2
root@0e079fdb75fe:~# ls
CMakeCache.txt  build                      dockerbuilds   test.bash
CMakeFiles      cmake-3.17.0-Linux-x86_64  rtabmap-tango
Makefile        cmake_install.cmake        rtabmap.bash
root@0e079fdb75fe:~# cd dockerbuilds/
root@0e079fdb75fe:~/dockerbuilds# git pull
remote: Enumerating objects: 5, done.
remote: Counting objects: 100% (5/5), done.
remote: Compressing objects: 100% (1/1), done.
remote: Total 3 (delta 2), reused 3 (delta 2), pack-reused 0
Unpacking objects: 100% (3/3), done.
From https://github.com/kaatrasa/dockerbuilds
   78813d3..ec9d41b  master     -> origin/master
Updating 78813d3..ec9d41b
Fast-forward
 CMakeLists.txt | 3 +++
 1 file changed, 3 insertions(+)
root@0e079fdb75fe:~/dockerbuilds# cd ..
root@0e079fdb75fe:~# rm test.bash 
root@0e079fdb75fe:~# cp dockerbuilds/test.bash .
root@0e079fdb75fe:~# ./test.bash 
rtabmap.bash android_install_prefix api_level (23 for tango, 24 for arengine)   # Example: build.bash /opt/android 24
root@0e079fdb75fe:~# ./test.bash /opt/android 24
CMake Warning (dev) at ../cmake-3.17.0-Linux-x86_64/share/cmake-3.17/Modules/FindPackageHandleStandardArgs.cmake:272 (message):
  The package name passed to `find_package_handle_standard_args` (eigen) does
  not match the name of the calling package (PCL).  This can lead to problems
  in calling code that expects `find_package` result variables (e.g.,
  `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /opt/android/arm64-v8a/share/pcl-1.8/PCLConfig.cmake:138 (find_package_handle_standard_args)
  /opt/android/arm64-v8a/share/pcl-1.8/PCLConfig.cmake:623 (find_eigen)
  /opt/android/arm64-v8a/share/pcl-1.8/PCLConfig.cmake:825 (find_external_library)
  CMakeLists.txt:7 (FIND_PACKAGE)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found eigen: /opt/android/arm64-v8a/include/eigen3  
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
[... 340 lines cut ...]
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
RTABMap_CORE_RELEASE
    linked by target "RtabmapPlugin" in directory /root/dockerbuilds
RTABMap_UTILITE_RELEASE
    linked by target "RtabmapPlugin" in directory /root/dockerbuilds

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
Scanning dependencies of target RtabmapPlugin
[ 50%] Building CXX object CMakeFiles/RtabmapPlugin.dir/main.cpp.o
[100%] Linking CXX shared library libRtabmapPlugin.so
/opt/android-ndk-r21/toolchains/llvm/prebuilt/linux-x86_64/lib/gcc/aarch64-linux-android/4.9.x/../../../../aarch64-linux-android/bin/ld: cannot find -lRTABMap_CORE_RELEASE-NOTFOUND
/opt/android-ndk-r21/toolchains/llvm/prebuilt/linux-x86_64/lib/gcc/aarch64-linux-android/4.9.x/../../../../aarch64-linux-android/bin/ld: cannot find -lRTABMap_UTILITE_RELEASE-NOTFOUND
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
CMakeFiles/RtabmapPlugin.dir/build.make:340: recipe for target 'libRtabmapPlugin.so' failed
make[2]: *** [libRtabmapPlugin.so] Error 1
CMakeFiles/Makefile2:92: recipe for target 'CMakeFiles/RtabmapPlugin.dir/all' failed
make[1]: *** [CMakeFiles/RtabmapPlugin.dir/all] Error 2
Makefile:100: recipe for target 'all' failed
make: *** [all] Error 2
```

Could this be problem with me using this script when installing rtabmap(https://github.com/introlab/rtabmap/blob/master/docker/bionic/android/rtabmap_api26/rtabmap.bash) or?
Thank you!

matlabbe, 2020-09-28:
A `make install` [here](https://github.com/introlab/rtabmap/blob/bedc771fa4acde22c1497811a88de50c42a248e8/docker/bionic/android/rtabmap_api26/rtabmap.bash#L63) is required if you want to link to it as a local library. rtabmap will be installed in android install directory (`/opt/android`), thus cmake should be able to find it (with `-DCMAKE_INSTALL_PREFIX=/opt/android/arm64-v8a`).

kaatrasa, 2020-09-28:
I added the `make install` to rtabmap.bash, deleted rtabmap-tango folder and ran the rtabmap.bash again. I still got the same
error: 
```
-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
Scanning dependencies of target RtabmapPlugin
[ 50%] Building CXX object CMakeFiles/RtabmapPlugin.dir/main.cpp.o
[100%] Linking CXX shared library libRtabmapPlugin.so
/opt/android-ndk-r21/toolchains/llvm/prebuilt/linux-x86_64/lib/gcc/aarch64-linux-android/4.9.x/../../../../aarch64-linux-android/bin/ld: cannot find -lRTABMap_CORE_RELEASE-NOTFOUND
/opt/android-ndk-r21/toolchains/llvm/prebuilt/linux-x86_64/lib/gcc/aarch64-linux-android/4.9.x/../../../../aarch64-linux-android/bin/ld: cannot find -lRTABMap_UTILITE_RELEASE-NOTFOUND
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
CMakeFiles/RtabmapPlugin.dir/build.make:340: recipe for target 'libRtabmapPlugin.so' failed
make[2]: *** [libRtabmapPlugin.so] Error 1
CMakeFiles/Makefile2:92: recipe for target 'CMakeFiles/RtabmapPlugin.dir/all' failed
make[1]: *** [CMakeFiles/RtabmapPlugin.dir/all] Error 2
Makefile:100: recipe for target 'all' failed
make: *** [all] Error 2
```

matlabbe, 2020-09-28:
I checked on another project using rtabmap android as library, and there is the minimal requirements to successfully pass the linking. The `CMakeLists.txt`:
```cmake
cmake_minimum_required(VERSION 2.8)
PROJECT( MyProject )

SET(CMAKE_MODULE_PATH "${PROJECT_SOURCE_DIR}")
FIND_PACKAGE(RTABMap COMPONENTS Core Utilite REQUIRED)
FIND_PACKAGE(OpenCV REQUIRED)
FIND_PACKAGE(PCL REQUIRED)

find_package(GTSAM REQUIRED)

SET(INCLUDE_DIRS
   ${RTABMap_INCLUDE_DIRS}
   ${OpenCV_INCLUDE_DIRS}
   ${PCL_INCLUDE_DIRS}
)

SET(LIBRARIES
   ${RTABMap_LIBRARIES}
   gtsam
   ${OpenCV_LIBRARIES}
   ${PCL_LIBRARIES}
)

# OpenMP ("-fopenmp" should be added for flann included in PCL)
# the gcc-4.2.1 coming with MacOS X is not compatible with the OpenMP pragmas we use, so disabling OpenMP for it
if((NOT APPLE) OR (NOT CMAKE_COMPILER_IS_GNUCXX) OR (GCC_VERSION VERSION_GREATER 4.2.1) OR (CMAKE_CXX_COMPILER_ID STREQUAL "Clang"))
  find_package(OpenMP)
endif()
if(OPENMP_FOUND)
  set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} ${OpenMP_C_FLAGS}")
  set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} ${OpenMP_CXX_FLAGS}")
  set(CMAKE_INSTALL_OPENMP_LIBRARIES TRUE)
  message (STATUS "Found OpenMP")
  if(PCL_OMP)
    add_definitions(-DPCL_OMP)
  endif(PCL_OMP)
else(OPENMP_FOUND)
  message (STATUS "Not found OpenMP")
endif()

ADD_DEFINITIONS(${PCL_DEFINITIONS})

INCLUDE_DIRECTORIES(${INCLUDE_DIRS})

include (GenerateExportHeader)
add_library(RtabmapPlugin SHARED main.cpp)
TARGET_LINK_LIBRARIES(RtabmapPlugin ${LIBRARIES})
```

The example `main.cpp`:
```cpp
#include "rtabmap/core/Rtabmap.h"

int main(int argc, char* argv[])
{
        rtabmap::Rtabmap rtabmap;
        return 0;
}
```

Build commands (based on the [android 24 bionic android docker file](https://github.com/introlab/rtabmap/tree/master/docker/bionic/android/rtabmap_api24)):
```bash
$ ~/cmake-3.17.0-Linux-x86_64/bin/cmake -DCMAKE_TOOLCHAIN_FILE=$ANDROID_NDK/build/cmake/android.toolchain.cmake -DANDROID_ABI=arm64-v8a -DANDROID_NDK=$ANDROID_NDK -DANDROID_NATIVE_API_LEVEL=24 -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/opt/android/arm64-v8a -DCMAKE_FIND_ROOT_PATH="/opt/android/arm64-v8a/bin;/opt/android/arm64-v8a;/opt/android/arm64-v8a/share" -DOpenCV_DIR=/opt/android/arm64-v8a/sdk/native/jni -DBUILD_SHARED_LIBS=OFF ..
$ make

-- Build files have been written to: /root/test/build
[ 50%] Building CXX object CMakeFiles/RtabmapPlugin.dir/main.cpp.o
[100%] Linking CXX shared library libRtabmapPlugin.so
[100%] Built target RtabmapPlugin
```

cheers,
Mathieu

kaatrasa, 2020-09-29:
Hey, many thanks for great example!

I was able to build my plugin now. Only problem is that, after building my Unity project, the application is "unable to find" my Rtabmap-plugin even though it is included in the build. I think this could be caused by not including all dependencies in the Unity project (now only librtabmap_core.so, librtabmap_utilite.so and libNativeRtabmap.so are included). Do you know if I should be including other files also?

closed 2020-10-07 by kaatrasa

TheBricktop, 2021-02-25:
hello @kaatrasa would you mind showing code or some demo of how it is performing on android? Im really interested in slam capacity and speed of it on mobile devices.

kaatrasa, 2021-02-26:
> hello @kaatrasa would you mind showing code or some demo of how it is performing on android? Im really interested in slam capacity and speed of it on mobile devices.

Hey @TheBricktop, unfortunately I cannot share the code (at least yet), but you could try matlabbe's RTAB-Map Android application. It has some limitations but overall works very similar to my plugin. 

About SLAM performance
1) The SLAM runs real time even in large environments on android. Especially for large environments, SIFT features (not enabled in the default app, because calculating them for an image takes ~0.3-0.4 seconds on Huawei P30 Pro) are by far the best for the re-localisation accuracy & frequency. 

2) The depth image range in mapping phase is also very important if the the environment is large/open, because the loop closure transformation is done frame-to-frame with PnP so the features in the map's node require depth values. The depth range on Huawei P30 Pro is around 4 meters which works well in small environments but for large environments many features will not have depth value which makes calculating the transformation hard/impossible on many areas. One option is to use some other hardware for the mapping stage and then localize on this map with your android device. Another option we are exploring is to replace RTAB-Map's frame-to-frame registeration with global feature map in localisation mode. With very simple implementation this already looked very promising.

TheBricktop, 2021-02-26:
Hey @kaatrasa unfortunately i dont have any tango enabled phone now but ive happen to have realsense d435, is the Rtabmap-android build with realsense enabled?
1. Thats immensly promising. Wouldnt fast or orb be even more faster for feature extraction?
2. Currently im looking at a way to include slam capabilities into mobile sbc grade device with rgbd camera. Im really interested if its roubust enough to provide localization for XR headtracking.

kaatrasa, 2021-02-26:
Hey @TheBricktop, the RTAB-Map android application only requires a phone that can run ARCore/AREngine. Mapping phase requires depth sensor (and AREngine support because ARCore does not have API for raw depth images, also I think that Huawei P30 Pro camera intrinsics are hard-coded, because AREngine API fails). I have used Realsense D435/D455 with T265 for external odometry on a laptop and then transfered this map to a phone. The android build does not have realsense enabled by default, but if you manage to run the D435 on your android device then you could use that as the image source (you might need to make few modifications to https://github.com/introlab/rtabmap/blob/master/corelib/src/camera/CameraRealSense2.cpp).

1. Yes, both were faster than SIFT and would probably work well in room/apartment scale environment. However in my tests, SIFT  was able to localize more often and more accurately in large environments (university building). Also see matlabbe's https://github.com/introlab/rtabmap_ros/issues/233#issuecomment-379308078

2. We are using RTAB-Map for XR on mobile devices and I suspect the jumps on re-localisations can be even more distracting for headtracking so you will need some method to handle those. Or maybe global feature map could provide smooth enough re-localisations? https://github.com/introlab/rtabmap/issues/539#issuecomment-632805998

TheBricktop, 2021-06-23:
Hey @kaatrasa  could You point me toward some start point on how to build rtabmap on docker? Ive build some libs using docker but i have some problems.

kaatrasa, 2021-06-29:
Hey @TheBricktop, 

The default RTAB-Map android application should be built by default.
If you wish to build your own plugin or application, then you could start with the minimum example matlabbe provided above.

Summary:

1. Get the docker container
```
docker pull introlab3it/rtabmap:android-deps
docker run -it --name=rtabmap-plugin-android introlab3it/rtabmap:android-deps 
```
2. Run the modified version of the [rtabmap.bash], (https://github.com/introlab/rtabmap/blob/master/docker/bionic/android/rtabmap_api26/rtabmap.bash) script,
so that you can use RTAB-Map as a local library for your plugin.
```
#!/bin/bash

if [ $# -ne 2 ]; then
    echo "rtabmap.bash android_install_prefix api_level (23 for tango, 24 for arengine)   # Example: build.bash /opt/android 24"
    exit 1
fi

prefix=$1
api=$2

cd ~

# get rtabmap
git clone https://github.com/introlab/rtabmap.git rtabmap-tango
cd 

# tango
wget 'https://docs.google.com/uc?authuser=0&id=12rHHkYM5k-UnQn-xGXs9JqYWhSXrgJr3&export=download' -O TangoSDK_Ikariotikos_C.zip
unzip -qq TangoSDK_Ikariotikos_C.zip
rm TangoSDK_Ikariotikos_C.zip
cp -r lib_tango_client_api/include/* $prefix/arm64-v8a/include/.
cp -r lib_tango_client_api/lib/arm64-v8a/* $prefix/arm64-v8a/lib/.
rm -r lib_tango_client_api
wget 'https://docs.google.com/uc?authuser=0&id=1AqVuEVu5284X6OgrGWu12VTrx4pY99Jb&export=download' -O TangoSupport_Ikariotikos_C.zip
unzip -qq TangoSupport_Ikariotikos_C.zip
rm TangoSupport_Ikariotikos_C.zip
cp -r lib_tango_support_api/include/* $prefix/arm64-v8a/include/.
cp -r lib_tango_support_api/lib/arm64-v8a/* $prefix/arm64-v8a/lib/.
rm -r lib_tango_support_api
wget 'https://docs.google.com/uc?authuser=0&id=1s5iPJ7xiridj9Jj--gCy2XiQFniheVm6&export=download' -O TangoSDK_Ikariotikos_Java.jar
mv TangoSDK_Ikariotikos_Java.jar rtabmap-tango/app/android/libs/.

# ARCore
wget 'https://docs.google.com/uc?authuser=0&id=1VsibeqRYpS5pjmrG-vYTXyiPg8kbIfVN&export=download' -O arcore.zip
unzip -qq arcore.zip
rm arcore.zip
cp -r arcore1_18/include/* $prefix/arm64-v8a/include/.
cp -r arcore1_18/arm64-v8a/* $prefix/arm64-v8a/lib/.
cp arcore1_18/*.jar rtabmap-tango/app/android/libs/.
rm -r arcore1_18

# AREngine
wget 'https://docs.google.com/uc?authuser=0&id=1rdaD2Z1QBv-SUeUy0oBmg3C2odfxTHgR&export=download' -O arengine.zip
unzip -qq arengine.zip
rm arengine.zip
cp -r arengine/include/* $prefix/arm64-v8a/include/.
cp -r arengine/arm64-v8a/* $prefix/arm64-v8a/lib/.
cp arengine/*.jar rtabmap-tango/app/android/libs/.
rm -r arengine

# resource tool
cd rtabmap-tango/build
cmake -DANDROID_PREBUILD=ON ..
make
cd ../..

# rtabmap
mkdir rtabmap-tango/build/arm64-v8a
cd rtabmap-tango/build/arm64-v8a
~/cmake-3.17.0-Linux-x86_64/bin/cmake -DCMAKE_TOOLCHAIN_FILE=$ANDROID_NDK/build/cmake/android.toolchain.cmake -DANDROID_ABI=arm64-v8a -DANDROID_NDK=$ANDROID_NDK -DANDROID_NATIVE_API_LEVEL=$api -DBUILD_SHARED_LIBS=OFF -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=$prefix/arm64-v8a -DCMAKE_FIND_ROOT_PATH="$prefix/arm64-v8a/bin;$prefix/arm64-v8a;$prefix/arm64-v8a/share" -DBUILD_EXAMPLES=OFF -DBUILD_TOOLS=OFF -DOpenCV_DIR=$prefix/arm64-v8a/sdk/native/jni ../..
make
make clean
make install
```
3. Copy the minimum example from matlabbe's comment and run the given build command.
4. To copy the plugin from docker to your pc run
`docker cp rtabmap-plugin-android:/root/<PLUGINPATH>/libRtabmapPlugin.so .`

TheBricktop, 2021-06-29:
Thank you very much, now I only need to modify it so it includes realsense 2 for depth camera, good thing that I have some prebuild aar around.