introlab/rtabmap #607: RegistrationIcp cannot do registration with a null guess.

mdasari823, 2020-10-01:
Hi I get the following error when I use Lidar odometry.

`[ERROR] (2020-10-01 12:35:10.986) RegistrationIcp.cpp:1301::computeTransformationImpl() RegistrationIcp cannot do registration with a null guess.`

@matlabbe, in your paper here, you mentioned that 

'**_In case ICP cannot find a transformation, odometry is lost. In contrast to visual odometry, lidar odometry cannot recover from being lost when the motion prediction is null, in order to avoid large odometry errors. Lidar odometry must then be reset._**' 

How is lidar odometry fundamentally different from visual odometry here? Is there an option in rtabamp, to see how bad (i.e., large errors that you are referring to) lidar odometry can be if we just let it continue doing odometry even when motion prediction is null?

matlabbe, 2020-10-01:
Hi,

You can set `Odom/ResetCountdown` parameter to 1 to force an automatic reset of odometry from latest pose.

ICP does unknown correspondences estimation as for visual odometry, it dos known correspondences estimation. With visual approach, we can reliably recover from being lost because we can do feature matching. For ICP, there is no implementation of unique lidar feature matching that could be used to recover from being lost.

cheers,
Mathieu

mdasari823, 2020-10-02:
Thanks @matlabbe, this is what I was looking for. But like you said, it seems it produces really large errors. Initially, I thought this is because of the computation capacity I have on my host laptop. But looks like it needs some sort of correction mechanism. Do you have any suggestions for improving this? Thank you.

matlabbe, 2020-10-07:
If the scans don't overlap, there is not so much we can do with ICP.