introlab/rtabmap #609: Using sift and SuperPoint

Zumbalamambo, 2020-10-06:
I have managed to install rtabmap using `ros-noetic`. Unfortunately, the sift and the superpoint options are greyed out. How do I enable them while building?

Zumbalamambo, 2020-10-07:
The docker also doesn't have those options enabled! @matlabbe 
I tried to install with docker. The sift, surf, superpoint are all greyed out :(

matlabbe, 2020-10-07:
Which OpenCV version are you using? Prior to 4.4, you need to enable NONFREE option when building OpenCV to have SIFT. 
You can do `rtabmap --version` to see to which OpenCV version you built against, or check in About dialog. SURF requires NONFREE option for all OpenCV versions. For SuperPoint, rtabmap should be built with libtorch (c++) dependency. The cmake log should show which dependency has been found before building.

closed 2020-10-09 by Zumbalamambo