introlab/rtabmap #610: [Bug] _lastGlobalLoopClosureId

MotorTrick, 2020-10-07:
Hi, I was just checking your info topic. _lastGlobalLoopClosureId should be only for global loop closures right? (green flash on the gui + red links on the map). Still the outputs also proximity detection loop closures on Loop/Last_id and Loop/id. Is this right? Shouldn't be only for the global ones? Is there a way to get _only_ the global ones from some topic/info?

matlabbe, 2020-10-07:
`Loop/Id` does have combined loop closure and proximity detection ids:
https://github.com/introlab/rtabmap/blob/bd80811ea10c933b19f8a41b63b8c9af8efa0598/corelib/include/rtabmap/core/Statistics.h#L55

In Statistics object, you can get loop closure and proximity detection ids separately:
https://github.com/introlab/rtabmap/blob/bedc771fa4acde22c1497811a88de50c42a248e8/corelib/include/rtabmap/core/Statistics.h#L256-L259

If you are on ROS, the [Info](https://github.com/introlab/rtabmap_ros/blob/f7f9e82cb17c9f391d37f27259c0ab0dcd55d914/msg/Info.msg#L8-L11) msg does have ids split too:
```
int32 refId
int32 loopClosureId
int32 proximityDetectionId
int32 landmarkId
```

MotorTrick, 2020-10-08:
yup but loopClosureId is (almost) always zero apart from when either a local/global loop closure happens. So there is no real way to get _only_ the global loop closure directly. Unless I check Loop/Last_id against local loop closures ids. If they're equal it's a local loop closure if they're different it's a global one.

matlabbe, 2020-10-08:
Hi, I am using both here:
https://github.com/introlab/rtabmap/blob/eef0a23b1b92a6ed4874eb06d27ae692be8cbcf4/tools/Reprocess/main.cpp#L697
...
https://github.com/introlab/rtabmap/blob/eef0a23b1b92a6ed4874eb06d27ae692be8cbcf4/tools/Reprocess/main.cpp#L708-L715
and it works as expected. In recent version, we check for proximity detection before loop closure detection, if proximity closes on same id than the loop closure detection, the loop closure id is suppressed. We can know that when LoopSuppressed_hypothesis_id statistic is not zero.

Just to make sure, which loopClosureId are you referring, in ROS or in Statistics?

MotorTrick, 2020-10-08:
The one published in the topic /rtabmap/info. The count is indeed correct, both in the rtabmap gui and if I listen to the topic in rviz. But I'm sadly not able to get only the global loop closure event through that topic.

matlabbe, 2020-10-08:
Here is an output of rostopic echo `/rtabmap/info/loopClosureId` on left and rostopic echo `/rtabmap/info/proximityDetectionId` on right:
[image]
The outputs are aligned. They match the color background in rtabmapviz at the same time (green for left and yellow for right). Which rtabmap_ros version are you using?

MotorTrick, 2020-10-08:
The count is indeed correct, both in the rtabmap gui and if I listen to the topic in rviz. I think the "issue" is this one 

> In recent version, we check for proximity detection before loop closure detection, if proximity closes on same id than the loop closure detection, the loop closure id is suppressed.

I was mostly testing locally so that was probably what happened. I'll check better tomorrow and get back to you, it's sleep time here were I am.

MotorTrick, 2020-10-08:
(Thanks for your patience and support man ;) I've answered also to my other two issues if you happen to have time ;) )

matlabbe, 2020-10-08:
You can decrease `RGBD/ProximityPathFilteringRadius` to 0.1 (default 1 meter) for example to trigger less often proximity detections.

MotorTrick, 2020-10-12:
Ok, I checked and seems that loopClosureId is indeed working >.< sorry..

closed 2020-10-12 by MotorTrick