introlab/rtabmap #612: Map size increasing

AWilson10, 2020-10-09:
We have found that each time we start a mapping session , the size of the map grows, even though we are mapping the same area. Is there a way to maintain a fixed map size and cull old nodes / transforms?

matlabbe, 2020-10-09:
The most effective way to avoid growing the map during navigation is to switch to localization mode when you know you are revisiting the same area and you don't want to update it. Otherwise, some other options in SLAM mode to use less memory:

- `Mem/ReduceGraph` (experimental): this approach is explained in this [paper](https://introlab.3it.usherbrooke.ca/mediawiki-introlab/images/8/87/LabbeAURO2017.pdf) (section 4.3). In summary, if loop closures are found, the graph won't grow. As discussed in the paper, having also the possibility to set a maximum density of nodes in the map could be a nice new option to have.
- `Mem/BinDataKept` (default true): If false, binary data (RGB and depth images) are not saved in database.
- `Mem/NotLinkedNodesKept` (default true): If false, nodes not in graph won't be saved to database (when the robot is not moving for example).

The last two are enabled by default for debugging purpose and recovery.

AWilson10, 2020-10-10:
Thanks for the speedy response.

closed 2020-10-10 by AWilson10