introlab/rtabmap #613: Creating a pointcloud from a known camera pose and rgb image

Zumbalamambo, 2020-10-09:
I would like to create the point cloud from the extrinsic and intrinsic camera matrix and the images of the known pose.

I have selected the following values in source,

1. I have set the source type to `RGB` and `images` under RGB section

2. I have pointed to the directory that contains about 140 images.

3. I have selected also the pose file in the kitti format in the `odometry file`.

Unfortunately, when I run, it throws the following error,

`[FATAL] (2020-10-09 14:23:04.905) CameraImages.cpp:440::readPoses() Condition (outputPoses.size() == inOutStamps.size()) not met! [140 vs 0]
`

how can I fix this?

tkircher, 2020-10-12:
This seems like more of a photogrammetry problem. I recommend giving meshroom a try. They actually just put out a new release a couple days ago.

Zumbalamambo, 2020-10-12:
@tkircher Does meshroom have an option to export depth images? 

The problem that I have is that I have created a database using Tango. Now that the RGB images are of size 540x960 while depth images are of size 135x240. Since the RGB and the depth images are of different resolution I'm not sure how can I create an rgbd image out of this for 2D-3D correspondence.

tkircher, 2020-10-12:
@Zumbalamambo If you already have depth information, and it's aligned to the images, scale it up to match the RGB data then generate an appropriate RGBD file.

closed 2020-10-12 by Zumbalamambo

matlabbe, 2020-10-12:
RTAB-Map can support lower resolution depth images, as long as the RGB dimension modulo depth dimension is 0 and that depth image is registered to RGB image. RTAB-Map will scale the depth as needed for 3D map or motion estimation. If you created the database using RTAB-Map Tango app, it should be already compatible (like the scans in this [tutorial](https://github.com/introlab/rtabmap/wiki/Multi-Session-Mapping-with-RTAB-Map-Tango)). 

With KITTI format, a timestamp file is required. Set "Timestamps file" or check "Use file names as timestamps". If images are numbered like 0.jpg,1.jpg,2.jpg,3.jpg,4.jpg... The framerate will be 1 Hz. To increase frame rate, set Input Rate to 5 Hz for example, and uncheck "Synchronize capture rate with timestamps.".