introlab/rtabmap #614: D455 vs. D435i for 3D Room scanning

peta-peta, 2020-10-12:
Dear all, 
I want to scan rooms and create water tight meshes. I have started exploring intel realsense L515 with RTABMAP-Ros but I am not happy with the results. Mapping is very unstable and sensitive to daylight. Since I can see many successful applications using realsense D435i I am humbled to buy one too and use it directly with RTABMap. Today i learned that Intel has already released a new version D455 and I am wondering if any of you has some experience with it using RTABmap for mapping and odometry. Which one shall I buy, or do you suggest even a different camera or a combination of different devices for room scanning with RTABMap - Room has to be a watertight model with floor, walls, windows and ceiling.

Cheers Peter

rlabs-oss, 2020-10-16:
Hi, 
    I own the D455, L515 and D415 (but not the D435) - the L515 I got working with rtabmap following the earlier posts on the camera. For the D455 it works out of the box. - Once the IMU gravity is adjusted the D455 works very well. Ive used it to scan my house, outside structures, roads etc. The only issue I have with the cameras is the lack of portability.*1 I tried building rtabmap for the pi4 except it segfaults on start. Realsense viewer is working. 

*1 Ive also built my own 3D lens for a canon EOS M as a portable rig, knock off of the Lumix 3D which I also own, neither will calibrate in rtabmap even with ffmpeg tricks to crop/resize/lenscorrect the stream before passing it to the v4l2 loopback (and opening that in rtabmap) -Looking into opencv2 and how that mangles reality next. 
 
D455 top, L515 bottom - both scanned with rtabmap (and changes for the L515 code)
[https://urlme.net/dl/Sofa.jpg](url)
Not complete due to the laptop dropping frames using the L515.
Mapping with the D455
[https://urlme.net/dl/mapping.gif](url)

peta-peta, 2020-10-16:
Hi @rlabs-oss 
thank you very much for your detailed explanation. I will give the D455 a try.
Regarding the L515, can you share your launch script with us? I suppose you run `roslaunch  rtabmap_ros` with a launch script, right?
Best wishes,
Peter

rlabs-oss, 2020-10-16:
Hi, 
     I added the changes to the standalone app as suggested in this issue: https://github.com/introlab/rtabmap/issues/574 then rebuilt it. Main changes being the L500 Sensor:
`		}
		else if ("L500 Depth Sensor" == module_name)
		{
				sensors[1] = elem;	
		}
		else if ("Wide FOV Camera" == module_name)`

and the depth sensor selection

`	if ((!ir_ && !irDepth_ && video_profile.stream_type() == RS2_STREAM_DEPTH) // L515 fix, use first available depth stream
				|| (video_profile.width()  == cameraWidth_ &&
				video_profile.height() == cameraHeight_ &&
				video_profile.fps()    == cameraFps_))
				//D400 series:`

Currently I'm trying to figure out how to select the L515's infrared stream rather than the RGB stream which will be more useful in the dark.
-M

matlabbe, 2020-10-18:
For rtabmap standalone, a new commit has been added for L515: https://github.com/introlab/rtabmap/commit/3047b7da6b734628bdfe67eef1550d61a928a0d7

peta-peta, 2020-10-20:
I tried the new commit for realsense L515. 
I executed ` ./rtabmap-rgbd_camera 11`  and got this error 
`[ERROR] (2020-10-20 20:04:53.761) CameraRealSense2.cpp:1374::captureImage() Missing frames (received 1)`
Attached you can also find the full log file.
[log.txt](https://github.com/introlab/rtabmap/files/5410658/log.txt)

matlabbe, 2020-10-20:
Can you comment this [line](https://github.com/introlab/rtabmap/blob/fbc30042c4d0b1a25b29ea3876e2bf625ec9bb42/corelib/src/camera/CameraRealSense2.cpp#L1013) and see if it changes something?

peta-peta, 2020-10-20:
I comment 
`//sensors[i].set_option(rs2_option::RS2_OPTION_GLOBAL_TIME_ENABLED, 1);`
and recompiled `cmake ..` and `make -j4` (or do I need to clean bin and build folders?)

Same Error.

rlabs-oss, 2020-10-20:
I can detail the workarounds I made - 1st fix was to the gyro which if using Ubuntu20.04 the kernel stream sync isn't active.
 
[This line](https://github.com/introlab/rtabmap/blob/fbc30042c4d0b1a25b29ea3876e2bf625ec9bb42/corelib/src/camera/CameraRealSense2.cpp#L392) 
to:
`if(((iterA == iterB) && (stamp == iterA->first)) || isL500_)'

Since steams maybe de synced, I ignored them for the L500

And [this line](https://github.com/introlab/rtabmap/blob/fbc30042c4d0b1a25b29ea3876e2bf625ec9bb42/corelib/src/camera/CameraRealSense2.cpp#L1009)
to
` if((!isL500_) && sensors[i].supports(rs2_option::RS2_OPTION_GLOBAL_TIME_ENABLED))'

[Desired frames](https://github.com/introlab/rtabmap/blob/fbc30042c4d0b1a25b29ea3876e2bf625ec9bb42/corelib/src/camera/CameraRealSense2.cpp#L1145) is also problematic if the IMU info doesnt arrive on time, comment it out
`		UTimer timer;
		int desiredFramesetSize = 2;
		//if(isL500_)
		//	desiredFramesetSize = 2; 

More hacks really since if the kernel driver /stream sync is present the changes should not be needed.

peta-peta, 2020-10-20:
@matlabbe  and @rlabs-oss  thank you very much! It works like charm.
I ve tried already some tool under ros, but the results with rtabmap are already very impressive and much more stable! I will keep you posted about my progress.
Cheers Peter

rlabs-oss, 2020-10-26:
The D455 + T265 quick room scan. The scanning was faster than shown in the video since I was post processing it and recording for upload. 
https://youtu.be/IDWoLNk0l5o

samhoff20, 2020-11-03:
Hi guys, even after the new commit I don't see the l515 in my source list? How do I initialize the camera?

rlabs-oss, 2020-11-03:
Manually setup the camera via the Select sensor driver/More Options button, or Detection - Select source - more options. 
 - Source Type RGB-D
Grabber for RGB-D devices...
- Driver Realsense2
- IR Emitter enabled
- IR mode unchecked
- Use Depth... unchecked. 
640 Stream width 
480 Stream height
30 Hz rate
In the standalone app should work.

samhoff20, 2020-11-03:
Thank you! I seem to be having some realsense issues also but I will try this once I get them worked out!

peta-peta, 2020-11-03:
I confirm that solution provided by rslabs-oss is working without problems. I was also able to use the standard settings of the D435 provided by RTABmap standalone tool for the L515.
Settings by rlabs-oss work better though.
Cheers Peter

samhoff20, 2020-11-03:
So trying the custom source and the 415 both get me this output. Do you have any insight?
[image]
[image]

samhoff20, 2020-11-03:
[image]

rlabs-oss, 2020-11-03:
Hi,
I have not had a chance to try 0.20.6 with the [workarounds](https://github.com/introlab/rtabmap/issues/614#issuecomment-713072426) added but will check it now.
-M

rlabs-oss, 2020-11-03:
Leaving the gyro code alone (which will warn: "CameraRealSense2.cpp:409::getPoseAndIMU() Could not find gyro data to interpolate at image time 76198.835000 (earliest is 1604436689619.677734). Are sensors synchronized?" when using the L515 and [Commenting out](https://github.com/introlab/rtabmap/blob/fbc30042c4d0b1a25b29ea3876e2bf625ec9bb42/corelib/src/camera/CameraRealSense2.cpp#L1145) those lines to read
		//if(isL500_)
		//	desiredFramesetSize = 3;

And adding the previous hack for global sync in [this line](https://github.com/introlab/rtabmap/blob/fbc30042c4d0b1a25b29ea3876e2bf625ec9bb42/corelib/src/camera/CameraRealSense2.cpp#L1009)
So it reads: 
` 			 if((!isL500_) && sensors[i].supports(rs2_option::RS2_OPTION_GLOBAL_TIME_ENABLED))`

Will enable the L515 at 640x480x30 in release 0.20.6 without causing the D415 to stop working. 
I tested this in dual mode T265 + D415. 
Single: D415, D455 (640x480x30, 60hz, and 1280x720x30)
L515: 640x480x30

(librealsense 2.39.0 October 1st release, and all camera firmwares updated)

samhoff20, 2020-11-03:
Thanks for the help. I tried the above edits but still can't initialize the camera. This is the output of the terminal running rtab:
[image]

rlabs-oss, 2020-11-04:
@samhoff20 Is the camera working with the realsense-viewer without error? Turn on all streams/modules in the viewer and watch the console log.

The difference for the changes is shown below. 

 diff ~/external-source/GitHub_RTAB/rtabmap/corelib/src/camera/CameraRealSense2.cpp corelib/src/camera/CameraRealSense2.cpp 
1009c1009
<                        if(sensors[i].supports(rs2_option::RS2_OPTION_GLOBAL_TIME_ENABLED))
---
>                        if((!isL500_) && sensors[i].supports(rs2_option::RS2_OPTION_GLOBAL_TIME_ENABLED))
1145,1146c1145,1146
<               if(isL500_)
<                       desiredFramesetSize = 3;
---
>               //if(isL500_)
>               //      desiredFramesetSize = 3;

check the output from ldd against rtabmap in the bin folder to show which version of librealsense2 its using. 

ldd rtabmap | grep real
        librealsense2.so.2.39 => /usr/local/lib/librealsense2.so.2.39

samhoff20, 2020-11-04:
This is the error I'm getting in the realsense terminal when trying to press play on rtabmap. 
`realsense-viewer and having both depth sensor and rgb camera on (not motion module though).
 04/11 07:44:02,123 ERROR [139668014343936] (types.h:313) set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 04/11 07:44:02,124 ERROR [139668014343936] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 04/11 07:44:02,247 ERROR [139667856922368] (types.h:313) xioctl(VIDIOC_DQBUF) failed for fd: 23 Last Error: No such device
 04/11 07:44:02,247 ERROR [139667856922368] (backend-v4l2.cpp:1354) xioctl(VIDIOC_DQBUF) failed for fd: 23 Last Error: No such device
 04/11 07:44:02,247 ERROR [139667659785984] (types.h:313) xioctl(VIDIOC_DQBUF) failed for fd: 27 Last Error: No such device
 04/11 07:44:02,247 ERROR [139667659785984] (backend-v4l2.cpp:1354) xioctl(VIDIOC_DQBUF) failed for fd: 27 Last Error: No such device
 04/11 07:44:02,247 ERROR [139667571734272] (types.h:313) xioctl(VIDIOC_DQBUF) failed for fd: 41 Last Error: No such device
 04/11 07:44:02,247 ERROR [139667571734272] (backend-v4l2.cpp:1354) xioctl(VIDIOC_DQBUF) failed for fd: 41 Last Error: No such device
 04/11 07:44:03,124 ERROR [139668014343936] (types.h:313) acquire: Cannot open '/dev/video0 Last Error: No such file or directory
 04/11 07:44:03,125 ERROR [139668014343936] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: acquire: Cannot open '/dev/video0 Last Error: No such file or directory
`

Then this last ERROR[139668014343936] repeats endlessly.

samhoff20, 2020-11-04:
If I am just in the realsense viewer and I turn on both depth sensor and rgb turn on fine, but I get the following error when turning on the motion module if that is an issue: `$ realsense-viewer
 04/11 07:46:18,485 WARNING [140248687920960] (ds5-motion.cpp:480) IMU Calibration is not available, default intrinsic and extrinsic will be used.
 04/11 07:46:18,492 WARNING [140247727863552] (backend-hid.cpp:681) HID set_power 1 failed for /sys/devices/pci0000:00/0000:00:14.0/usb2/2-7/2-7:1.7/0003:8086:0B64.0026/HID-SENSOR-200073.2.auto/iio:device0/buffer/enable
 04/11 07:46:18,540 WARNING [140247711078144] (sensor.cpp:970) HID timestamp not found, switching to Host timestamps.
`

rlabs-oss, 2020-11-04:
To clarify: realsense-viewer should not be open when using rtabmap they are both independent applications. 
Also are you using the D415 or the L515 ? The changes above are only related to the L515. The D415 should be unaffected or as functional as in the previous releases.

rlabs-oss, 2020-11-04:
> Then this last ERROR[139668014343936] repeats endlessly.

I do not experience that. With just realsense-viewer open (and not rtabmap) do you see the same error? If so it could be related to the USB bandwidth, USB port type, how the usb cable is plugged in -sounds strange but its better to plug into the usb port first then the camera.
In realsense-viewer, with the camera connected click the Info button. is the "Usb Type Descriptor" 3.2? it should look something like this for the L515:
[image]

If any one of the L500 depth sensor, RGB camera and Motion module in realsense-viewer cannot be activated or generate lots of errors those should be resolved before using rtabmap.

samhoff20, 2020-11-05:
I am using the L515.

Here is a snapshot of the info depth, rgb and motion turned on.
[image]

Anderson191919, 2020-11-22:
Hello guys,
I have an issue with the L515 on windows with rtabmap (v 0.20.6 from https://ci.appveyor.com/project/matlabbe/rtabmap/branch/master/artifacts). unfortunately I get following error(s) when starting on rtabmap (like samhoff20)

Anderson191919, 2020-11-22:
[image]

Anderson191919, 2020-11-22:
I have used these settings (#614 (comment)).

Thank you very much in advance.
best regards

Anderson191919, 2020-11-22:
[image]

rlabs-oss, 2020-11-22:
Hi, 
Problem is related to lack of hardware timestamps from the realsense SDK for the frames (inc Gyro stream) which need a custom kernel module see:- 
https://github.com/IntelRealSense/librealsense/blob/master/doc/frame_metadata.md
Its possible to check this in the realsense viewer. Top left, clilck 2D and turn all modules ON - The Motion Module Gyro stream window will indicate a red triangle error. 

In RTAB-Map, if the rs2_option::RS2_OPTION_GLOBAL_TIME_ENABLED is used together with an expected number of frames (RGB, Depth, Gyro) then for the L515 and a non-custom kernel it will stall. 

As a workaround, ==HACK, the checks for synced frames for the L515 can be disabled:  
However this will cause gyro interpolation issues since the frame arriving from the gyro has a vastly different timestamp than the images thus this warning: "CameraRealSense2.cpp:409::getPoseAndIMU() Could not find gyro data to interpolate at image time 186557.513000 (earliest is 1606059194997.468750). Are sensors synchronized?" is seen constantly (set logging to fatal to ignore those.) 

In summary the changes to the current version (as of 22nd November) are:
rtabmap/corelib/src/camera/CameraRealSense2.cpp
Line 1016 to 
1016  if((!isL500_) && sensors[i].supports(rs2_option::RS2_OPTION_GLOBAL_TIME_ENABLED))
and
Lines 1152-1153 commented out
1152		//if(isL500_)
1153		//	desiredFramesetSize = 3;
  
Windows - Ive not tested but suspect it maybe the same issue based on previous posts. 
Hope this helps.

Anderson191919, 2020-11-22:
[image]

Anderson191919, 2020-11-22:
Thank you for your effort. Do you mean this error in the realsense viewer?

Anderson191919, 2020-11-22:
I am not so good to make changes to cpp-files and then rebuild the project. Have you an idea to make the L515 run in rtabmap 0.20.6?

Anderson191919, 2020-11-22:
I have read that some realsense modules are not calibrated from factory. Am I supposed to do it? What is your opinion?

rlabs-oss, 2020-11-22:
> Thank you for your effort. Do you mean this error in the realsense viewer?
Not seen that, I get "CameraRealSense2.cpp:409::getPoseAndIMU() Could not find gyro data to interpolate at image time 324922.068000 (earliest is 1606066114724.759033). Are sensors synchronized?"

rlabs-oss, 2020-11-22:
> I am not so good to make changes to cpp-files and then rebuild the project. Have you an idea to make the L515 run in rtabmap 0.20.6?

Only by building a version with the changes as [proposed](https://github.com/introlab/rtabmap/issues/614#issuecomment-731769818)

matlabbe, 2020-11-22:
In the commit above, I added the changes from @rlabs-oss. There is a new option called "Global time sync" in Preferences->Source->RealSense2. By default it is checked, but if you uncheck it, it will do exactly @rlabs-oss proposed. 

The windows build has started, just wait https://ci.appveyor.com/project/matlabbe/rtabmap/branch/master to finish the new artifact.

matlabbe, 2020-11-22:
Please let me know if it works with a L515, as I cannot test it myself.

Anderson191919, 2020-11-22:
I will definitely. Thank you very much :)

rlabs-oss, 2020-11-22:
> Please let me know if it works with a L515, as I cannot test it myself.

Tested & Confirmed to work,

Global Time sync On - CameraThread.cpp:154::mainLoop() no more images... 
- as expected due to lack of sync.

Off - working with single warning "CameraRealSense2.cpp:442::getPoseAndIMU() As globalTimeSync option is off, the latest received gyro and accelerometer will be re-stamped with image time. This message is only shown once."

matlabbe, 2020-11-22:
Thx @rlabs-oss, I could not test the warning message I added for imu/acc, it is exactly what I wanted. Note that I missed that [change](https://github.com/introlab/rtabmap/commit/bdc7be40b4f21ae595de1ba7c7740c5f85063bb1) in the other commit, causing imu not being published.

Anderson191919, 2020-11-22:
@matlabbe thank you, but still the same error :(

Anderson191919, 2020-11-22:
[image]

Anderson191919, 2020-11-22:
[image]

Anderson191919, 2020-11-22:
Could it be really due to calibration issues? I have not calibrated it since I got it from factory. did you manually calibrate the l515? Thanks for your efforts

matlabbe, 2020-11-22:
I need the first error in the log to debug. The one starting with "Given stream configuration is not supported by the device!"

rlabs-oss, 2020-11-22:
Windows build is still in progress last time I checked. My earlier comment was for my local Linux build.

rlabs-oss, 2020-11-22:
> Could it be really due to calibration issues? I have not calibrated it since I got it from factory. did you manually calibrate the l515? Thanks for your efforts
I also see (ds5-motion.cpp:480) IMU Calibration is not available, default intrinsic and extrinsic will be used. when using realsense-viewer yet this does not affect its use in RTAB-Map, However that shouldn't be ruled out so I will test the calibration procedure as described in [this document](https://dev.intelrealsense.com/docs/imu-calibration-tool-for-intel-realsense-depth-camera) and check if there is any improvement or change.

Anderson191919, 2020-11-23:
[image]

Anderson191919, 2020-11-23:
Here is my error log. Maybe I have to select manually the device-id? Anyway thank you for your efforts :)

Anderson191919, 2020-11-23:
Or do I need to install Point Cloud Library, GTSAM or g2o?

Anderson191919, 2020-11-23:
[LogRtabmap.txt](https://github.com/introlab/rtabmap/files/5581991/LogRtabmap.txt)

Anderson191919, 2020-11-23:
> 
> 
> I need the first error in the log to debug. The one starting with "Given stream configuration is not supported by the device!"
> 
> No need to explicitly set 640x480x30 anymore in the Preferences dialog, in the code I just search for 640x480x30 when L500 is detected.

What very strange is, is that although rtabmap detects a L500 it is not adjusting the pixel and frequency automatically :/

Anderson191919, 2020-11-23:
Is rtabmap trying to use my laptop cam?

Anderson191919, 2020-11-23:
Sorry that I am frequently writing. Another problem could be maybe windows, because it is detecting the L500 as two seperate inputs in the device manager: 
[image]

In comparison an azure kinect - which works properly in rtabmap - is displayed as one device:
[image]

What is your opinion? Thank you very much in advance

matlabbe, 2020-11-23:
Based on the error [log](https://github.com/introlab/rtabmap/issues/614#issuecomment-732015849), it looks like only the RGB camera is detected at the device ID "0". However, I would have expect that 640x480 would still exist even if only the RGB camera is detected (here the minimum resolution is 960x540). If you set device ID to 1 or 2 in Preferences->Source, would you have a different output? 

What is the version or your RealSenseViewer? Does your sensor have the latest firmware version? What is it? (RealSenseViewer->Info).

@rlabs-oss Can you launch rtabmap with Logger level set to Info to show those logs: https://github.com/introlab/rtabmap/blob/80199f23b59c3982b70ebb3b2ad412067a2eca3e/corelib/src/camera/CameraRealSense2.cpp#L682-L689. I would like to compare which video profiles are detected in Linux in comparison to Windows above. Also, what is the firmware installed on your L515? Thank you so far for this debugging!

rlabs-oss, 2020-11-23:
> @rlabs-oss Can you launch rtabmap with Logger level set to Info to show those logs:

See [output](https://urlme.net/dl/rtabmap_l515.txt
)
640x480x30 is a valid RGB mode.  I will download the win version and test it on the same machine running win10. and post the same info shortly...

Anderson191919, 2020-11-23:
> 
> 
> Based on the error [log](https://github.com/introlab/rtabmap/issues/614#issuecomment-732015849), it looks like only the RGB camera is detected at the device ID "0". However, I would have expect that 640x480 would still exist even if only the RGB camera is detected (here the minimum resolution is 960x540). If you set device ID to 1 or 2 in Preferences->Source, would you have a different output?
> 
> What is the version or your RealSenseViewer? Does your sensor have the latest firmware version? What is it? (RealSenseViewer->Info).
> 
> @rlabs-oss Can you launch rtabmap with Logger level set to Info to show those logs:
> 
> https://github.com/introlab/rtabmap/blob/80199f23b59c3982b70ebb3b2ad412067a2eca3e/corelib/src/camera/CameraRealSense2.cpp#L682-L689
> . I would like to compare which video profiles are detected in Linux in comparison to Windows above. Also, what is the firmware installed on your L515? Thank you so far for this debugging!
> 
> **EDIT**: Looking at this [post](https://github.com/IntelRealSense/realsense-ros/issues/1246#issuecomment-650836142), it seems there is no 640x480 RGB profile... not sure how the current code is working on Linux?! Well, [there](https://github.com/IntelRealSense/realsense-ros/issues/1246#issuecomment-651978035) they say 640x480 works?!

If i set the device id to 1 or 2 I get the following error: 

[image]

I got this realsense viewer version: 
[image]

and this L500 firmware version:
[image]

Anderson191919, 2020-11-23:
@matlabbe and @rlabs-oss thank you very much for your efforts. It helps me a lot. Maybe we can find the solution for this issue ^^

rlabs-oss, 2020-11-23:
Just noticed librealsense is now [V2.40.0](https://github.com/IntelRealSense/librealsense/releases) which may cause changes.

Using librealsense 2.39 on Windows 10 Pro, the L515 connected via USB3.2: RTAB-Map started and mapped as in Linux. -The default resolution is 848x480x60 - I changed that to 640x480x30 before starting. 

log output is [here](https://urlme.net/dl/rtabmap_windows.txt)

In device manager I see the L515 under USB devices and under Cameras Depth + RGB. 

Ill check if this is related to the new V2.40 librealsense and find out if that changes the behavior.

[25 more replies]