introlab/rtabmap #615: import point cloud file

wangwayjay, 2020-10-13:
I want to import an existed point cloud file as a base and run the rtabmap to continue reconstruction based on the point cloud file, the format of the point cloud file may be pcd or ply, the camera device is the Azure Kinect DK. Could you help me acheive this goal please? thank you very much.

wangwayjay, 2020-10-13:
Could I first convert the point cloud file to a db file, then load the db file as a base? How to convert a point cloud file to a db file is still a problem.

matlabbe, 2020-10-17:
See http://official-rtab-map-forum.67519.x6.nabble.com/import-point-cloud-file-td7022.html#a7046

closed 2020-10-17 by matlabbe

mattiasmar, 2024-03-15:
@matlabbe is this still not possible? 
I would like to initialize the map of rtabmap with .ply file and run rtabmap localization against it. Do you see how that would be possible?

matlabbe, 2024-03-16:
Well, local ICP localization could be possible (with `RGBD/ProximityGlobalScanMap=true`), but it is a workflow that is not supported. To import the PLY in rtabmap, we would assume the map is made from single node containing the whole point cloud. I have a work-in-progress [PR](https://github.com/introlab/rtabmap/pull/1161) to use rtabmap standalone without camera, but till then, you have to provide also an image. The image could be black pixels with fake calibration.

[image]
[image]

Global localization would not work, so to make proximity detection working when switching to localization mode, the sensor/robot would need to start at origin of the map (`RGBD/StartAtOrigin=true`).