introlab/rtabmap #617: Units of the 3D Projection value

Zumbalamambo, 2020-10-23:
I'm using the following code to create a 3D projection.

```

import cv2
import numpy as np

# camera matrix
mtx = [8.6767822265625000e+02, 0., 4.8268579101562500e+02, 0.,
       8.6760119628906250e+02, 2.7117459106445312e+02, 0., 0., 1.]

# parse calib matrix
fx = mtx[0]
cx = mtx[2]

fy = mtx[4]
cy = mtx[5]

depth_img = cv2.imread("RTAB-Map/bundler/depth/1326.png", cv2.IMREAD_ANYDEPTH)
depth_row, depth_column = depth_img.shape

# to store 3d points
pts3d = []

for i in range(depth_row):
    for j in range(depth_column):

        # retrieve depth value
        depth = depth_img[i][j]
        x, y = i, j

        if depth > 0.0:
            x3D = (x - cx) * depth / fx
            y3D = (y - cy) * depth / fy
            z3D = depth

            # populate 3d points array
            pts3d.append([x3D, y3D, z3D])
        else:
            pts3d.append([-1, -1, -1])
            
row,col = 540,960
pts3d = np.array(pts3d).reshape(row,col,3)
print(pts3d[row-1,col-1])

```
In the very last line, I have tried to access xyz values. It prints `[ 50.36409239 615.20491168 776.        ]`

They seem to be very big.what are the coordinates of these values?

Zumbalamambo, 2020-10-24:
Is the Unit in `cm` because if the pts3d is divided by 100, it gives an almost accurate 3d position in the real world?

matlabbe, 2020-11-06:
Depth values are either in mm or meters.

closed 2020-11-09 by Zumbalamambo