introlab/rtabmap #619: RVIZ demo not working as presented in the tutorial

yohannes-taye, 2020-10-25:
Here is my issue 

I have just setup rtabmap and the visualization in rtabmapviz is working just as intended but while trying to use RViz its not showing the reconstruction at all. I looked at the demo on the website and the reconstruction is rendered by subscribing to two topics `/assembled_cloud `which is not showing up in the topics list while I run` rostopic list`. Am I missing something here? 
[image]

matlabbe, 2020-11-06:
In the demo, we use MapCloud plugin to show the 3D map. Make sure the topic is set in the plugin. What does `rostopic list` show?

yohannes-taye, 2020-11-10:
@matlabbe I did manage to fix the issue. I used the MapCloud plugin but PointCloud2 topics are still not working.