introlab/rtabmap #623: Localization parameters [question]

NicoNorena, 2020-10-28:
Are the Bayes Filter parameters the only setting when in localization mode so that `Localization_pose` Topic gets published more often?

Would the `MaxOdomCacheSize` param help in any way?

matlabbe, 2020-11-07:
To get the pose published more often, you can increase `Rtabmap/DetectionRate` (default 1 Hz), or even set it to 0 to publish as fast as possible. 

`RGBD/MaxOdomCacheSize` is used to detect wrong loop closures in localization mode. Here is the description:
```
$ rtabmap --params | grep MaxOdomCacheSize
```

> Param: RGBD/MaxOdomCacheSize = "0"                         [Maximum odometry cache size. Used only in localization mode (when Mem/IncrementalMemory=false) and when RGBD/OptimizeMaxError!=0. This is used to verify localization transforms to make sure we don't teleport to a location very similar to one we previously localized on. When the cache is full, the whole cache is cleared and the next localization is automatically accepted without verification. Set 0 to disable caching.]

NicoNorena, 2020-11-10:
thanks. I will play the `Rtabmap/DetectionRate` in localization mode.

in what scenerio would it help to have `Bayes/FullPredictionUpdate` set to `true`  ?
and how did determine  the values for  `Bayes/PredictionLC` ?

closed 2021-08-18 by NicoNorena

matlabbe, 2021-09-03:
`Bayes/FullPredictionUpdate` is more a testing parameter, what I remember is that in some cases regenerating the full prediction may be faster than iterating all modifications to do. However I think the new incremental version is always faster. Computation time could be compared with:
```
rtabmap-report Timing/Posterior_computation/ms rtabmap.db
```

For `Bayes/PredictionLC`, I would refer you to this [paper](https://introlab.3it.usherbrooke.ca/mediawiki-introlab/images/b/bc/TRO2013.pdf), Section III-C, the transition model explanation (point 4).

cheers,
Mathieu