introlab/rtabmap #625: RGBD/SavedLocalizationIgnored not working

rahulmoorthy19, 2020-10-29:
Hello, 

Thank you for the this great mapping technique. Really helping me a lot.

I was working on a project. when the robot is in completely different room in localization mode. Even then the localization is happening at the last localized position. I had also set the RGBD/SavedLocalizationIgnored parameter as true but even after that the error persisted. Help on this would be much appreciated. I have added the rtabmap and rgbdodom parameters below for your reference. 

  <node pkg="rtabmap_ros" type="rgbd_odometry" name="rgb_odometry" >
    <param name="Vis/FeatureType" value="2"/>
    <param name="Vis/MaxFeatures" value="4000"/>
    <param name="Vis/MinInliers" value="20"/>
    <param name="Reg/Force3DoF"    value="true" />
    <param name="Odom/Strategy"    value="1" />
    <param name="frame_id"               type="string" value="base_link"/>
    <param name="Odom/ResetCountdown"    type="string" value="1"/>
  </node>

  <node pkg="rtabmap_ros" type="rtabmap" name="rtabmap" output="screen" args="$(arg rtabmap_args)">
    <remap from="scan" to="/scan"/>
    <param name="RGBD/ProximityBySpace"     type="string" value="false"/>
    <param name="wait_for_transform_duration"     type="double" value="5"/>
    <param name="RGBD/LinearUpdate" value="0"/>
    <param name="RGBD/AngularUpdate" value="0"/>
    <param name="Rtabmap/DetectionRate" value="20"/>
    <param name="Rtabmap/MemoryThr" value="2000"/>
    <param name="Kp/DetectorStrategy" value="2"/>
    <param name="Kp/MaxFeatures" value="2000"/>
    <param name="queue_size" type="int" value="10"/>
    <param name="Reg/Force3DoF"    value="true" />
    <param name="Optimizer/Slam2D" value="true" />
    <param name="grid_size" type="double" value="50"/> 

    <param name="Reg/Force3DoF"             type="string" value="true"/>
    <param name="database_path"        type="string" value="$(arg database_path)"/>
    <param name="subscribe_scan" type="bool" value="true"/>
    <param name="subscribe_scan_cloud" type="bool" value="false"/>
    <param name="Grid/FromDepth" type="string" value="false"/>
    <param name="odom_frame_id" type="string" value="odom"/>
    <param name="RGBD/SavedLocalizationIgnored" type="string" value="true"/>
    <remap from="grid_map" to="/map"/>
    <remap from="planner_goal" to="/move_base_simple/goal"/>

    <param     if="$(arg localization)" name="Mem/IncrementalMemory" type="string" value="false"/>
    <param unless="$(arg localization)" name="Mem/IncrementalMemory" type="string" value="true"/>
    <param name="Mem/InitWMWithAllNodes" type="string" value="$(arg localization)"/>

  </node>

Thank You!!

rahulmoorthy19, 2020-10-29:
It always starts by saying Rtabmap.cpp:1076::process() Update map correction based on last localization saved in database! correction = xyz=0.000000,0.000000,0.000000 rpy=0.000000,-0.000000,0.000000, nearest id = 1 of last pose = xyz=0.000000,0.000000,0.000000 rpy=0.000000,-0.000000,0.000000, odom = xyz=0.000000,0.000000,0.000000 rpy=0.000000,-0.000000,0.000000

matlabbe, 2020-11-07:
I am not sure I understand the question. What is the version of rtabmap you are using? In latest code, when we set `RGBD/SavedLocalizationIgnored` to true, it is assumed that the robot starts from origin of the map.

rahulmoorthy19, 2020-11-07:
Yea...Is solved that error. Closing the issue

closed 2020-11-07 by rahulmoorthy19

Delbina, 2021-06-28:
> Yea...I solved that error. Closing the issue

Hi,
I have the same issue. in which file you add this parameter? how did you solve this issue?
thanks