introlab/rtabmap #626: Occupied cells in grid_map misaligned

naktamello, 2020-11-02:
Hello Mathieu,
I would like to report a strange behavior when using 2D laser scan as input (melodic-devel).

The problem is same as reported by [this post](http://official-rtab-map-forum.67519.x6.nabble.com/rtabmap-grid-map-not-aligned-with-2d-scan-td6595.html) in rtabmap forum.

As can be seen in the images below, occupied cells are translated while empty cells are correctly aligned with laser scan.

[image]
[image]

Here is my db file: [database file](https://drive.google.com/file/d/19CgfixXCYV_UUs_EChNEwUiGJBt7jHKV/view?usp=sharing)

Currently I'm using the following workaround to fix the issue (referring to file [util3d_mapping.cpp](https://github.com/introlab/rtabmap/blob/melodic-devel/corelib/src/util3d_mapping.cpp#L107-L144) ):
The problem goes away when I populate `occupied` using `map8S` returned by `create2DMap`
instead of `occupied = scanHit.clone();`
(It also seemed strange to me that `create2DMap` is doing some work to collect occupied cells ([L663-L685](https://github.com/introlab/rtabmap/blob/melodic-devel/corelib/src/util3d_mapping.cpp#L663-L685)) just to be discarded.)
```
        std::list<int> emptyIndices;
	std::list<int> occupiedIndices;
	for(unsigned int i=0; i< map8S.total(); ++i)
	{
		if(map8S.data[i] == 0)
		{
			emptyIndices.push_back(i);
		}
		else if (map8S.data[i] == 100){
			occupiedIndices.push_back(i);
		}
	}

    . . .

	occupied = cv::Mat();
	if(occupiedIndices.size())
	{
		occupied = cv::Mat(1, (int)occupiedIndices.size(), CV_32FC2);
		int i = 0;
		for(std::list<int>::iterator iter=occupiedIndices.begin();iter!=occupiedIndices.end(); ++iter)
		{
			int y = *iter / map8S.cols;
			int x = *iter - y*map8S.cols;
			cv::Vec2f * ptr = occupied.ptr<cv::Vec2f>();
			ptr[i][0] = (float(x))*cellSize + xMin;
			ptr[i][1] = (float(y))*cellSize + yMin;
			++i;
		}
	}
```

matlabbe, 2020-11-07:
Which rtabmap version are you using? When I regenerate the occpuancy grids in databaseViewer, it seems ok:
Before:
[image]

After:
[image]

naktamello, 2020-11-08:
I am using 0.20.0 built from source (tagged 0.20.0-melodic). The same issue occurred with version installed with apt.

Can it be that database-viewer doesn't use `occupancy2DFromLaserScan` from util3d_mapping.cpp?

It looks like empty cells go through `create2DMap` which transforms scan data with `util3d::transformLaserScan` with pose provided, while occupied cells are copied directly: `occupied = scanHit.clone();`.

Maybe there is a mismatch between an assumption rtabmap makes about how the frames should be aligned and my setup?(map, laser, odom)

If you have any guesses, please let me know I will do some tests and report back.

matlabbe, 2020-11-08:
As you are using rtabmap from source, can you try master branch? DatabaseViewer uses the same code for regenerating the occupancy grid.

naktamello, 2020-11-10:
Yes indeed, the above commit fixes the issue. It is working as expected in the latest master branch. (rtabmap ver 0.20.6).

Thank you.

closed 2020-11-10 by naktamello