introlab/rtabmap #629: L515 - Windows RTAB-MAP support

shuixing85, 2020-11-04:
Hi all,

Just wondering is it possible to package an EXE package to support realsense L515 camera?
or else, is there an alternative way to access the "*.bag" file generated by the intel realsense viewer?

Many thanks for any replies/helps,
William

matlabbe, 2020-11-07:
Not officially supported (as I could not test it myself yet), but I updated the windows binaries created from automated build to include realsense2 driver. You can give a try: https://ci.appveyor.com/project/matlabbe/rtabmap/branch/master/artifacts

shuixing85, 2020-11-08:
> Not officially supported (as I could not test it myself yet), but I updated the windows binaries created from automated build to include realsense2 driver. You can give a try: https://ci.appveyor.com/project/matlabbe/rtabmap/branch/master/artifacts

A BIG THANKS!
I spent almost a whole weekend trying to build from source, and I had 0 experience on those VS/coiding, so it progresses rather slowly.

shuixing85, 2020-11-08:
Hi there, thanks for the help on the buildings again, 

tried to run the exe file and zip file and get this error:

it seems L515 is not listed in the support list, 
[image]

if you don't mind, could you please spend 5mins to include L515 and rebuild it again?

Much Appreciated, 
William

matlabbe, 2020-11-08:
Are you source you downloaded the version here? https://ci.appveyor.com/project/matlabbe/rtabmap/branch/master/artifacts

It seems you are using rtabmap 0.20.3, not 0.20.6.

shuixing85, 2020-11-09:
> Are you source you downloaded the version here? https://ci.appveyor.com/project/matlabbe/rtabmap/branch/master/artifacts
> 
> It seems you are using rtabmap 0.20.3, not 0.20.6.

Ah, silly me!
Yeah you are right, I'm using the wrong package!
Tried the zip package and L515 is recognized as normal!
However, I encountered the problem here:
[image]

https://github.com/introlab/rtabmap/issues/614#issuecomment-713042309

I can see only one step to the goal, can you please help me to fix this?

AndreHeinz, 2020-11-19:
Same message here...

matlabbe, 2020-11-19:
Can you try this: https://github.com/introlab/rtabmap/issues/614#issuecomment-721266476

AndreHeinz, 2020-11-19:
Dear Mathieu

Not working.
Checked physical connection with realsense viewer, ok. [Image]

Settings:
[Image]

Result:
Given stream config is not supported by the device.
And your new commited change "L500 is detected...."
[Image]

I'm gonna give it up and try it with ros on ubuntu.

Freundliche Grüsse
Andreas Heinz
Bauleiter

Walter Dietsche
Baumanagement AG
Loëstrasse 99, CH-7000 Chur
Tel. +41 81 353 44 88
Mob. +41 79 861 21 31
www.wdietsche.ch<http://www.wdietsche.ch>

Can you try this: #614 (comment)<https://github.com/introlab/rtabmap/issues/614#issuecomment-721266476>

matlabbe, 2020-11-19:
If you are going on Ubuntu, it will be easier to build rtabmap from source. If you do so, you can try those modifs of this [comment](https://github.com/introlab/rtabmap/issues/614#issuecomment-713072426). In particular those edits:

> And this [line](https://github.com/introlab/rtabmap/blob/fbc30042c4d0b1a25b29ea3876e2bf625ec9bb42/corelib/src/camera/CameraRealSense2.cpp#L1009)
to
```cpp
if((!isL500_) && sensors[i].supports(rs2_option::RS2_OPTION_GLOBAL_TIME_ENABLED))
```

> [Desired frames](https://github.com/introlab/rtabmap/blob/fbc30042c4d0b1a25b29ea3876e2bf625ec9bb42/corelib/src/camera/CameraRealSense2.cpp#L1145) is also problematic if the IMU info doesnt arrive on time, comment it out
```cpp
int desiredFramesetSize = 2;
//if(isL500_)
// desiredFramesetSize = 3;
```

matlabbe, 2020-11-19:
I think the last commit had a bug, because if you set resolution 640x480 with 30 FPS, you should not see the error message "L500 sensor is detected". It could be fixed by this commit https://github.com/introlab/rtabmap/commit/fdaaa6ccfa42f591f7bc792e6bc05239211de666

AndreHeinz, 2020-11-19:
Many many thanks for your support on this Topic.
Very kind.

I will give it a try at weekend and provide feedback.

And perhaps i can give a bit back, even if coding skills are just basic level.
f.e.: need some german localization in the gui?

Regards from switzerland…
Andi

Anderson191919, 2020-11-22:
Hello guys,
I have an issue with the L515 on windows with rtabmap (v 0.20.6 from https://ci.appveyor.com/project/matlabbe/rtabmap/branch/master/artifacts). unfortunately I get following error(s) when starting on rtabmap

Anderson191919, 2020-11-22:
[image]

Anderson191919, 2020-11-22:
I have used these settings (#614 (comment)).

Thank you very much in advance.
best regards

Anderson191919, 2020-11-22:
[image]

matlabbe, 2020-11-22:
The workaround stated [here](https://github.com/introlab/rtabmap/issues/614#issuecomment-731782168) may fix the problem. However, if you still get that long list of errors, make sure to include the first error message just over "Available configurations...".

shuixing85, 2020-12-15:
> Hello guys,
> I have an issue with the L515 on windows with rtabmap (v 0.20.6 from https://ci.appveyor.com/project/matlabbe/rtabmap/branch/master/artifacts). unfortunately I get following error(s) when starting on rtabmap

Hi Anderson, 
I originally run this on my desktop PC without any errors 
and run another time on a laptop and similar errors jump up,

I think the main problem is RTAB cannot decide which cameras are available to access, I disabled the laptop camera and leaving the L515 as the only connected one, and it works.

Hope this can help you,
William

matlabbe, 2020-12-15:
@shuixing85 Thanks for the information. Not sure if it can help, but we can select a different device in Source->"ID of the device".

roppppp, 2021-11-01:
Regarding .bag files : open3d also supports them , I dont know to which extent but see for yourself
http://www.open3d.org/docs/release/tutorial/sensor/realsense.html#reading-from-realsense-bag-files
I think their latest release also supports real time capture and processing of l515
https://www.youtube.com/watch?v=pLCVCH7ypI4

One should make a versus of realtime RTabMap vs Open3d