introlab/rtabmap #632: iPhone Lidar

Zumbalamambo, 2020-11-09:
Just like the RTABMap for Tango devices, is there any app for iOS devices with Lidar as well? :)

matlabbe, 2020-11-14:
RTAB-Map for iOS is in the pipeline for 2021

Zumbalamambo, 2020-12-19:
@matlabbe I saw that iOS point cloud has been used in the [code](https://github.com/introlab/rtabmap/blob/f9abcf9e353d72c7eedd9eafe4c788a863f0b323/corelib/src/camera/CameraImages.cpp).

Any instructions on how I can use it? :)

matlabbe, 2020-12-19:
This approach uses data exported from 3d Scanner App on IOS. In Preferences dialog, select RGB-D Images source, scroll down to set RGB and depth paths. Set decimation to 2 so that default features work correctly. RGB images + JSON files should be in the same folder and depth images in another folder. Check the option "Load config file for each frame". Optionally, you can enable RGB-D SLAM -> "Use odometry instead of IMU orientation..." with Graph Optimization->"Gravity sigma..." to 0.1.

This is experimental, but it can give a glimpse of what we can achieve with iPhone/iPad Lidar.

rooi, 2021-03-31:
Hi,

Could you please indicate which 3D Scanner App is used to export data? There are multiple apps in the App Store called 3D Scanner App and all of them seem to build 3D models of objects and/or rooms themselves and allow export in a 3D format instead of RGB and depth images.
E.g.: This app: https://www.3dscannerapp.com

There is also the Record3D app but this app allows exporting RGBD video's

Thanks!

matlabbe, 2021-04-02:
Yes this one (https://www.3dscannerapp.com/) would export data in image directory.

rooi, 2021-04-09:
Thanks @matlabbe 

I can browse the scan using the files app and locate the jpgs and json files. The export all data button seems to export the same data.
However, I don't see the depth images and the jpg don't seem to include depth either (checked using exiftool). How did you find the depth images, or am I overlooking something?

The following files are in the export folder:
world_map.arkit
frame_00000.jpg
frame_00001.jpg
...
textured_output.jpg
thumb_00000.jpg
frame_00000.json
frame_00001.json
...
info.json
textured_output.mtl
export.obj
export_refined.obj

closed 2021-04-09 by Zumbalamambo

matlabbe, 2021-04-27:
For those interested to get a beta version of RTAB-Map on iOS, go to http://official-rtab-map-forum.67519.x6.nabble.com/Beta-testers-wanted-for-iPhone-iPad-with-LiDAR-td7842.html

cheers,
Mathieu

matlabbe, 2021-06-09:
Released on App Store: https://apps.apple.com/ca/app/rtab-map-3d-lidar-scanner/id1564774365

matlabbe, 2021-11-23:
@rooi With latest 3D Scanner App, it doesn't record as many depth images than RGB images. We have to remove JSON files and depth images who don't have corresponding RGB frame. For example here I put all JSON and RGB files in same rgb directory:
[image][image]
Same for the depth directory:
[image][image]

The instructions of this [post](https://github.com/introlab/rtabmap/issues/632#issuecomment-748525481) above should still work to use RGB-D Images input source:
[image]

360wcui, 2023-08-17:
Does the 3D Scanner App method work in 2023?  The scanner app now produces conf_*.png files.

matlabbe, 2023-08-18:
Maybe not, I am not following thoroughly 3D Scanner App. `conf` sounds like confidence images.