introlab/rtabmap #633: Compiling rtabmap with libtorch 1.7.0

jsseng, 2020-11-10:
I am trying to compile rtabmap with libtorch 1.7.0.  I am getting this error regarding C++14:

```
In file included from /home/jseng/Downloads/libtorch/include/torch/csrc/api/include/torch/torch.h:3:0,
                 from /home/jseng/rtabmap/corelib/src/superpoint_torch/SuperPoint.h:9,
                 from /home/jseng/rtabmap/corelib/src/Features2d.cpp:48:
/home/jseng/Downloads/libtorch/include/torch/csrc/api/include/torch/all.h:4:2: error: #error C++14 or later compatible compiler is required to use PyTorch.
 #error C++14 or later compatible compiler is required to use PyTorch.
```

I am using gcc 7.5.0 (Ubuntu 18.04) and I thought C++14 was standard.  I also tried setting -std=C++14, but I am not sure if it is being reset somewhere.

matlabbe, 2020-11-10:
Try with with a newest cmake version (at least 3.16).

jsseng, 2020-11-10:
That did the trick!  Can I ask a question relating to SuperPoint run time?  I am now getting update times of greater than 1 sec. with SuperPoint.  I believe it is using the GPU when I use the command to monitor usage:  nvidia-smi -l 1 (I can see rtabmap_ros).  Does a typical forward pass through SuperPoint take this long typically?  Thanks!

matlabbe, 2020-11-14:
Which kind of GPU are you using? On my GTX 1650, I get more like 300-350 ms per frame. On CPU, I can see over 1 second though.

jsseng, 2020-11-14:
I am using an old K620.  I timed the forward pass of the network and it took about 600ms.  I am using images from a Zed 2 (1280x720).  When I increased the SuperPoint threshold, it brought the overall runtime down (around 800ms total including NMS).  I have been trying it on a Jetson Xavier and I was getting similar numbers at max power mode.  What resolution images do you use?

jsseng, 2020-11-15:
I reduced my resolution and it runs faster.  I will go ahead and close this.

closed 2020-11-15 by jsseng

matlabbe, 2020-11-16:
In this [video](https://www.youtube.com/watch?v=G-5jesjNfLc), loop closure detection was done with SuperPoint features using MyntEyeS images of resolution 752x480.