introlab/rtabmap #637: Consistent "Segmentation fault (core dumped)" crashes during SLAM using Azure Kinect

aashutosh1997, 2020-11-12:
I am getting segmentation faults after some time of SLAM using Azure Kinect device, as shown on screenshot below
[image]

Upon further inspection using GDB, found the error to be in malloc 
```
...
[New Thread 0x7fff1dffb700 (LWP 23856)]
[New Thread 0x7fff1ffff700 (LWP 23857)]
[Thread 0x7fff1ffff700 (LWP 23857) exited]
[Thread 0x7fff1dffb700 (LWP 23856) exited]

Thread 8 "rtabmap" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff3f7fe700 (LWP 23451)]
__GI___libc_free (mem=0x145) at malloc.c:3113
3113	malloc.c: No such file or directory.
```
[image]

It seems like the program is trying to free illegal memory, I'm not sure where this is occurring, how do I solve it?

tkircher, 2020-11-12:
https://github.com/ethz-asl/eigen_catkin/wiki/Eigen-Memory-Issues

matlabbe, 2020-11-12:
What is the output of your cmake?
```
$ cd rtabmap/build
$ cmake ..
```

That kind of bug is indeed often related to Eigen issues like @tkircher referred. Most notably, the incompatible use of `-march=native` flag with PCL, G2O and GTSAM. Either all or none of those libraries should be built with `-march=native`. 

PCL binaries ar built without that flag, so g2o and gtsam should disable that flag too before building them.

IF PCL is built from source, it has probably added `-march=native` by default. If it is the case, when doing cmake on rtabmap, it will tell if PCL has been built with `-march=native` or not, if so, make sure that g2o and/or gtsam are also built from source with that flag.

With g2o, set either `cmake -DBUILD_WITH_MARCH_NATIVE=ON` or `cmake -DBUILD_WITH_MARCH_NATIVE=OFF` depending if PCL is built with `-march=native` or not.

aashutosh1997, 2020-11-12:
@matlabbe thanks for the quick response.
I indeed got pcl from ```apt get```, and built g2o with march-native ON because otherwise it was throwing the following upon launching rtabmap.
```
double free or corruption (out)
Aborted (core dumped)
```

The other option would be to build pcl from source with march=native, I'll try that and get back to the issue.

matlabbe, 2020-11-16:
I would not rebuild PCL from source, to save you time. When you build g2o without `-march=native`, show backtrace of gdb for that error:
```
double free or corruption (out)
Aborted (core dumped)
```
Can you also show cmake output of rtabmap?

aashutosh1997, 2020-11-23:
@matlabbe Sorry about the late response, I am guilty of not taking your advice and building pcl anyway, but it didn't help. So just kept using the docker version for the past week.

Anyway the following build was done with MARCH_NATIVE set to OFF. The cmake output for rtabmap is as follows
```
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Deprecation Warning at CMakeLists.txt:67 (cmake_policy):
  The OLD behavior for policy CMP0043 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found CUDA: /usr/local/cuda (found suitable exact version "10.2") 
-- Checking for module 'eigen3'
--   Found eigen3, version 3.3.7
-- Found Eigen: /usr/local/include/eigen3 (Required is at least version "3.1") 
-- Eigen found (include: /usr/local/include/eigen3, version: 3.3.7)
-- Checking for module 'libusb-1.0'
--   Found libusb-1.0, version 1.0.21
CMake Warning (dev) at /usr/local/share/cmake-3.17/Modules/FindPackageHandleStandardArgs.cmake:272 (message):
  The package name passed to `find_package_handle_standard_args` (USB_10)
  does not match the name of the calling package (OpenNI).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /usr/local/share/pcl-1.10/Modules/FindOpenNI.cmake:28 (find_package_handle_standard_args)
  /usr/local/share/pcl-1.10/PCLConfig.cmake:176 (find_package)
  /usr/local/share/pcl-1.10/PCLConfig.cmake:319 (find_openni)
  /usr/local/share/pcl-1.10/PCLConfig.cmake:543 (find_external_library)
  CMakeLists.txt:213 (FIND_PACKAGE)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found USB_10: /usr/lib/x86_64-linux-gnu/libusb-1.0.so  
-- OpenNI found (include: /usr/include/ni, lib: /usr/lib/libOpenNI.so)
-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.3/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
[... 335 lines cut ...]

Program received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  0x00007ffff4ef3fb7 in __GI_raise (sig=sig@entry=6)
    at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff4ef5921 in __GI_abort () at abort.c:79
#2  0x00007ffff4f3e967 in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7ffff506bb0d "%s\n") at ../sysdeps/posix/libc_fatal.c:181
#3  0x00007ffff4f459da in malloc_printerr (str=str@entry=0x7ffff506d818 "double free or corruption (out)") at malloc.c:5342
#4  0x00007ffff4f4cf6a in _int_free (have_lock=0, p=0x555555af0630, av=0x7ffff52a0c40 <main_arena>) at malloc.c:4308
#5  0x00007ffff4f4cf6a in __GI___libc_free (mem=0x555555af0640)
    at malloc.c:3134
#6  0x00007fffe452e36d in g2o::EdgeSE2PointXYCalib::~EdgeSE2PointXYCalib() ()
    at /usr/local/lib/libg2o_types_slam2d.so
#7  0x00007fffe47a1885 in g2o::Factory::registerType(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, g2o::AbstractHyperGraphElementCreator*) () at /usr/local/lib/libg2o_core.so
#8  0x00007fffe4523062 in __static_initialization_and_destruction_0(int, int) [clone .constprop.26] () at /usr/local/lib/libg2o_types_slam2d.so
#9  0x00007ffff7de38f3 in call_init (env=0x7fffffffdcd8, argv=0x7fffffffdcc8, argc=1, l=<optimized out>) at dl-init.c:72
#10 0x00007ffff7de38f3 in _dl_init (main_map=0x7ffff7ffe170, argc=1, argv=0x7fffffffdcc8, env=0x7fffffffdcd8) at dl-init.c:119
#11 0x00007ffff7dd40ca in _dl_start_user () at /lib64/ld-linux-x86-64.so.2
---Type <return> to continue, or q <return> to quit---
#12 0x0000000000000001 in  ()
#13 0x00007fffffffe09b in  ()
#14 0x0000000000000000 in  ()
```

matlabbe, 2020-11-23:
It looks like PCL is still built with `-march=native`:
```
--   PCL_VERSION = 1.10.1
--   PCL_COMPILE_OPTIONS = -march=native;-msse4.2;-mfpmath=sse
```
If you want to avoid recompiling PCL (with can be long), you could try rebuilding g2o with `-march=native` instead:
```
cd g2o/build
cmake -DBUILD_WITH_MARCH_NATIVE=ON ..
make
make install
```

aashutosh1997, 2020-11-23:
Well yes it does build successfully if I build g2o with ```cmake -DBUILD_WITH_MARCH_NATIVE=ON ..``` but it still crashes because of the original issue which was this

> I am getting segmentation faults after some time of SLAM using Azure Kinect device, as shown on screenshot below
> [image]
> 
> Upon further inspection using GDB, found the error to be caused while trying to free illegal memory:
> 
> ```
> ...
> Thread 8 "rtabmap" received signal SIGSEGV, Segmentation fault.
> [Switching to Thread 0x7fff3f7fe700 (LWP 30218)]
> __GI___libc_free (mem=0x135) at malloc.c:3113
> 3113	malloc.c: No such file or directory.
> (gdb) b
> Breakpoint 1 at 0x7ffff4f58a6d: file malloc.c, line 3113.
> (gdb) bt
> #0  0x00007ffff4f58a6d in __GI___libc_free (mem=0x135) at malloc.c:3113
> #1  0x00007ffff6ea132f in g2o::SparseBlockMatrix<Eigen::Matrix<double, -1, -1, 0, -1, -1> >::~SparseBlockMatrix() () at /usr/local/lib/librtabmap_core.so.0.20
> #2  0x00007ffff6e98b1c in rtabmap::OptimizerG2O::optimize(int, std::map<int, rtabmap::Transform, std::less<int>, std::allocator<std::pair<int const, rtabmap::Transform> > > const&, std::multimap<int, rtabmap::Link, std::less<int>, std::allocator<std::pair<int const, rtabmap::Link> > > const&, cv::Mat&, std::__cxx11::list<std::map<int, rtabmap::Transform, std::less<int>, std::allocator<std::pair<int const, rtabmap::Transform> > >, std::allocator<std::map<int, rtabmap::Transform, std::less<int>, std::allocator<std::pair<int const, rtabmap::Transform> > > > >*, double*, int*) () at /usr/local/lib/librtabmap_core.so.0.20
> #3  0x00007ffff6b5ea48 in rtabmap::Rtabmap::optimizeGraph(int, std::set<int, std::less<int>, std::allocator<int> > const&, std::map<int, rtabmap::Transform, std::less<int>, std::allocator<std::pair<int const, rtabmap::Transform> > > const&, bool, cv::Mat&, std::multimap<int, rtabmap::Link, std::less<int>, std::allocator<std::pair<int const, rtabmap::Link> > >*, double*, int*) const ()
>     at /usr/local/lib/librtabmap_core.so.0.20
> #4  0x00007ffff6b5ed24 in rtabmap::Rtabmap::optimizeCurrentMap(int, bool, std::map<int, rtabmap::Transform, std::less<int>, std::allocator<std::pair<int const, rtabmap::Transform> > >&, cv::Mat&, std::multimap<int, rtabmap::Link, std::less<int>, std::allocator<std::pair<int const, rtabmap::Link> > >*, double*, int*) const () at /usr/local/lib/librtabmap_core.so.0.20
> #5  0x00007ffff6b73af4 in rtabmap::Rtabmap::process(rtabmap::SensorData const&, ---------Type <return> to continue, or q <return> to quit---
> rtabmap::Transform, cv::Mat const&, std::vector<float, std::allocator<float> > const&, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, float, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, float> > > const&) ()
>     at /usr/local/lib/librtabmap_core.so.0.20
> #6  0x00007ffff6b9736f in rtabmap::RtabmapThread::process() ()
>     at /usr/local/lib/librtabmap_core.so.0.20
> #7  0x00007ffff6b98f38 in rtabmap::RtabmapThread::mainLoop() ()
>     at /usr/local/lib/librtabmap_core.so.0.20
> #8  0x00007ffff67fdb82 in UThreadC<void>::ThreadMainHandler(UThreadC<void>::Instance*) () at /usr/local/lib/librtabmap_utilite.so.0.20
> #9  0x00007ffff2f4b6db in start_thread (arg=0x7fff3f7fe700) at pthread_create.c:463
> #10 0x00007ffff4fe271f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
> ```
> 
> [image]
> 
> Looks like it is related to g2o, how do I solve it?

So probably the problem is with Eigen like @tkircher mentioned

tkircher, 2020-11-23:
Definitely take a look at this:

https://github.com/RainerKuemmerle/g2o/issues/445

There's basically no distribution that has recent enough versions of libraries for what I'm doing, so I have to build everything from source, starting from libusb. It is actually very difficult to get magma, suitesparse, eigen, libpcl, opencv, and so on to all work together with high performance. For most purposes, it makes sense to trade CPU performance for stability and at least in this case make up for the difference with CUDA.

matlabbe, 2020-11-23:
The PCL binaries with "apt install" are built **without** `-march=native`, it is why I recommend using the binaries. The g2o binaries are also built without `-march=native`. I think that GTSAM binaries are also built without `-march=native`.

aashutosh1997, 2020-11-24:
When I built pcl from source, I couldn't find the architecture flag. However, rtab-map cmake output was showing that "-march=native" flag was used as you pointed out. 

https://github.com/PointCloudLibrary/pcl/issues/2239#issuecomment-370753752

It looks like these architecture flags are being added unconditionally, even in the PCL1.8.1 binary for bionic.

aashutosh1997, 2020-11-24:
I checked the generated ```PCLConfig.cmake``` for my build of pcl and it did have "-march=native" appended as mentioned here https://github.com/PointCloudLibrary/pcl/issues/2239#issuecomment-370753752.

However the code that you mentioned is the current master so ```-DPCL_ENABLE_SSE=OFF``` should do that.

I have also confirmed that ```PCLConfig.cmake``` in PCL1.8.1 binary for bionic does not have the architecture flags appended so it is not built with "-march=native" flag as you have mentioned. However RTAB-Map is still crashing when I use g2o with -DBUILD_WITH_MARCH_NATIVE=OFF

> it was throwing the following upon launching rtabmap.
> 
> ```
> double free or corruption (out)
> Aborted (core dumped)
> ```

For now I built rtab-map without g2o, and am just using the ceres solver which works with no problem.

matlabbe, 2020-11-24:
you can do `make VERBOSE=1` to make sure the `-march-native` is not used when building g2o. I am currently using g2o (master up to date) built from source on bionic with pcl binaries (1.8.1) and I don't have problems. 

```bash
$ make VERBOSE=1
[...]
[  6%] Building CXX object g2o/core/CMakeFiles/core.dir/optimizable_graph.cpp.o
cd /home/mathieu/workspace/g2o/build/g2o/core && /usr/bin/c++  -DUNIX -Dcore_EXPORTS -I/usr/include/eigen3 -I/home/mathieu/workspace/g2o -I/home/mathieu/workspace/g2o/build  -Wall -W -O3 -DNDEBUG -O3 -fPIC   -msse2 -msse3 -mssse3 -msse4.1 -msse4.2 -std=gnu++11 -o CMakeFiles/core.dir/optimizable_graph.cpp.o -c /home/mathieu/workspace/g2o/g2o/core/optimizable_graph.cpp
[  6%] Linking CXX shared library ../../../lib/libg2o_core.so
```

aashutosh1997, 2020-11-25:
It is working now, just had to make sure that nothing (pcl, gtsam and g2o) is built for native architecture. PCL in apt repository is not built with ```-march=native``` flag, but if built from source there is no such flag in CMakeList, instead ```-DPCL_ENABLE_SSE=OFF``` would disable automatic append like Mathieu said. 

Really appreciate your help here @matlabbe and @tkircher. I'm closing this.

closed 2020-11-25 by aashutosh1997