introlab/rtabmap #640: addData() Missing some information (images empty or missing calibration)!?

ErfolgreichCharismatisch, 2020-11-14:
Hello.
I am using an avi video rgb as input.
I calibrated the camera. The video plays calibrated using test, so the avi can be loaded.
I press okay
I press the play button
I get
`[ERROR] (2020-11-14 13:13:27) OdometryThread.cpp:139::rtabmap::OdometryThread::addData() Missing some information (images empty or missing calibration)!?`
And nothing happens.

ErfolgreichCharismatisch, 2020-11-14:
It doesn't work with monocular video.

matlabbe, 2020-11-30:
Yes, "It doesn't work with monocular video." RTAB-Map is not a mono-SLAM approach.

closed 2020-11-30 by matlabbe

nirmalsnair, 2024-02-26:
Thanks for the clarification @matlabbe, I was also running into this same issue. Perhaps remove the RGB option from the menu to avoid confusion or is there any reason for retaining the RGB source type?

matlabbe, 2024-02-27:
For specific applications, we can use RGB input. For examples:
* Appearance-based loop closure detection only ([original RTAB-Map < 2013](https://introlab.3it.usherbrooke.ca/mediawiki-introlab/images/b/bc/TRO2013.pdf))
* Global localization of monocular camera with a map previously created with RGB-D or stereo cameras.
* Enabling 2D-2D registration approach with F2F->optical flow (experimental monocular SLAM)

Normally, you should get a message box or errors if the current config with monocular camera is not supported. To avoid confusion, I [removed the option of the main menu](https://github.com/introlab/rtabmap/commit/6f7e5cc97b03f3794e9d001d9bf97dfda9fa2996) (still available under Preferences->Source dialog for the examples above).

cheers,
Mathieu

nirmalsnair, 2024-02-27:
Oh, I see, thanks @matlabbe.