introlab/rtabmap #644: Bizarre and persistent noise from Azure Kinect Motion Sensor (IMU)

aashutosh1997, 2020-11-23:
I have been using the Azure Kinect device for SLAM with RTAB-Map, the result however is not great: more often than not there is a bizarre jump in the pose graph, this causes loop closures to fail and as a result there are multiple artifacts from the same structure in the real world.

For example, look at the pose graph below, the device was simply held stable in a rotating chair, and all of a sudden there is a jump.
[image]
[image]

I have to believe that this is because of noise in the motion sensor data, or error in retrieving and processing the data. Could this just be a limitation of the Azure Kinect sensor, or any sort of physical interference?

matlabbe, 2020-11-23:
It is like the odometry jumps. Is this the default 720p image? You may try another kind of features, like ORB-Octree under Preferences->Visual registration->Feature Detector (scroll down). You can also tune GFTT distance. See "Important" section here: https://github.com/introlab/rtabmap/releases/tag/0.20.2:

> Important: some new cameras can output HD images (and even higher resolutions), the default GFTT/BRIEF feature doesn't work well with high resolution images. It is recommended to increase GFTT/MinDistance to 10 for example or use scale invariant features. The new ORB-OCTREE approach can be a good choice to keep speed with binary descriptors.

You can also try IR mode to use the large field of view of the depth sensor, or even switch to ICP odometry: http://official-rtab-map-forum.67519.x6.nabble.com/Kinect-For-Azure-L515-ICP-lighting-invariant-mapping-td7187.html

Regards,
Mathieu

aashutosh1997, 2020-11-24:
I was infact using 1080p images, but the problem existed with 720p as well. Changing the feature detector showed promise of improvement, increasing the Minimum Euclidean Distance to 10 for GFFT (using GFFT+ORB) worked better than using out-of-box ORB-Octree.

Closing this one, thanks alot @matlabbe !

closed 2020-11-24 by aashutosh1997