introlab/rtabmap #645: Question about RTABMAP Signatures and VWDictionary

jsseng, 2020-11-25:
I am working on adding a feature to cache a pre-built map to use with RTABMAP in localization mode.  The idea is to save the VWDictionary and the flann Index since these structures take a while to load on startup.  I have code which will save the VWDictionary to a binary file and also the flann Index (about 750,000 descriptors).  Before saving, the flann index trees are re-built into a contiguous block of memory, so they can be quickly loaded into a memory block at startup.

My question is does the initialization of Signatures here:

https://github.com/introlab/rtabmap/blob/7859313bebb988cc710703894eb457fdeb47c04a/corelib/src/Memory.cpp#L231

 set any pointers into the _vwd dictionary?  If so, then I may have problems with reloading the _vwd dictionary.

Basically, my approach has been once this line has been reached in the code,

https://github.com/introlab/rtabmap/blob/7859313bebb988cc710703894eb457fdeb47c04a/corelib/src/Memory.cpp#L406

I save all the contents of _vwd and the flann index.  Sometimes, it loads the contents back correctly, but now I feel like I am missing some pointers into the _vwd contents.

I have a repository I can share if you would like to take a look at what I am trying to do with the loadDataFromDb() function.  I have been getting about a factor of 10x speed up on initialization.

matlabbe, 2020-11-30:
Hi,

the dictionary is loaded here on startup: 
https://github.com/introlab/rtabmap/blob/96c1c81e22959009b236fc9bd2bbb3cd06daa65c/corelib/src/Memory.cpp#L402

Then just after we populate the inverted index:
https://github.com/introlab/rtabmap/blob/96c1c81e22959009b236fc9bd2bbb3cd06daa65c/corelib/src/Memory.cpp#L409-L429

Note sure afterwards what you are missing. This is however a very interesting feature to add to increase initialization time. You could share me a fork with the modifications, maybe we could save the compressed vocabulary inside the database file directly instead of a separate file.

cheers,
Mathieu

jsseng, 2020-11-30:
Hi Matthieu,
  Thanks for the response.  I had to hack some files in the flann library in order to get into the classes, but if it is worth pursuing, I can keep cleaning it up.  Everything seems to be working as I found a bug on my part.  Here is the repository:  https://github.com/jsseng/herbie_rtabmap

  A few things:

  * I have found that this line:

https://github.com/introlab/rtabmap/blob/96c1c81e22959009b236fc9bd2bbb3cd06daa65c/corelib/src/Memory.cpp#L395

does not get run, even when I pass '--Mem/IncrementalMemory false' on the command line.  I am not sure if it is an initialization issue.

 * I am having problems adding a command line parameter.  Even when I add it to Parameters.h, because it does not show up in the QT application, I can't pass it to a Memory object (it is missing from the ParametersMap).

 * The save datafiles can be very big.  Over 1GB for the flann because the ORB 256-bit descriptors are floats and I just write them out as floats.  I think this can be optimized.  The vwdictionary is smaller (60MB for my map).

 * I am wondering what the performance would be like storing the data in the database because I think there is overhead with sqlite3 accesses.

 * I use mmap() to map the flann into the same virtual addresses every time.  This allows the pointers to be stored in the cached file.

I am getting startup times of 2.5 seconds (1.2 seconds for loading signatures and 1.3 seconds for vwdictionary and the flann index) for a map that used to take 25 seconds.  I need to try this on my Jetson Xavier which takes about 1+ minutes to load the same map.

Thanks for taking a look!