introlab/rtabmap #646: Some doubts about OccupancyGrid::segmentCloud() in OccupancyGrid.hpp

liuzhenboo, 2020-11-26:
I have some question about OccupancyGrid::segmentCloud() at line 38 in the file OccupancyGrid.hpp.
I don't understand the following sentence at line 85 and 90: 
`	// add pose rotation without yaw
`
and 
`	cloud = util3d::transformPointCloud(cloud, Transform(0,0, projMapFrame_?pose.z():0, roll, pitch, 0));
`
why add pose rotation without yaw??

Why is this code not like this:

`	cloud = util3d::transformPointCloud(cloud, Transform(pose.x(),pose.y(), pose.z():0, roll, pitch, yaw));
`
I hope to get your help, thank you very much！！

matlabbe, 2020-11-29:
Please use links like these by clicking on the line numbers in Github:
https://github.com/introlab/rtabmap/blob/0022b6b993ba68a3cfef5e63aef5daee0ba38fef/corelib/include/rtabmap/core/impl/OccupancyGrid.hpp#L85-L90

https://github.com/introlab/rtabmap_ros/blob/b19a82ef34c24da8ccf28d184f84e95a81ea6d9a/src/nodelets/obstacles_detection.cpp#L322-L332

The published segmented cloud would be in the same frame than the input cloud, the transformed cloud as above is transformed back here:
https://github.com/introlab/rtabmap_ros/blob/b19a82ef34c24da8ccf28d184f84e95a81ea6d9a/src/nodelets/obstacles_detection.cpp#L374-L393

We could use x,y and yaw, but they are not used by the segmentation approach (only z for height if we want to segment in map frame, and roll and pitch for alignment with horizon).

liuzhenboo, 2020-12-02:
> Please use links like these by clicking on the line numbers in Github: [...]

thank you very much！ I get it.

closed 2020-12-02 by liuzhenboo