introlab/rtabmap #647: What is the accuracy of the scale of the depth image

Zumbalamambo, 2020-11-28:
I have used the following code to retrieve the scales out of the depth image.
```

import cv2

# camera matrix
mtx = [8.6767822265625000e+02, 0., 4.8268579101562500e+02, 0.,
       8.6760119628906250e+02, 2.7117459106445312e+02, 0., 0., 1.]

# parse calib matrix
fx = mtx[0]
cx = mtx[2]

fy = mtx[4]
cy = mtx[5]

depth_img = cv2.imread("RTAB-Map/bundler/depth/1326.png", cv2.IMREAD_ANYDEPTH)
depth_row, depth_column = depth_img.shape

# to store 3d points
pts3d = []

for i in range(depth_row):
    for j in range(depth_column):

        # retrieve depth value
        depth = depth_img[i][j]
        x, y = i, j

        if depth > 0.0:
            x3D = (x - cx) * depth / fx
            y3D = (y - cy) * depth / fy
            z3D = depth

            # populate 3d points array
            pts3d.append([x3D, y3D, z3D])
        else:
            pts3d.append([-1, -1, -1])
      row, col = 960, 540
      pts3d = np.array(pts3d).reshape(row, col, 3)
```

1. I picked 4 two dimensional coordinates of known dimensions.  -> known scale
2. I used `pts3d` that I have measured to calculate the dimension  -> Measured scale

The known scale and the dimension scales are 40-50cm off. Are there any ways to fix it? I'm using tango by the way.

matlabbe, 2020-11-29:
Is this the camera calibration of the RGB image or the depth image? With Tango, RTAB-Map keeps the calibration of the RGB image, then rescales it to depth resolution when needed. For example, if resolution of RGB image is 4x the size of the depth image, divide `fx`, `fy`, `cx` and `cy` by 4.

Zumbalamambo, 2020-11-30:
@matlabbe it's the camera calibration value of the rgb images. Instead of diving the calibration values by 4, I have rescaled the depth images 4 times to be same as the size of the rgb image. In that case, how accurate the scale depth image could be?

matlabbe, 2020-11-30:
To use rgb calibration "as is", the depth image should be the same size than rgb image. However, scaling up a depth image is not trivial, as interpolating edges would generate false depth values, or there would be pixels without depth.

Zumbalamambo, 2020-11-30:
@matlabbe is there any option in rtabmap to output the depth image to have the same size as that of the rgb image?

matlabbe, 2020-11-30:
It depends of the depth sensor used. For standard RGB-D cameras, the depth image is often the same size than RGB images. For Tango, the depth resolution is small by default, so we keep as is (there is no advantage to scale up the depth, it won't be more accurate and will cost more memory to hold the same information).

Zumbalamambo, 2020-11-30:
So is it advicable to downscale the rgb image and the camera matrix to the same shape as that of the depth image to do any computation?

matlabbe, 2020-11-30:
In RTAB-Map, we keep RGB and depth resolution can be different without problems (as I said, we just have to rescale the calibration fx,fy,cs,cy when projecting the depth values). It depends on what you want to do.

Zumbalamambo, 2020-11-30:
I would like to use the 3d points from depth image and 2d points from rgb image to find the pose using solvepnp. So I guess I just have to downscale the camera params by 4 and also the 2d points by 4. Is this the right?

matlabbe, 2020-11-30:
In rtabmap, to get depth of a RGB pixel, we divide pixel position in RGB image by ratio of depth on rgb, then get the nearest depth value in the depth image. For example, if we have RGB images of 640x480 and depth images of 160x120, we have a RGB pixel at x=400, y=200, so the depth is at pixel x=100, y=50 in depth image. We project depth pixel (100,50) to 3D using divided RGB calibration by 4. So you have the 3D feature and its color.

Zumbalamambo, 2020-11-30:
Thank you! I will try this. Very clean implementation!

closed 2020-11-30 by Zumbalamambo