introlab/rtabmap #649: Finding a way of getting transform with respect to map given node label

rahulmoorthy19, 2020-12-01:
Hello,

I am using RTABMAP for a project. In that I have given custom labels to the nodes. As I read from the node message definition. It has a pose values but it is written in the description that "Pose from odometry not corrected". What does that exactly mean?

One more thing which I would like to ask- Is there any way to get the pose of that labeled node with respect to map frame. Please let me know the steps if any way exists. Will implement the same.

Thank You in advance

matlabbe, 2020-12-01:
Hi,

It means it is the odomety pose directly (pose in odometry frame). To have the pose in map frame, use [rtabmap_ros/MapGraph](http://docs.ros.org/en/noetic/api/rtabmap_ros/html/msg/MapGraph.html) message included in [rtabmap_ros/MapData](http://docs.ros.org/en/noetic/api/rtabmap_ros/html/msg/MapData.html) or he standalone message.

rahulmoorthy19, 2020-12-01:
I have one doubt in it..In python can we have this transform....or is there any mathematical formula to have the transform as no frame id are there to have a listener

rahulmoorthy19, 2020-12-01:
Got it matlabbe....It was written not corrected, That is why I got confused. Thanks a lot.

closed 2020-12-10 by matlabbe