introlab/rtabmap #650: Compilation Error with Ubuntu 16.04 & libpointmatcher-1.3.1

David9696, 2020-12-01:
Hi,

When I tried to compile the rtabmap package, I got the following errors:

[ 25%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Registration.cpp.o
[ 26%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/RegistrationIcp.cpp.o
/home/test/rtabmap/corelib/src/RegistrationIcp.cpp: In member function ‘virtual void rtabmap::RegistrationIcp::parseParameters(const ParametersMap&)’:
/home/test/rtabmap/corelib/src/RegistrationIcp.cpp:596:85: error: no matching function for call to ‘PointMatcherSupport::Registrar<PointMatcher<float>::Matcher>::reg(const char [23], std::shared_ptr<PointMatcherSupport::Registrar<PointMatcher<float>::Matcher>::GenericClassDescriptor<KDTreeMatcherIntensity<float> > >)’
 cher.MatcherRegistrar.reg("KDTreeMatcherIntensity", std::make_shared<Desc>() );
                                                                              ^
In file included from /usr/local/include/pointmatcher/PointMatcher.h:64:0,
                 from /home/test/rtabmap/corelib/src/RegistrationIcp.cpp:45:
/usr/local/include/pointmatcher/Registrar.h:158:8: note: candidate: void PointMatcherSupport::Registrar<Interface>::reg(const string&, PointMatcherSupport::Registrar<Interface>::ClassDescriptor*) [with Interface = PointMatcher<float>::Matcher; std::__cxx11::string = std::__cxx11::basic_string<char>]
   void reg(const std::string &name, ClassDescriptor* descriptor)
        ^
/usr/local/include/pointmatcher/Registrar.h:158:8: note:   no known conversion for argument 2 from ‘std::shared_ptr<PointMatcherSupport::Registrar<PointMatcher<float>::Matcher>::GenericClassDescriptor<KDTreeMatcherIntensity<float> > >’ to ‘PointMatcherSupport::Registrar<PointMatcher<float>::Matcher>::ClassDescriptor*’
corelib/src/CMakeFiles/rtabmap_core.dir/build.make:1531: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/RegistrationIcp.cpp.o' failed
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/RegistrationIcp.cpp.o] Error 1
CMakeFiles/Makefile2:267: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/all' failed
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2

May I know how to solve it?

Thank you.

matlabbe, 2020-12-02:
It seems similar to https://github.com/introlab/rtabmap/commit/96c1c81e22959009b236fc9bd2bbb3cd06daa65c

You may update libpointmatcher or downgrade to 1.3.0. I'll still check if we could change the preprocessor IF to 10301

matlabbe, 2020-12-02:
Well, the issue is that they [didn't increase libpointmatcher version over 1.3.1 yet](https://github.com/ethz-asl/libpointmatcher/blob/fba9de44d3916cc6e7d61481337a89a98e7a1a38/pointmatcher/PointMatcher.h#L72-L75), while the master branch is now using shared pointers. If I adjust the preprocessor version to 10301, it will break the build with latest libpointmatcher version... :(