introlab/rtabmap #651: Ghosting in map

AWilson10, 2020-12-02:
Hi. 
I have attached an image of the map in rtab. Notice the ghosting affect. I think this may be why we see big jumps when trying to localise within the map. Note that when localising, mapping is off. We create the map in a separate session (with a handheld d415 and a t265 for odometry). We localise on our robot using the tf tree for odometry.
I have also attached the file we use when mapping. The most important parameters I have pasted below.
Is it possible if you could point out why we get the ghosting from the attached information. 
Thanks in advanced for any information you can give.

      <param name="RGBD/LinearUpdate"             type="double"   value="0.1"/>
      <param name="RGBD/AngularUpdate"            type="double"   value="0.01"/>
      <param name="RGBD/LinearSpeedUpdate"        type="double"   value="0"/>
      <param name="RGBD/AngularSpeedUpdate"       type="double"   value="0"/>
      <param name="RGBD/SavedLocalizationIgnored" type="bool"     value="true"/>
      <param name="Rtabmap/DetectionRate"         type="string"   value="0"/>
      <param name="Bayes/VirtualPlacePriorThr"    type="string"   value="0.915"/>
      <param name="Vis/MinInliers"                type="string"   value="20"/> 
      <param name="Vis/InlierDistance"            type="string"   value="0.2"/> 
      <param name="Vis/MeanInliersDistance"       type="string"   value="20"/>
      <param name="Rtabmap/LoopGPS"               type="bool"     value="false"/>
      <param name="Mem/STMSize"                   type="string"   value="3"/>
      <param name="odom_tf_angular_variance"      type="double"   value="0.0001"/>
      <param name="odom_tf_linear_variance"       type="double"   value="0.0001"/>
      <param name="Optimizer/Strategy"            type="string"   value="1"/>
      <param name="Kp/DetectorStrategy"           type="int"      value="8"/> 
      <param name="Vis/FeatureType"               type="int"      value="8"/> 
      <param name="Vis/EstimationType"            type="int"      value="1"/>
      <param name="Vis/ForwardEstOnly"            type="bool"     value="false"/>
      <param name="Vis/RefineIterations"          type="int"      value="10"/>  
      <param name="Vis/EpipolarGeometryVar"       type="double"   value="0.017"/> 
      <param name="Vis/MaxFeatures"               type="int"      value="10000"/> 
      <param name="Vis/BundleAdjustment"          type="int"      value="1"/> 
      <param name="OdomF2M/BundleAdjustment"      type="int"      value="1"/> 
      <param name="OdomF2M/BundleAdjustmentMaxFrames"      type="int"      value="0"/> 
      <param name="g2o/Solver"                    type="int"      value="1"/> 
      <param name="g2o/Optimizer"                 type="int"      value="0"/> 
      <param name="g2o/PixelVariance"             type="double"   value="100.0"/> 
      <param name="Icp/CorrespondenceRatio"       type="string"   value="0.01"/> 

[image]

matlabbe, 2020-12-10:
The TF between the kinect and T265 may be wrong. T265 may be drifting for some reasons. Do you have the database to share?

AWilson10, 2020-12-11:
Hi. It wouldn't let me upload the file so I have put in google drive. here is the link. Thanks for your help.
I just looked through all the maps we have created and the ghosting doesnt look that bad. This may be a clue as the image above is when we are using the tf tree, on the robot  etc and localised. When i look at the map with no odometry (not on the robot just at my desktop) the map doesnt look as bad. 
https://drive.google.com/drive/folders/11v9Bq5SfmI5XzwKG6cWJq8t6dSj0K7A9?usp=sharing